espressif/esp-idf #10504: Two classic BT HID devices connecting at the same time fail with ERR_SDP (BT_SDP: process_service_attr_rsp occurs after ESP_HIDH_OPEN_EVENT).

benryves, 2023-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

DOIT ESP32 DEVKIT V1

### Power Supply used.

USB

### What is the expected behavior?

I am attempting to connect two Wii remotes (which are classic Bluetooth HID devices) to an ESP32. The Wii remotes have previously been successfully paired individually (after searching for them and sending the appropriate PIN using the BT GAP API before using the HIDH API to open the device).

If the ESP32 is reset the Wii remotes will reconnect to it automatically, with no need to search for them again via GAP API or by using the HIDH API to open the device. If only one Wii remote is switched on at a time, then both connect successfully. Ideally this should also happen if both Wii remotes are switched on at the same time.

### What is the actual behavior?

If both Wii remotes are switched on at the same time then they will both fail to connect with ERR_SDP. If a Wii remote is then switched off (resulting in bta_hh_co_close being called) then BT_HIDH generates a Device Not Found and CLOSE ERROR: NO_CONNECTION and the ESP32 crashes and reboots.

The only thing I can see that indicates what the issue might be is that when the Wii remotes connect successfully the log shows two events from BT_SDP: process_service_attr_rsp **before** the HIDH open callback is invoked. When the connections fail these BT_SDP: process_service_attr_rsp events come **after** the HIDH open callback.

### Steps to reproduce.

1. Pair two Wii remotes to an ESP32.
2. Reset the ESP32 with both Wii remotes connected, OR reset the ESP32 and switch on both Wii remotes by pressing a button on them simultaneously.
3. If the two Wii remotes try to reconnect at the same time, neither will properly reconnect. (If they stagger the times that they try to reconnect then all is fine).

### Debug Logs.

```plain
********** Successful connection: **********

W (7000) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
E (8150) BT_BTM: tBTM_SEC_DEV:0x3ffd18d4 rs_disc_pending=0
W (8400) BT_SDP: process_service_attr_rsp
W (8710) BT_SDP: process_service_attr_rsp
W (8720) BT_APPL: new conn_srvc id:23, app_id:5
I (8740) BT: HIDH open 00:21:bd:07:b6:41: 
BDA:00:21:bd:07:b6:41, Status: OK, Connected: YES, Handle: 0, Usage: GAMEPAD
Name: , Manufacturer: , Serial Number:
PID: 0x0306, VID: 0x057e, VERSION: 0x0600
Report Map Length: 217
    VENDOR   INPUT REPORT, ID:  63, Length:  21
    VENDOR   INPUT REPORT, ID:  62, Length:  21
    VENDOR   INPUT REPORT, ID:  61, Length:  21
    VENDOR   INPUT REPORT, ID:  55, Length:  21
    VENDOR   INPUT REPORT, ID:  54, Length:  21
    VENDOR   INPUT REPORT, ID:  53, Length:  21
    VENDOR   INPUT REPORT, ID:  52, Length:  21
    VENDOR   INPUT REPORT, ID:  51, Length:  17
    VENDOR   INPUT REPORT, ID:  50, Length:  10
    VENDOR   INPUT REPORT, ID:  49, Length:   5
    VENDOR   INPUT REPORT, ID:  48, Length:   2
    VENDOR   INPUT REPORT, ID:  34, Length:   4
    VENDOR   INPUT REPORT, ID:  33, Length:  21
    VENDOR   INPUT REPORT, ID:  32, Length:   6
    VENDOR  OUTPUT REPORT, ID:  26, Length:   1
    VENDOR  OUTPUT REPORT, ID:  25, Length:   1
    VENDOR  OUTPUT REPORT, ID:  24, Length:  21
    VENDOR  OUTPUT REPORT, ID:  23, Length:   6
    VENDOR  OUTPUT REPORT, ID:  22, Length:  21
    VENDOR  OUTPUT REPORT, ID:  21, Length:   1
    VENDOR  OUTPUT REPORT, ID:  20, Length:   1
    VENDOR  OUTPUT REPORT, ID:  19, Length:   1
    VENDOR  OUTPUT REPORT, ID:  18, Length:   2
    VENDOR  OUTPUT REPORT, ID:  17, Length:   1
   GAMEPAD  OUTPUT REPORT, ID:  16, Length:   1
I (8850) MAIN: Assigning slot 0

********** Failed connection: **********

W (5730) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (6670) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
E (6880) BT_BTM: tBTM_SEC_DEV:0x3ffd18d4 rs_disc_pending=0
E (7090) BT_HIDH: OPEN ERROR: ERR_SDP
E (7090) BT: HIDH open failed (-1)
W (7100) BT_HIDH: HID-Host Rcvd L2CAP disc cfm, unknown CID: 0x44
W (7160) BT_SDP: process_service_attr_rsp
W (7480) BT_SDP: process_service_attr_rsp

********** Crash after failed connection: **********

W (93351) BT_APPL: bta_hh_co_close: dev_handle = 1, app_id = 5
W (93351) BT_APPL: bta_dm_act no entry for connected service cbs
E (93361) BT_BTC: Error: cannot find device with handle 1
E (93361) BT_HIDH: Device Not Found
E (93371) BT_HIDH: CLOSE ERROR: NO_CONNECTION
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400db050  PS      : 0x00060130  A0      : 0x8014991c  A1      : 0x3ffd3d30  
0x400db050: hidh_callback at D:/Documents/Documents/GitHub/LightGunCommando/hosts/wii_remote/firmware/main/bt.c:217

A2      : 0x00000000  A3      : 0x3f405718  A4      : 0x00000004  A5      : 0x3ffd7894
A6      : 0x000000fd  A7      : 0x0000abab  A8      : 0x800db04e  A9      : 0x3ffd3d10
A10     : 0x00016cbb  A11     : 0x00000000  A12     : 0x00000000  A13     : 0xff000000
A14     : 0x00000004  A15     : 0x0000abab  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400db04d:0x3ffd3d30 0x40149919:0x3ffd3d90 0x401494b6:0x3ffd3dc0 0x4014958d:0x3ffd3e00 0x4009503d:0x3ffd3e20
0x400db04d: hidh_callback at D:/Documents/Documents/GitHub/LightGunCommando/hosts/wii_remote/firmware/main/bt.c:217

0x40149919: handler_execute at D:/Documents/Documents/Electronics/ESP/esp-idf/components/esp_event/esp_event.c:137

0x401494b6: esp_event_loop_run at D:/Documents/Documents/Electronics/ESP/esp-idf/components/esp_event/esp_event.c:582 (discriminator 3)

0x4014958d: esp_event_loop_run_task at D:/Documents/Documents/Electronics/ESP/esp-idf/components/esp_event/esp_event.c:107 (discriminator 15)

0x4009503d: vPortTaskWrapper at D:/Documents/Documents/Electronics/ESP/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: e127f984b20ef496

Rebooting...
```

boblane1, 2023-01-09:
Hi @benryves,

Thanks for reporting!

>Reset the ESP32 with both Wii remotes connected, OR reset the ESP32 and switch on both Wii remotes by pressing a button on them simultaneously.

Do the two procedures have the same result(ERR_SDP)? And can you always reproduce the issue?  I can't reproduce the issue as you described.  So please help enable the following config:

`Component config > Bluetooth > Bluedroid Options > BT DEBUG LOG LEVEL > L2CAP layer`  -> Event
`Component config > Bluetooth > Bluedroid Options > BT DEBUG LOG LEVEL > APPL layer`  -> Debug
`Component config > Log output > Default log verbosity`  -> Debug

The crash may be caused by referring to the NULL pointer in the hidh_callback function, please check the pointer before using it.

boblane1, 2023-01-10:
Hi @benryves,

I am glad you have tried to find out the reason by yourself. The root reason is that the Bluetooth stack on ESP32 only supports one SDP procedure each time. In your case, the second HID connection is occurred while the first SDP procedure has not been finished. You can work around the issue by either staggering the two reconnection or adding the additional reconnection procedures when the previous reconnection is failed. If you choose the second solution, please apply the following patch first.

[fix_hidh_connection_bug.zip](https://github.com/espressif/esp-idf/files/10381323/fix_hidh_connection_bug.zip)

benryves, 2023-01-29:
Hi @boblane1,

Thank you very much for your advice and the patch. Unfortunately, this is all a bit beyond me - I find it extremely difficult to follow the Bluedroid code as a lot of it is indirect (e.g. via callback tables), not to mention the sheer volume of it, so I'm not really sure where I'd go to fix this. I've spent a few hours digging through the code but not really closer to a proper solution.

My workaround is to count the number of successfully connected and opened devices. If an incoming connection fails to open and there are no known successfully opened devices, I wait one second then deinit() and then re-init() all Bluetooth components. By doing this the devices that failed to open do get disconnected properly (and so you can try to reconnect them without having to pull their batteries out).

It's all a bit of a hack but I'm not really sure what the proper way to handle this is, but as far as user experience is concerned this at least masks the problem a bit better than before.

boblane1, 2023-01-29:
Hi @benryves,

Maybe the deinit and re-init procedure is unnecessary. I think it's better to initiate the reconnection procedure directly, since it will not break the established connection. I will handle like the following comments.

```C
void hidh_callback(void *handler_args, esp_event_base_t base, int32_t id, void *event_data)
{
...
    case ESP_HIDH_OPEN_EVENT: {
        if (param->open.status == ESP_OK) {
            const uint8_t *bda = esp_hidh_dev_bda_get(param->open.dev);
            ESP_LOGI(TAG, ESP_BD_ADDR_STR " OPEN: %s", ESP_BD_ADDR_HEX(bda), esp_hidh_dev_name_get(param->open.dev));
            esp_hidh_dev_dump(param->open.dev, stdout);
        } else {
            ESP_LOGE(TAG, " OPEN failed!");
            // push the bda to your own task work queue
        }
        break;
    }
...
}

your_own_app_task()
{
    // 1. fetch from the work queue and check if it is reconnection event
    // 2. delay some time
    // 3. call `esp_hidh_dev_open` to trigger the reconnection
}
```

benryves, 2023-01-29:
Hi @boblane1,

Thank you for the suggestion, that would certainly seem like the easiest answer.

How would you know the BDA of the device that failed to connect? `param->open.dev` is 0 when the connection fails, and attempting to use `esp_hidh_dev_name_get` to retrieve the BDA from 0 unsurprisingly results in a LoadProhibited Guru Meditation Error.

I'm not initiating the connection myself, the Wii remote is reconnecting automatically. I don't call `esp_hidh_dev_open` in this situation, I just handle `hidh_callback` with `ESP_HIDH_OPEN_EVENT`. Is there some callback I can subscribe to to see which devices have tried to connect, maybe?

benryves, 2023-01-29:
Hi @boblane1,

I've made some changes to a couple of files to copy the BDA to the `ESP_HIDH_OPEN_EVENT`'s `esp_hidh_event_data_t` so that it can be recovered to trigger the reconnection: [propagate_hidh_open_bdaddr.patch](https://github.com/espressif/esp-idf/files/10529290/propagate_hidh_open_bdaddr.patch)

If the `ESP_HIDH_OPEN_EVENT` fails and I then try to `esp_hidh_dev_open` using the queued BDA after a short delay it does "successfully" open the device - I get a successful status report from `ESP_HIDH_OPEN_EVENT`, `esp_hidh_dev_dump` prints out information about the connected device, and the user application assigns the controller to a controller slot. However, when it tries to send data to the controller I get a series of `HID_ERR_NO_CONNECTION` errors:

```
I (4227) BT: HIDH open 00:22:d7:6a:4b:bc: 
BDA:00:22:d7:6a:4b:bc, Status: OK, Connected: YES, Handle: 1, Usage: GAMEPAD
Name: , Manufacturer: , Serial Number:
PID: 0x0306, VID: 0x057e, VERSION: 0x0600
Report Map Length: 217
    VENDOR   INPUT REPORT, ID:  63, Length:  21
    VENDOR   INPUT REPORT, ID:  62, Length:  21
    VENDOR   INPUT REPORT, ID:  61, Length:  21
    VENDOR   INPUT REPORT, ID:  55, Length:  21
    VENDOR   INPUT REPORT, ID:  54, Length:  21
    VENDOR   INPUT REPORT, ID:  53, Length:  21
    VENDOR   INPUT REPORT, ID:  52, Length:  21
    VENDOR   INPUT REPORT, ID:  51, Length:  17
    VENDOR   INPUT REPORT, ID:  50, Length:  10
    VENDOR   INPUT REPORT, ID:  49, Length:   5
    VENDOR   INPUT REPORT, ID:  48, Length:   2
    VENDOR   INPUT REPORT, ID:  34, Length:   4
    VENDOR   INPUT REPORT, ID:  33, Length:  21
    VENDOR   INPUT REPORT, ID:  32, Length:   6
    VENDOR  OUTPUT REPORT, ID:  26, Length:   1
    VENDOR  OUTPUT REPORT, ID:  25, Length:   1
    VENDOR  OUTPUT REPORT, ID:  24, Length:  21
    VENDOR  OUTPUT REPORT, ID:  23, Length:   6
    VENDOR  OUTPUT REPORT, ID:  22, Length:  21
    VENDOR  OUTPUT REPORT, ID:  21, Length:   1
    VENDOR  OUTPUT REPORT, ID:  20, Length:   1
    VENDOR  OUTPUT REPORT, ID:  19, Length:   1
    VENDOR  OUTPUT REPORT, ID:  18, Length:   2
    VENDOR  OUTPUT REPORT, ID:  17, Length:   1
   GAMEPAD  OUTPUT REPORT, ID:  16, Length:   1
I (4277) MAIN: Running: 0:02, <NO VIDEO>
I (4347) MAIN: Assigning slot 0
I (4357) MAIN: Initialising player 0 Wii remote
E (4357) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4367) BT_APPL: HID_HostWriteDev status:4
E (4417) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4417) BT_APPL: HID_HostWriteDev status:4
E (4467) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4467) BT_APPL: HID_HostWriteDev status:4
E (4517) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4517) BT_APPL: HID_HostWriteDev status:4
E (4517) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4517) BT_APPL: HID_HostWriteDev status:4
E (4567) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4567) BT_APPL: HID_HostWriteDev status:4
E (4617) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4617) BT_APPL: HID_HostWriteDev status:4
E (4667) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4667) BT_APPL: HID_HostWriteDev status:4
E (4717) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4717) BT_APPL: HID_HostWriteDev status:4
E (4767) BT_HIDH: HID_ERR_NO_CONNECTION dev_handle 1
E (4767) BT_APPL: HID_HostWriteDev status:4
I (5347) MAIN: Running: 0:03, <NO VIDEO>
I (6347) MAIN: Running: 0:04, <NO VIDEO>
I (7347) MAIN: Running: 0:05, <NO VIDEO>
I (8347) MAIN: Running: 0:06, <NO VIDEO>
I (9347) MAIN: Running: 0:07, <NO VIDEO>
I (10347) MAIN: Running: 0:08, <NO VIDEO>
I (11347) MAIN: Running: 0:09, <NO VIDEO>
```

If I then press some buttons on the controller I get similar error messages about the device not being found:
[... 2 lines cut ...]
E (355597) BT_HIDH: Device Not Found: handle 0
E (355737) BT_HIDH: Device Not Found: handle 0
E (356047) BT_HIDH: Device Not Found: handle 0
E (356137) BT_HIDH: Device Not Found: handle 0
I (356347) MAIN: Running: 5:54, <NO VIDEO>
E (357237) BT_HIDH: Device Not Found: handle 0
I (357347) MAIN: Running: 5:55, <NO VIDEO>
E (357357) BT_HIDH: Device Not Found: handle 0
```

Similar to before, it's as if the controller is connected successfully but something is not properly associated internally.

I then tried applying your earlier patch, and this does change things but not for the better - this time after the reconnection attempt with the previously-received BDA I just get an `ERR_SDP` error, and the application goes into and endless loop of trying to reconnect and failing. I undid the changes from the patch and it goes back to an initial `ERR_SDP` followed by a "successful" opening that doesn't work (the BT_HIDH: Device Not Found: handle 0 errors mentioned above).

----

I spent some time digging through `bt/host/bluedroid/stack/hid/hidh_api.c` and it looks like `HID_HostAddDev` is called for the connecting device, then `HID_HostRemoveDev` after it fails to connect, followed by `HID_HostAddDev` after the failure to open the device. When I then re-open the device, `HID_HostAddDev` is not called again with the updated data. I changed `HID_HostRemoveDev` so that it immediately returns (and doesn't remove the device), which I know is obviously not the right thing to do, however as an experiment does actually work and allows the reopened device to be properly accessible from the user application.

[... cut ...]
At this point, I'm not sure what the best solution is, short of exposing more of the inner guts of `hidh_api.c` to the user application (e.g. the `hh_cb.devices` array) so it has a better idea of what connections have failed and what it can do to recover.

----

There is also one other solution, which is to crash and trigger a reset. By doing this the Wii remotes don't know they were intentionally disconnected (as they do from an deinit()/init() cycle) and so will automatically try to reconnect when the device comes back up instead of switching themselves off. This is a pretty awful solution, but certainly the easiest...

boblane1, 2023-01-30:
Hi @benryves,

When the HID connection procedure fails for SDP error, the stack doesn't destroy the L2CAP layer connection. Try the following patch again and reconnect, please let me know if the issue is solved, thanks!

[fix_hidh_connection_bug_20230130.zip](https://github.com/espressif/esp-idf/files/10531995/fix_hidh_connection_bug_20230130.zip)

benryves, 2023-02-03:
Hi @boblane1,

Thank you very much for putting that together, unfortunately after applying those patches I still get stuck in an endless `ERR_SDP` loop whenever I try to reopen the device that failed to connect previously. I do get an error message about an invalid device handle of 255 (possibly set to -1 somewhere?) though, which I've not seen before...

```
W (5765) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (6255) MAIN: Running: 0:04, <NO VIDEO>
W (6775) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
E (6925) BT_BTM: tBTM_SEC_DEV:0x3ffd1b10 rs_disc_pending=0

W (7115) BT_SDP: process_service_attr_rsp

E (7175) BT_HIDH: OPEN ERROR: ERR_SDP
E (7175) BT: HIDH open failed (-1)
I (7175) BT: trying to reconnect
I (7175) BT: reconnecting 00:22:d7:6a:4b:bc...
W (7235) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (7235) BT_APPL: bta_hh_co_close: dev_handle = 255, app_id = 0
W (7235) BT_APPL: bta_hh_co_close: Oops, dev_handle (255) is invalid...
W (7245) BT_APPL: bta_dm_act no entry for connected service cbs
E (7245) BT_BTC: Error: cannot find device with handle 255
E (7255) BT_HIDH: CLOSE ERROR: NO_CONNECTION
W (7255) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16
I (7265) BT: HIDH close 00:22:d7:6a:4b:bc: 
I (7265) MAIN: Running: 0:05, <NO VIDEO>
W (7435) BT_SDP: process_service_attr_rsp

I (8175) BT: re-opening 00:22:d7:6a:4b:bc
I (8275) MAIN: Running: 0:06, <NO VIDEO>
I (9275) MAIN: Running: 0:07, <NO VIDEO>
I (10275) MAIN: Running: 0:08, <NO VIDEO>
I (11275) MAIN: Running: 0:09, <NO VIDEO>
I (12275) MAIN: Running: 0:10, <NO VIDEO>
I (13275) MAIN: Running: 0:11, <NO VIDEO>
W (13285) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x43

W (13295) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (13295) BT_HIDH: OPEN ERROR: ERR_SDP
E (13295) BT: HIDH open failed (-1)
I (13305) BT: trying to reconnect
I (13305) BT: reconnecting 00:22:d7:6a:4b:bc...
I (14275) MAIN: Running: 0:12, <NO VIDEO>
I (14305) BT: re-opening 00:22:d7:6a:4b:bc
I (15275) MAIN: Running: 0:13, <NO VIDEO>
I (16275) MAIN: Running: 0:14, <NO VIDEO>
I (17275) MAIN: Running: 0:15, <NO VIDEO>
I (18275) MAIN: Running: 0:16, <NO VIDEO>
I (19275) MAIN: Running: 0:17, <NO VIDEO>
W (19415) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

W (19415) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (19415) BT_HIDH: OPEN ERROR: ERR_SDP
E (19415) BT: HIDH open failed (-1)
I (19425) BT: trying to reconnect
I (19425) BT: reconnecting 00:22:d7:6a:4b:bc...
I (20275) MAIN: Running: 0:18, <NO VIDEO>
I (20435) BT: re-opening 00:22:d7:6a:4b:bc
I (21275) MAIN: Running: 0:19, <NO VIDEO>
I (22275) MAIN: Running: 0:20, <NO VIDEO>
I (23275) MAIN: Running: 0:21, <NO VIDEO>
I (24275) MAIN: Running: 0:22, <NO VIDEO>
I (25275) MAIN: Running: 0:23, <NO VIDEO>
W (25555) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x43

W (25555) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (25555) BT_HIDH: OPEN ERROR: ERR_SDP
E (25565) BT: HIDH open failed (-1)
I (25565) BT: trying to reconnect
I (25565) BT: reconnecting 00:22:d7:6a:4b:bc...
I (26275) MAIN: Running: 0:24, <NO VIDEO>
I (26575) BT: re-opening 00:22:d7:6a:4b:bc
I (27275) MAIN: Running: 0:25, <NO VIDEO>
I (28275) MAIN: Running: 0:26, <NO VIDEO>
I (29275) MAIN: Running: 0:27, <NO VIDEO>
I (30275) MAIN: Running: 0:28, <NO VIDEO>
I (31275) MAIN: Running: 0:29, <NO VIDEO>
W (31685) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

W (31685) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (31685) BT_HIDH: OPEN ERROR: ERR_SDP
E (31695) BT: HIDH open failed (-1)
I (31695) BT: trying to reconnect
I (31695) BT: reconnecting 00:22:d7:6a:4b:bc...
I (32275) MAIN: Running: 0:30, <NO VIDEO>
I (32705) BT: re-opening 00:22:d7:6a:4b:bc
I (33275) MAIN: Running: 0:31, <NO VIDEO>
I (34275) MAIN: Running: 0:32, <NO VIDEO>
I (35275) MAIN: Running: 0:33, <NO VIDEO>
I (36275) MAIN: Running: 0:34, <NO VIDEO>
I (37275) MAIN: Running: 0:35, <NO VIDEO>
W (37825) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x43

W (37825) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (37825) BT_HIDH: OPEN ERROR: ERR_SDP
E (37835) BT: HIDH open failed (-1)
I (37835) BT: trying to reconnect
I (37835) BT: reconnecting 00:22:d7:6a:4b:bc...
I (38275) MAIN: Running: 0:36, <NO VIDEO>
I (38845) BT: re-opening 00:22:d7:6a:4b:bc
I (39275) MAIN: Running: 0:37, <NO VIDEO>
I (40275) MAIN: Running: 0:38, <NO VIDEO>
I (41275) MAIN: Running: 0:39, <NO VIDEO>
I (42275) MAIN: Running: 0:40, <NO VIDEO>
I (43275) MAIN: Running: 0:41, <NO VIDEO>
W (43955) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

W (43955) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (43955) BT_HIDH: OPEN ERROR: ERR_SDP
E (43965) BT: HIDH open failed (-1)
I (43965) BT: trying to reconnect
I (43965) BT: reconnecting 00:22:d7:6a:4b:bc...
I (44275) MAIN: Running: 0:42, <NO VIDEO>
I (44975) BT: re-opening 00:22:d7:6a:4b:bc
```

(In this case the "trying to reconnect" message appears when the connection fails, "reconnecting 00:22:d7:6a:4b:bc..." appears when the pending connection is successfully queued for later retrieval by a task, "re-opening 00:22:d7:6a:4b:bc" appears in the reconnection task when `esp_hidh_dev_open` returns successfully).

boblane1, 2023-02-06:
Hi @benryves,

In the previous patch, when the open operation failed, it comes with both `ESP_HIDH_OPEN_EVENT` and `ESP_HIDH_CLOSE_EVENT` events. Please initiate the reconnect operation in the `ESP_HIDH_CLOSE_EVENT` event which is cased by open failure or just delay some time to recover the stack. Please try again to see if the issue persists.

benryves, 2023-02-06:
Hi @boblane1

Thanks for the feedback, previously I was only using the patch you supplied in [fix_hidh_connection_bug_20230130.zip](https://github.com/espressif/esp-idf/files/10532999/fix_hidh_connection_bug_20230130.zip), I reset to a clean version of 5.0 and then applied [fix_hidh_connection_bug.zip](https://github.com/espressif/esp-idf/files/10381323/fix_hidh_connection_bug.zip) and then the later patch. After applying both patches I now see the `HIDH_CLOSE` event for the first (failed) connection attempt, but subsequent failures don't appear to trigger this event.

When the `HIDH_OPEN` event fails I store the BDA in a queue and start a task that waits for one second, then pulls items from the queue and tries to reopen them with a one second delay between them. I've made my logging a bit clearer to show that it's queueing up connection attempts (rather than immediately reopening the connection) as well as extended the initial delay to three seconds. (As I only get the `HIDH_CLOSE` event for the first failed attempt I suppose handling this event instead of a failed `HIDH_OPEN` would prevent my code from going into an infinite loop, it would only retry once).

```
W (1921) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (2881) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
E (3071) BT_BTM: tBTM_SEC_DEV:0x3ffd297c rs_disc_pending=0

W (3241) BT_SDP: process_service_attr_rsp

E (3271) BT_HIDH: OPEN ERROR: ERR_SDP
E (3281) BT: HIDH open failed (-1)
I (3281) BT: will try to reconnect
I (3281) BT: queued 00:21:bd:07:b6:41 for reconnection...
W (3301) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (3301) BT_APPL: bta_hh_co_close: dev_handle = 255, app_id = 0
W (3301) BT_APPL: bta_hh_co_close: Oops, dev_handle (255) is invalid...
W (3311) BT_APPL: bta_dm_act no entry for connected service cbs
E (3311) BT_BTC: Error: cannot find device with handle 255
E (3321) BT_HIDH: CLOSE ERROR: NO_CONNECTION
I (3321) BT: HIDH close 00:21:bd:07:b6:41: 
W (3331) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16
W (3621) BT_SDP: process_service_attr_rsp

I (3911) MAIN: Running: 0:00, <NO VIDEO>
I (4911) MAIN: Running: 0:01, <NO VIDEO>
I (5911) MAIN: Running: 0:02, <NO VIDEO>
I (6281) BT: re-opening 00:21:bd:07:b6:41
I (6911) MAIN: Running: 0:03, <NO VIDEO>
I (7911) MAIN: Running: 0:04, <NO VIDEO>
I (8911) MAIN: Running: 0:05, <NO VIDEO>
I (9911) MAIN: Running: 0:06, <NO VIDEO>
I (10911) MAIN: Running: 0:07, <NO VIDEO>
W (11391) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

W (11391) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (11391) BT_HIDH: OPEN ERROR: ERR_SDP
E (11391) BT: HIDH open failed (-1)
I (11401) BT: will try to reconnect
I (11401) BT: queued 00:21:bd:07:b6:41 for reconnection...
I (11911) MAIN: Running: 0:08, <NO VIDEO>
I (12911) MAIN: Running: 0:09, <NO VIDEO>
I (13911) MAIN: Running: 0:10, <NO VIDEO>
I (14411) BT: re-opening 00:21:bd:07:b6:41
I (14911) MAIN: Running: 0:11, <NO VIDEO>
I (15911) MAIN: Running: 0:12, <NO VIDEO>
I (16911) MAIN: Running: 0:13, <NO VIDEO>
I (17911) MAIN: Running: 0:14, <NO VIDEO>
I (18911) MAIN: Running: 0:15, <NO VIDEO>
W (19521) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x43

W (19521) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (19531) BT_HIDH: OPEN ERROR: ERR_SDP
E (19531) BT: HIDH open failed (-1)
I (19531) BT: will try to reconnect
I (19541) BT: queued 00:21:bd:07:b6:41 for reconnection...
I (19911) MAIN: Running: 0:16, <NO VIDEO>
I (20911) MAIN: Running: 0:17, <NO VIDEO>
I (21911) MAIN: Running: 0:18, <NO VIDEO>
I (22541) BT: re-opening 00:21:bd:07:b6:41
I (22911) MAIN: Running: 0:19, <NO VIDEO>
I (23911) MAIN: Running: 0:20, <NO VIDEO>
I (24911) MAIN: Running: 0:21, <NO VIDEO>
I (25911) MAIN: Running: 0:22, <NO VIDEO>
I (26911) MAIN: Running: 0:23, <NO VIDEO>
W (27651) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

W (27651) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
E (27651) BT_HIDH: OPEN ERROR: ERR_SDP
E (27651) BT: HIDH open failed (-1)
I (27661) BT: will try to reconnect
I (27661) BT: queued 00:21:bd:07:b6:41 for reconnection...
I (27911) MAIN: Running: 0:24, <NO VIDEO>
I (28911) MAIN: Running: 0:25, <NO VIDEO>
```

Would it be useful if I was to try to put together the smallest program I can that demonstrates the issue? (You'd need at least two Wii remotes and a push button to perform the initial sync, then once they're paired you could reset the ESP32 until they both tried to reconnect at the same time and triggered the issue).

boblane1, 2023-02-06:
Hi @benryves,

>W (11391) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x42

>W (11391) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4

`st 0x4` means the SDP connection is timeout. It seems that the Wii can not response the reconnection of ESP32(HID Host) when the previous connection is failed.

benryves, 2023-02-06:
Thank you for the confirmation, @boblane1!

My previous solution was to de-init and re-init the Bluetooth stack on error, this cleanly disconnects the Wii remotes and allows them to be reconnected by pressing a button on them to wake them up (when they encounter an error otherwise they get stuck in an unresponsive state unless you hold the power button down or pull a battery out). I think I may stick to that solution, as it's the least confusing for the user.

closed 2023-03-13 by espressif-bot