espressif/esp-idf #10506: Unable to transmit any data over Wi-Fi after heavy traffic and poor internet quality

DimaSes99, 2023-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-MINI-1 custom board

### Power Supply used.

Battery

### What is the expected behavior?

The FW does not lose connection with internet and access point after heavy traffic and poor internet quality.

### What is the actual behavior?

The FW works normal for some time, sends and receives packets to AWS MQTT broker. After some time, under heavy internet traffic (a lot of messages been sent by a device) and with poor internet quality the device loses connection with MQTT broker (KeepAlive timeout) and then the device is unable to reconnect with the broker because of DNS errors.
Under this condition, the device is unable to ping any internet resource (for example, espressif.com) and, also, the device cannot ping access point itself by its IP address. The device remains connected with the access point under this condition.
There are two ways to recover the device from this state:
1. Perform reset
2. Disconnect and connect to access point

### Steps to reproduce.

1. Use smartphone with 2G mode only as a hot-spot
2. Power up the device and wait for Wi-Fi and broker connection
3. Verify that everything is working - the device is able to send and receive messages
4. Verify that the device is able to ping espressif.com and the hot-spot using icmp-echo
5. Put the device in a state where it starts sending a lot of data to MQTT broker
6. Wait for some time until the device loses connection
7. The device is unable to resolve MQTT broker hostname, is unable to ping espressif.com and is unable to ping the hot-spot by its IP address

### Debug Logs.

```plain
The log we are constantly seeing when a device is under the condition:

E<115820> esp-tls: couldn't get hostname for : <our_broker_url>: getaddrinfo() returns 202, addrinfo =0x0
```

### More Information.

We have a bunch of devices and all of them have this problem. Sometimes it can work for hours under heavy internet traffic, and sometimes it can lose connection after a few minutes. It is highly depended on the connection speed/quality. 
Additional information:
* MQTT library is used: [link](https://github.com/FreeRTOS/coreMQTT)
* Application constantly performs BLE advertising with ~100ms interval. 
* SDK config is attached

What we tried to resolve an issue:
1. Disable BLE, completely disable all Wi-Fi power saving - the problem remains, but it takes more time for the FW to go to unrecoverable disconnect
2. Increase number of Wi-Fi RX and TX buffers - the problem looks resolved, but there are another problems in our application because of the lack of memory

The issue looks similar to [#8953](https://github.com/espressif/esp-idf/issues/8953)

[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/10368629/sdkconfig.defaults.txt)

nishanth-radja, 2023-01-09:
@DimaSes99 can you pls attach the console logs?

DimaSes99, 2023-01-09:
> 

There are a lot of application logs but here is is.
At lines 224-225 there is ping statistic after successful connect.
At the line 677 there is connection lose
At lines 716-726 the FW attempts to reconnect and then tries
[unrecoverable_disconnect_2.txt](https://github.com/espressif/esp-idf/files/10372542/unrecoverable_disconnect_2.txt)
 to ping hot-spot again

Espressif-liuuuu, 2023-01-13:
hi @DimaSes99 , here is recommended configuration for coexistence of Wi-Fi and BLE and pls give a try
```
CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE=y
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=6
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=12
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=12
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=6
```
```
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=12000
CONFIG_LWIP_TCP_WND_DEFAULT=12000
```

txf-, 2023-03-17:
I have also hit this bug. And I am also using coreMQTT (with the coreMQTT-agent). 

I do not have BLE enabled, however, but I am transmitting at a rate of 10hz.

landaming, 2024-01-04:
I had this problem too

KonssnoK, 2024-02-14:
this recalls quite closely issues that we are experiencing on the field with 25000+ devices.

What i can add:
- it happens both on LTE and Wifi (we have 2 netif interfaces)
- it is not the netif interface that stops working ( the drivers receive data when the timeouts occur )
- sometimes the downtime is for few seconds/minutes
- sometimes the downtime is persistent until a reboot of the device is triggered

Espressif-liuuuu, 2024-05-15:
Hi @txf-  @KonssnoK  @landaming , do we still have the issue? Feel free to open a new ticket and share more information if it does.

AxelLin, 2024-06-06:
@Espressif-liuuuu 
This issue is changed to "Reviewing" status, I'm wondering if you found the root cause? (What was fixed?)

Espressif-liuuuu, 2024-06-06:
> @Espressif-liuuuu This issue is changed to "Reviewing" status, I'm wondering if you found the root cause? (What was fixed?)

Hi, its awaiting response, no more detail for later comments.   
Hi @DimaSes99 , for the original issue, did it work by the attached configuration?

txf-, 2024-06-06:
> Hi @txf- @landaming , do we still have the issue? Feel free to open a new ticket and share more information if it does.

Don't keep the issue open on my account. Last time I looked at this was in Feb. at that time I still had the issue.  However I'm not currently working on this ,  so I can't investigate this again.

closed 2024-07-17 by Sherry616