espressif/esp-idf #10507: GPIO does not work on ESP32-S3 WROOM-1 (N8R8)

nyh-workshop, 2023-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello Espressif,

I have a  [Cytron Maker Feather AIOT S3](https://my.cytron.io/p-v-maker-feather-aiot-s3-simplifying-aiot-with-esp32?r=1)  that uses the ESP32-S3 WROOM-1 (N8R8). It has full support for Arduino and the standard "Blink" examples all work with most of the GPIO.

However, when I'm using the ESP-IDF 5.0 (Installed through the VScode Extension) and compile it (without any errors), and then upload it into the same module, none of the GPIO gets toggled, or even switched on for a start.

There are no errors in the console - everything else look fine there. Here's the code snippet and the console:
```
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_log.h"

#include "driver/gpio.h"

void app_main(void)
{
    printf("Hello World!\n");

    gpio_set_direction(GPIO_NUM_40, GPIO_MODE_OUTPUT);
    
    gpio_set_level(GPIO_NUM_40, 0);

    while (1)
    {
        gpio_set_level(GPIO_NUM_40, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        gpio_set_level(GPIO_NUM_40, 1);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```

```
I (24) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 23:53:41
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00177000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=089cch ( 35276) map
I (117) esp_image: segment 1: paddr=000189f4 vaddr=3fc91f00 size=03154h ( 12628) load
I (122) esp_image: segment 2: paddr=0001bb50 vaddr=40374000 size=044c8h ( 17608) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19c08h (105480) map
I (153) esp_image: segment 4: paddr=00039c30 vaddr=403784c8 size=09944h ( 39236) load
I (161) esp_image: segment 5: paddr=0004357c vaddr=50000000 size=00010h (    16) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (181) octal_psram: vendor id    : 0x0d (AP)
I (182) octal_psram: dev id       : 0x02 (generation 3)
I (182) octal_psram: density      : 0x03 (64 Mbit)
I (187) octal_psram: good-die     : 0x01 (Pass)
I (192) octal_psram: Latency      : 0x01 (Fixed)
I (197) octal_psram: VCC          : 0x01 (3V)
I (202) octal_psram: SRF          : 0x01 (Fast Refresh)
I (208) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (214) octal_psram: BurstLen     : 0x01 (32 Byte)
I (219) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (226) octal_psram: DriveStrength: 0x00 (1/1)
I (231) esp_psram: Found 8MB PSRAM device
I (235) esp_psram: Speed: 40MHz
I (239) cpu_start: Pro cpu up.
I (243) cpu_start: Starting app cpu, entry point is 0x40375304
0x40375304: call_start_cpu1 at C:/Users/yongh/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (980) esp_psram: SPI SRAM memory test OK
I (989) cpu_start: Pro cpu start user code
I (989) cpu_start: cpu freq: 240000000 Hz
I (990) cpu_start: Application information:
I (992) cpu_start: Project name:     esp32-ulp-hello
I (998) cpu_start: App version:      1
I (1002) cpu_start: Compile time:     Jan  8 2023 23:52:31
I (1008) cpu_start: ELF file SHA256:  0413444e8b0f1479...
I (1015) cpu_start: ESP-IDF:          v5.0
I (1019) heap_init: Initializing. RAM available for dynamic allocation:
I (1027) heap_init: At 3FC95AC8 len 00053C48 (335 KiB): D/IRAM
I (1033) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1040) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1046) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1053) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1061) spi_flash: detected chip: generic
I (1065) spi_flash: flash io: qio
W (1069) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the 
binary image header.
I (1083) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1103) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
Hello World!
```

I also have the SDKconfig file attached there. I'm suspecting that there are some settings that are not correct for this module. Even setting the SPI flash to 8MB (as this module has a 8MB flash) doesn't work there.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10368827/sdkconfig.txt)

Help is requested to examine if there are any configurations that are wrong that prevents any of the libraries from working. :D

PickaxeHit, 2023-01-08:
try to add `esp_rom_gpio_pad_select_gpio(GPIO_NUM_x)` before `gpio_set_direction()`.

nyh-workshop, 2023-01-09:
Hi @PickaxeHit , thanks for the help. Added this function before the `gpio_set_direction`, however it still doesn't work.

The blink code is based on most the examples available online and it works on the original ESP32.

0xjakob, 2023-01-09:
@nyh-workshop Are you trying to use the RGB-LED on that board?

nyh-workshop, 2023-01-09:
Hi @0xjakob , I'm using the normal LED on the board itself. The addressable LED is at GPIO46.

nopnop2002, 2023-01-09:
@nyh-workshop 

Try this.
```
gpio_reset_pin(GPIO_NUM_40);
gpio_set_direction(GPIO_NUM_40, GPIO_MODE_OUTPUT);
```

nyh-workshop, 2023-01-09:
Hi @nopnop2002 , thanks for the assistance.

This is the output after using the provided log at the GPIO40:
```
I (1113) MAIN: GPIO_NUM_40=40
I (1113) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
```

nyh-workshop, 2023-01-09:
I have also manually added the Enable W1TS on it just to see if it's set, it looks like it's just zero:
(Some of the getting the GPIO registers are from this [blog](https://blog.csdn.net/Marchtwentytwo/article/details/117753770))
```
    gpio_reset_pin(GPIO_NUM_14);

    esp_rom_gpio_pad_select_gpio(GPIO_NUM_14);

    // gpio_set_direction(GPIO_NUM_40, GPIO_MODE_OUTPUT);
    
    // gpio_set_level(GPIO_NUM_40, 0);

    gpio_config_t cfg;
    cfg.pin_bit_mask = 1ULL << GPIO_NUM_14;
    cfg.mode = GPIO_MODE_OUTPUT;
    cfg.pull_up_en = GPIO_PULLUP_DISABLE;
    cfg.pull_down_en = GPIO_PULLDOWN_DISABLE;
    cfg.intr_type = GPIO_INTR_DISABLE;
    gpio_config(&cfg);

    gpio_dev_t *hw = &GPIO;
    hw->enable_w1ts = (1 << GPIO_NUM_14);
    gpio_set_level(GPIO_NUM_14, 1);
    ESP_LOGI("MAIN", "ENABLE_W1TS=%d", hw->enable_w1ts);
    ESP_LOGI("MAIN", "W1TS=%d", hw->out_w1ts);
    ESP_LOGI("MAIN", "GPIO_NUM_14=%d",GPIO_NUM_14); 
```

Output:
```
I (1113) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1113) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1123) MAIN: ENABLE_W1TS=0
I (1133) MAIN: W1TS=0
I (1133) MAIN: GPIO_NUM_14=14
```

I suspect that the W1TS and W1TC must be enabled before using the `gpio_set_level()`?

nopnop2002, 2023-01-09:
@nyh-workshop 

```
    gpio_reset_pin(GPIO_NUM_14);

    esp_rom_gpio_pad_select_gpio(GPIO_NUM_14);
```

they have the same function.

nyh-workshop, 2023-01-10:
Hi @nopnop2002 , thanks for the assistance. 

I have examined the board and directly probed the pins on the ESP32-S3-WROOM1 - the pins are working as usual.

On the board supporting the ESP32 module, it has diagnostic LEDs connected to a few GPIOs and driven by a small MOSFET each. The board has an option to enable or disable the diagnostic LEDs - A high to the GPIO11 enables these MOSFET immediately.

For those who worked with Cytron's Maker Feather AIOT S3 and ESP-IDF and need to use the diagnostic LEDs - the GPIO11 must be at high first.

closed 2023-01-10 by nyh-workshop

bigfootXadite, 2023-04-04:
> try to add `esp_rom_gpio_pad_select_gpio(GPIO_NUM_x)` before `gpio_set_direction()`.

This worked for me!

xuanchibkd14, 2023-05-30:
it not working in gpio42, my code is:
```
gpio_reset_pin(XPT2046_IRQ);
gpio_config_t irq_config = {
.pin_bit_mask = BIT64(XPT2046_IRQ),
.mode = GPIO_MODE_INPUT,
.pull_up_en = GPIO_PULLUP_DISABLE,
.pull_down_en = GPIO_PULLDOWN_DISABLE,
.intr_type = GPIO_INTR_DISABLE,
};
esp_err_t ret = gpio_config(&irq_config);
```
i can't get input signal from XPT2046_IRQ, it define by gpio 42