espressif/esp-idf #10508: Using GPIO Lookup Macros with ESP-IDF V5

nopnop2002, 2023-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used the GPIO Lookup Macros below:
ADC1_GPIO36_CHANNEL

But I get the following warning:

```
In file included from ../main/main.c:12:
/home/nop/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~

```

# IDF Version:
```
$ idf.py --version
ESP-IDF v5.0
```

# Source code:

```
#include <stdio.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "soc/adc_channel.h"
#include "driver/adc.h"

#define TAG "MAIN"

void app_main()
{
        ESP_LOGI(TAG, "ADC1_GPIO32_CHANNEL=%d", ADC1_GPIO32_CHANNEL);
}
```

Icarus113, 2023-01-09:
Hi @nopnop2002 , the ADC related drivers under `driver/adc.h` include path are migrated to `esp_adc/adc_oneshot.h` and `esp_adc/adc_continuous.h`. Detailed migration guides are [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc). You can enable [CONFIG_ADC_SUPPRESS_DEPRECATE_WARN](https://github.com/espressif/esp-idf/blob/master/components/driver/Kconfig#L15) to disable the deprecation warning, if you still use the legacy driver.

nopnop2002, 2023-01-09:
@Icarus113 
Thanks for your comment.
I have read [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#adc) but could not find any mention of ADC1_GPIOxx_CHANNEL.

This document describes it this way.
```
The ADC oneshot mode driver has been redesigned.
The new driver is in esp_adc component and the include path is esp_adc/adc_oneshot.h.
The legacy driver is still available in the previous include path driver/adc.h.
The ADC continuous mode driver has been moved from driver component to esp_adc component.
The include path has been changed from driver/adc.h to esp_adc/adc_continuous.h.
```

But ADC1_GPIOxx_CHANNEL is not defined in "esp_adc/adc_continuous.h" nor "esp_adc/adc_oneshot.h".
When using ADC1_GPIO32_CHANNEL, it looks like I still need to use the legacy driver.

Icarus113, 2023-01-09:
I see, thanks for the report. Since `ADC1_GPIOxx_CHANNEL` isn't deprecated, we will make them work with both the legacy driver and the new driver. @nopnop2002 
There will be a commit for this later.

nopnop2002, 2023-01-09:
@Icarus113 

This document may also need revision.
[image]

Icarus113, 2023-01-12:
The doc you posted seems to be a version earlier than v5.0. So it should be correct. The fix has been merged, and will be introduced in next v5.0.x release as well.
@nopnop2002

nopnop2002, 2023-01-12:
@Icarus113 

See bottom left of my posted image.

It is written as v:latest.

Icarus113, 2023-01-12:
Maybe it's because your posted is in the `readthedocs` ? Sorry I'm not quite sure of the one you posted. But I'm sure the `adc.html` is the old one (prior to v5.0).
I think you can refer to [This](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/index.html)

@nopnop2002

nopnop2002, 2023-01-12:
>Maybe it's because what your posted is in the readthedocs?

Yes.
my post is readthedocs.

https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html#gpio-lookup-macros

closed 2023-01-13 by espressif-bot