espressif/esp-idf #10510: [esp32s3]wakeup cause is always 0

robbietree8, 2023-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

- esp-idf version: v4.4.3-210-g932b6699ac
- example: https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/deep_sleep_example_main.c
- board: esp32-s3-wroom-1

The wakeup cause is always 0 when wakeup from deep sleep from ext0 or from timer.

robbietree8, 2023-01-17:
I switched idf branch to release/v5.0, the issue does not existed.

Because I need to use esp-adf, the latest version I can use for esp-idf is release/v4.4. 

So I wonder if there is any fix can be back ported to release/v4.4?

ESP-Marius, 2023-04-27:
@robbietree8 are you still seeing this issue?

I'm unable to reproduce this. How are you connected to the ESP32-S3? Would you mind sharing the log?

robbietree8, 2023-04-27:
@ESP-Marius I can not reproduce it anymore, and I don't know how to explain it...

We can close this issue now.

closed 2023-04-27 by robbietree8