espressif/esp-idf #10511: Only one STA can join the SoftAP otherwise "max connection, deauth!"

wuyuanyi135, 2023-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2669-gd152d007c6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

I created a softAP with the following code (SSID and password omitted)
```
wifi_config_t wifi_config = {};
wifi_config.ap.max_connection = 15;
wifi_config.ap.authmode = WIFI_AUTH_WPA2_PSK;
wifi_config.ap.ssid_hidden = 1;
```
Then 
```
esp_wifi_set_config(WIFI_IF_AP, &wifi_config);
esp_wifi_start();
```

I would like see multiple STA could join this AP

### What is the actual behavior?

After first STA joins, if the second one tried to join, the SoftAP gives 
```
W (82022) wifi: [ieee80211_hostap.c,1339] max connection, deauth!
```

### Steps to reproduce.

1. Flash the SoftAP code
2. let multiple client join the AP

wuyuanyi135, 2023-01-09:
I bisected the issue. In `Component config → Wi-Fi` if ` Maximum espnow encrypt peers number` is set to `N` then the maximum number of client is implicitly limited to `17 - N`. For example, if `N=17`, no sta can associate to the AP. if `N=16`, only one sta can join the AP.

Is this a documentation issue or design flaw? I believe the inherent relation between ESPNOW peer number and STA limit is not documented yet.

Xiehanxin, 2023-01-11:
[image]
maybe your max_connection is too large.

wuyuanyi135, 2023-01-11:
@Xiehanxin No. In `master` branch the maximum allowed connection for ESP32S3 is 15. I also tried set it to a lower number like 5, but it behaved the same. `Maximum espnow encrypt peers` still affects the number of clients.

mhdong, 2024-02-28:
Hi @wuyuanyi135 
I'm very sorry for not responding in a timely manner. The document has been updated.
https://github.com/espressif/esp-idf/blob/5f4249357372f209fdd57288265741aaba21a2b1/docs/en/api-guides/wifi.rst#L1347

Sherry616, 2024-04-09:
Thanks for reporting, we will close this issue. Feel free to reopen if have more updates.

closed 2024-04-09 by Sherry616