espressif/esp-idf #10515: Add mbedTLS MBEDTLS_ERR_X509_CERT_VERIFY_FAILED return code

danewalton, 2023-01-10:
### Is your feature request related to a problem?

Currently, it seems the ESP IDF brings some error codes from the lower TLS layers but not all. Adding the above code would be very useful.

### Describe the solution you'd like.

Add the code please.

laukik-hase, 2023-02-14:
Hello, @danewalton!

Sorry for not getting back to you sooner.

While using `esp-tls` APIs (e.g. [`esp_tls_conn_new_sync()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#_CPPv421esp_tls_conn_new_syncPKciiPK13esp_tls_cfg_tP9esp_tls_t)), you can get the Mbed TLS error codes with the API [`esp_tls_get_and_clear_last_error()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#_CPPv432esp_tls_get_and_clear_last_error22esp_tls_error_handle_tPiPi) if the connection establishment fails. I've attached a usage example below.

```c
static void example_get_tls_error(esp_tls_t *tls)
{
    esp_tls_error_handle_t error_handle = {0};
    int esp_tls_code = 0, esp_tls_flags = 0;

    esp_err_t ret = esp_tls_get_error_handle(tls, &error_handle);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to get esp_tls error handle!");
        return;
    }

    ret = esp_tls_get_and_clear_last_error(error_handle, &esp_tls_code, &esp_tls_flags);
    ESP_LOGE(TAG, "Reason: %d - %s", ret, esp_err_to_name(ret));
    ESP_LOGE(TAG, "ESP_TLS: Error code - 0x%x", (unsigned int)esp_tls_code);
    ESP_LOGE(TAG, "ESP-TLS: Error flags - %d", esp_tls_flags);
}
```

mahavirj, 2023-02-27:
Closing as per the above comment. Please drop a comment if you still have any questions..

closed 2023-02-27 by mahavirj

danewalton, 2023-03-01:
@Laukik Hase just since this was closed.

laukik-hase, 2023-03-02:
Hello, @danewalton!

You can try using the API [esp_transport_get_error_handle()](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/include/esp_transport.h#L326-L338) when using `tcp_transport` APIs.