espressif/esp-idf #10516: Problem with SPI Flash API on Master Branch for esp32-c3

bmkuter, 2023-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0-rc1-4-g919b3e621e

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-c3 mini 1

### Power Supply used.

USB

### What is the expected behavior?

I am looking to initialise the default chip SPI flash via `esp_flash_default_chip(NULL)` to later read and write flash memory via `esp_flash_write()` and `esp_flash_read()`. I am using NULL here to specify the esp_flash_default_chip per [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_flash/index.html?highlight=esp_flash_init#_CPPv414esp_flash_initP11esp_flash_t). 

### What is the actual behavior?

Currently the system is returning error 0x102, `ESP_ERR_INVALID_ARG`. The code is:

```
    ...
    uint32_t target_address = 0x1000;
    uint32_t wipe_size = sizeof(read_buffer);
    memset(read_buffer, 0, wipe_size);
    ESP_LOGI(tag, "Start: %c", read_buffer[0]);

    ESP_LOGI(tag, "esp_flash_init:");
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_init(NULL);
    taskENABLE_INTERRUPTS();
    if (flash_error) 
    {
        ESP_LOGE(tag, "Error esp_flash_init! Error: %d", flash_error);
        while(1)
        {
            vTaskDelay(1000/portTICK_PERIOD_MS);
        }
    }
    ...
```

### Steps to reproduce.

Full code:

```
#include <string.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/semphr.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "nvs.h"
#include "esp_netif.h"
#include "esp_flash.h"

char read_buffer[4096];
char message_buffer[4096];

void app_main(void)
{
    char tag[] = "Main";
    esp_err_t flash_error; 

    // Note from Docs: 
    // New functions with prefix esp_flash accept an additional esp_flash_t* parameter. You can simply set it to NULL. This will make the function to run the main flash (esp_flash_default_chip).
    // https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/migration-guides/release-5.x/storage.html

    uint32_t target_address = 0x1000;
    uint32_t wipe_size = sizeof(read_buffer);
    memset(read_buffer, 0, wipe_size);
    ESP_LOGI(tag, "Start: %c", read_buffer[0]);

    ESP_LOGI(tag, "esp_flash_init:");
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_init(NULL);
    taskENABLE_INTERRUPTS();
    if (flash_error) 
    {
        ESP_LOGE(tag, "Error esp_flash_init! Error: %d", flash_error);
        while(1)
        {
            vTaskDelay(1000/portTICK_PERIOD_MS);
        }
    }

    ESP_LOGI(tag, "esp_flash_erase_region:");
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_erase_region(NULL, 0x1000, sizeof(message_buffer)); 
    taskENABLE_INTERRUPTS();
    if (flash_error) 
    {
        ESP_LOGE(tag, "Error esp_flash_erase_region! Error: %d", flash_error);
        while(1)
        {
            vTaskDelay(1000/portTICK_PERIOD_MS);
        }
    }

    memset(message_buffer, 'A', sizeof(message_buffer));
    ESP_LOGI(tag, "esp_flash_write:");
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_write(NULL, message_buffer, 0x1000, sizeof(message_buffer)); 
    taskENABLE_INTERRUPTS();
    if (flash_error) 
    {
        ESP_LOGE(tag, "Error esp_flash_write! Error: %d", flash_error);
        while(1)
        {
            vTaskDelay(1000/portTICK_PERIOD_MS);
        }
    }

    ESP_LOGI(tag, "esp_flash_read:");
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_read(NULL, read_buffer, 0x1000, sizeof(read_buffer)); 
    taskENABLE_INTERRUPTS();
    if (flash_error) 
    {
        ESP_LOGE(tag, "Error esp_flash_write! Error: %d", flash_error);
        while(1)
        {
            vTaskDelay(1000/portTICK_PERIOD_MS);
        }
    }

    ESP_LOGI(tag, "End: 0x%x", read_buffer[0]);
    while (1)
    {
        // ESP_LOGI(tag, "Hello World!");
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}

```

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x16e8
load:0x403cc710,len:0x96c
load:0x403ce710,len:0x2e2c
entry 0x403cc710
I (30) boot: ESP-IDF v5.0-rc1-4-g919b3e621e 2nd stage bootloader
I (30) boot: compile time 23:46:51
I (30) boot: chip revision: v0.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO        
I (43) boot.esp32c3: SPI Flash Size : 2MB        
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=081b0h ( 33200) map        
I (104) esp_image: segment 1: paddr=000181d8 vaddr=3fc8b400 size=014d0h (  5328) load      
I (108) esp_image: segment 2: paddr=000196b0 vaddr=40380000 size=06968h ( 26984) load      
I (121) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=16adch ( 92892) map       
I (151) esp_image: segment 4: paddr=00036b04 vaddr=40386968 size=04904h ( 18692) load      
I (155) esp_image: segment 5: paddr=0003b410 vaddr=50000010 size=00010h (    16) load      
I (160) boot: Loaded app from partition at offset 0x10000
I (163) boot: Disabling RNG early entropy source...
I (380) cpu_start: Pro cpu up.
I (388) cpu_start: Pro cpu start user code
I (388) cpu_start: cpu freq: 160000000 Hz
I (388) cpu_start: Application information:
I (391) cpu_start: Project name:     main
I (396) cpu_start: App version:      1
I (400) cpu_start: Compile time:     Jan 10 2023 00:11:42
I (406) cpu_start: ELF file SHA256:  2edb4e30b0d89006...
I (412) cpu_start: ESP-IDF:          v5.0-rc1-4-g919b3e621e
I (419) heap_init: Initializing. RAM available for dynamic allocation:
I (426) heap_init: At 3FC8F7B0 len 0004CF60 (307 KiB): DRAM
I (432) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (439) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (446) spi_flash: detected chip: generic
I (450) spi_flash: flash io: dio
W (454) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (467) cpu_start: Starting scheduler.
I (480) Main: Start: 
I (480) Main: esp_flash_init:
E (480) Main: Error esp_flash_init! Error: 258
```

igrr, 2023-01-10:
Other comments on the code you have posted:
```c++
    taskDISABLE_INTERRUPTS();
    flash_error = esp_flash_erase_region(NULL, 0x1000, sizeof(message_buffer)); 
    taskENABLE_INTERRUPTS();
```

1. You shouldn't disable interrupts around flash operations. Non-IRAM interrupts will be disabled automatically, and disabling all the interrupts like this may cause the Flash driver to not work correctly. (On ESP32-C3, currently the driver happens to work with interrupts disabled, but on multi-core chips the interrupts need to be enabled.)
2. Address 0x1000 is inside the bootloader binary, so by making this call you would overwrite the 2nd stage bootloader. If `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ABORTS` option is enabled (it is enabled by default), then the call to `esp_flash_erase_region(NULL, 0x1000, ...)` will abort due to an attempt to overwrite the bootloader.
3. `esp_flash_erase_region` size argument has to be multiple of SPI flash sector size, so the code looks relatively fragile: if someone changes the size of `message_buffer` to a value not divisible by SPI flash sector size, the code will start failing. It's better to round up the size of message buffer to a multiple of SPI flash sector size.

bmkuter, 2023-01-10:
Thank you for the insightful comments! It was in part the interrupts that were giving me issues. 

The end goal of the code is a manual programmer or sorts, so I need to be able to iterate over the entire flash space regardless, hence the flash_spi API. I am doing boundary calculations externally. When in the boot process is the main SPI initialized? Also is there a way to disable `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ABORTS` during runtime? Currently I have `CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED=y` but would love a more dynamic solution to this. 

Also do you happen to have a good reference for the bootloader, 2nd bootloader, application, etc address breakdown? I am looking at fig [3-1 in the TRM](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf) but can't correlate that to actual flash address. 

Thanks again!

igrr, 2023-01-10:
>  Also is there a way to disable CONFIG_SPI_FLASH_DANGEROUS_WRITE_ABORTS during runtime? Currently I've compiled my project with CONFIG_SPI_FLASH_DANGEROUS_WRITE_ALLOWED=y but would love a more dynamic solution to this.

You can do this by modifying the esp_flash_os_functions_t structure used by the flash chip.
Something along the lines of:
```c++
/* this is your custom callback function which should return ESP_OK if the region is writable */
static IRAM_ATTR esp_err_t custom_check_region_protected(void* arg, size_t start_addr, size_t size);

static esp_flash_os_functions_t custom_os_funcs;

/* fill the custom OS functions structure with the default values */
custom_os_funcs = *esp_flash_default_chip->os_func;
/* modify the region_protected pointer */
custom_os_funcs.region_protected = &custom_check_region_protected;
/* install the new OS functions structure into the esp_flash_t instance */
esp_flash_default_chip->os_func = &custom_os_funcs;
```

> Also do you happen to have a good reference for the bootloader, 2nd bootloader, application, etc address breakdown?

The 1st stage bootloader (a.k.a. ROM bootloader) is in the on-chip ROM. As it is ROM, it's not modifiable. It is not mapped into Flash in any way.

The 2nd stage bootloader is in Flash. The offset is 0x1000 for ESP32 and 0x0 for other chips ([bootloader docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/bootloader.html)).

The location of the application in Flash is determined by the [partition table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/partition-tables.html). Typically, with the default partition table, the application will be at offset 0x10000. This is fully customizable, though.

The addresses you see in Figure 3-1 in the TRM are the addresses in the CPU address space. Physical addresses in Flash can be mapped into the CPU address space via the Flash MMU. For example, the 2nd stage bootloader will map the code and read only data segments from somewhere in Flash to somewhere in the CPU address space.

(If this sounds confusing, perhaps you could to explain what you intend to achieve in your application? For most things, IDF provides a higher level API or approach which doesn't require caring about such low level details, and which is typically easier to understand.)