espressif/esp-idf #10517: mqtt/ssl_ds/configure_esp_secure_cert.py excution error in ESP32S3

Jinjungdong, 2023-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I run following procedure with "https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_ds/README.md"
I want to expect the configuration of DS. but some error is happend during configure the DS peripheral. 

### What is the actual behavior?

The error message is as below. the excute stops in the last line. Import error is happend.

PS D:\Project\AVR\VSC\esp_mqtt_ssl_tls_ds>python .\configure_esp_secure_cert.py --port COM7 --device-cert .\main\client.crt --private-key .\main\client.key --target_chip esp32s3 --configure_ds --skip_flash
Traceback (most recent call last):
  File "d:\Project\AVR\VSC\esp-idf/components/esptool_py/esptool/espefuse.py", line 15, in <module>
    import espressif.efuse.esp32 as esp32_efuse
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\__init__.py", line 1, in <module>
    from . import operations
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\operations.py", line 18, in <module>
    from . import fields
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\fields.py", line 17, in <module>
    from .. import base_fields
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\base_fields.py", line 14, in <module>
    from bitstring import BitArray, BitString, CreationError
ImportError: cannot import name 'BitString' from 'bitstring' (d:\Project\AVR\VSC\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\bitstring.py)

### Steps to reproduce.

I just follow the procdure with "https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_ds/README.md"

### Build or installation Logs.

```plain
PS D:\Project\AVR\VSC\esp_mqtt_ssl_tls_ds>python .\configure_esp_secure_cert.py --port COM7 --device-cert .\main\client.crt --private-key .\main\client.key --target_chip esp32s3 --configure_ds --skip_flash
Traceback (most recent call last):
  File "d:\Project\AVR\VSC\esp-idf/components/esptool_py/esptool/espefuse.py", line 15, in <module>
    import espressif.efuse.esp32 as esp32_efuse
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\__init__.py", line 1, in <module>
    from . import operations
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\operations.py", line 18, in <module>
    from . import fields
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\esp32\fields.py", line 17, in <module>
    from .. import base_fields
  File "d:\Project\AVR\VSC\esp-idf\components\esptool_py\esptool\espressif\efuse\base_fields.py", line 14, in <module>
    from bitstring import BitArray, BitString, CreationError
ImportError: cannot import name 'BitString' from 'bitstring' (d:\Project\AVR\VSC\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\bitstring.py)
```

### More Information.

I use the ESP-IDF with VSCode extention in Window11.

Jinjungdong, 2023-01-12:
If you need addtional information, just tell me. i will support as fast as possible. Thank you.

AdityaHPatwardhan, 2023-01-12:
Hi @Jinjungdong  The issue seems to be related to some python package on windows.
I don't see a similar issue on macOS.
I will check the python packages in windows to see if there is any difference.

Jinjungdong, 2023-01-12:
Thank you for your response. I am looking forward to solve the issue.

Jinjungdong, 2023-01-12:
There is a additional information. I only add a 'confgure_esp_secure_cert.py' file and then i run in my environment

ozanoner, 2023-01-20:
this worked for me:
pip install bitstring==3.1.9

AdityaHPatwardhan, 2023-01-23:
Hi @Jinjungdong , Can you please execute the script from IDF environment.
I think the dependency that @ozanoner mentioned should be met with IDF environment.
I will update that in the requirements.txt as well.
Closing the issue since looks like it is related to python dependency.
Please re-open if it still persists.
Thanks

closed 2023-01-23 by AdityaHPatwardhan

Jinjungdong, 2023-01-25:
Thank you for response. I will check it.