espressif/esp-idf #10520: Provisioner timeout, giving up transaction

Vanperdung, 2023-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying provisioning 2 node at the same time in ble_mesh_provisioner example. Provisioner sent 2 link open messages but 1 node did not receive it. 
[image]
Provisioner must wait until timeout 30s and send another link open message, the remaining node will be provisioned. 
[image]
Why is that and is there a way to fix it?

Shane555, 2023-03-20:
Any updates on this?

forx157, 2023-03-21:
Hi, @Vanperdung .
Please provide more detailed logs, including provisioner and nodes.

closed 2025-03-31 by Alvin1Zhang