espressif/esp-idf #10522: Missing IRAM_ATTR for esp_timer_is_active() and esp_timer_restart()

AxelLin, 2023-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. esp_timer_is_active() should has IRAM_ATTR.
   It just calls timer_armed() which has IRAM_ATTR.
   https://github.com/espressif/esp-nimble/issues/50#issuecomment-1340515243

2. It's strange that esp_timer_start_periodic() and esp_timer_stop() have IRAM_ATTR but esp_timer_restart() does not has IRAM_ATTR.
  https://github.com/espressif/esp-nimble/issues/50#issuecomment-1342246467

RoshanESP, 2023-01-11:
Hi @AxelLin , 
1. Can you please share the exception traceback log. 
2. Can you please callgraph output by referring to this [ comment ](https://github.com/espressif/esp-idf/issues/10079#issuecomment-1302262774)

AxelLin, 2023-01-11:
For esp_timer_is_active():
I think this code path already proves esp_timer_is_active() needs IRAM_ATTR since it is called by npl_freertos_callout_is_active(). 
https://github.com/espressif/esp-nimble/blob/56ff5b83337697895997a39bf6593255cdc37268/porting/npl/freertos/src/npl_os_freertos.c#L852-L860

For esp_timer_restart():
I didn't hit the issue because I don't use that function at the moment.
It looks strange that a start/stop functions have IRAM_ATTR but the the restart (suppose to be a shortcut of stop/start) does not has IRAM_ATTR.

AxelLin, 2023-02-04:
@RoshanESP @rahult-github 

Any update? Can you check if this needs fix?

KonstantinKondrashov, 2023-03-01:
Hi @AxelLin!
Thanks for bringing this issue up. I will help to fix it and backport it to the other versions.

AxelLin, 2023-03-02:
@KonstantinKondrashov 

For https://github.com/espressif/esp-nimble/blob/56ff5b83337697895997a39bf6593255cdc37268/porting/npl/freertos/src/npl_os_freertos.c#L852-L860
npl_freertos_callout_is_active() has IRAM_ATTR, so esp_timer_is_active() must has IRAM_ATTR.
In this case, since the code is not directly called by user's application.
I think linker.lf is not a good option. (You are asking application developers to workaround sdk bug).

If you want to add Kconfig option, you have to make sure you handle the dependency correctly.
in above example,
npl_freertos_callout_is_active() / esp_timer_is_active() / xTimerIsTimerActive() should be consistent.
Either with IRAM_ATTR or without IRAM_ATTR for all of them.

AxelLin, 2023-04-10:
> Hi @AxelLin! Thanks for bringing this issue up. I will help to fix it and backport it to the other versions.

Any update for the fix @KonstantinKondrashov ?
Below is the panic I hit, adding IRAM_ATTR to esp_timer_is_active() fixed it.

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x4038ca40  RA      : 0x4038b834  SP      : 0x3fca09c0  GP      : 0x3fc94400
0x4038ca40: uxListRemove at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:197

0x4038b834: xTaskRemoveFromEventList at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3900

TP      : 0x3fc65ef0  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x40382dcc
0x40382dcc: npl_freertos_callout_deinit at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:761

S0/FP   : 0x3fc9ce8c  S1      : 0x0000007c  A0      : 0x3fc9cea4  A1      : 0x3fc9cea4
A2      : 0xffffffff  A3      : 0x000a000a  A4      : 0x0000000a  A5      : 0x3fc9ce60
A6      : 0x3fc9c3ec  A7      : 0x00000000  S2      : 0x3fc9cea4  S3      : 0xffffffff
S4      : 0x3fc9ce90  S5      : 0x3c0f9000  S6      : 0x3c0f9000  S7      : 0x00043518
S8      : 0x3c0f9000  S9      : 0x3c0f9000  S10     : 0x00000000  S11     : 0x04a70dd7
T3      : 0x40381fa2  T4      : 0x40382ece  T5      : 0x40381f9e  T6      : 0x40382ec2
0x40381fa2: npl_freertos_callout_set_arg at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:926

0x40382ece: npl_freertos_callout_remaining_ticks at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:888

0x40381f9e: npl_freertos_callout_get_ticks at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:879

0x40382ec2: npl_freertos_callout_is_active at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:856

MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x000a0012
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

closed 2023-04-25 by espressif-bot