espressif/esp-idf #10523: ESP32-S3 HTTPS esp_mbedtls_handshake take more than 3s on average

LeoKemp223, 2023-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.2-378-g9269a536ac-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-Korvo-2

### Power Supply used.

USB

### What is the expected behavior?

When I use HTTP POST/GET requests, each request usually takes about 100-300ms
After change HTTP to HTTPS and add server certificate verification, each request takes more than 3 seconds
I've tried the following configurations, but the time does not change much
1、
    esp_tls_cfg_t cfg = {
        .use_global_ca_store = true,
    };
2、
    esp_tls_cfg_t cfg = {
        .cacert_buf = (const unsigned char *) server_root_cert_pem_start,
        .cacert_bytes = server_root_cert_pem_end - server_root_cert_pem_start,
    };
3、
    esp_tls_cfg_t cfg = {
        .crt_bundle_attach = esp_crt_bundle_attach,
    };

Then I tracked the source code and ran and found that each request took less than esp_mbedtls_handshake
E (9881) esp-tls-mbedtls: -------start  = 9270ms esp_mbedtls_handshake
I (10691) esp-x509-crt-bundle: Certificate validated
E (15741) esp-tls-mbedtls: -------end =15130ms esp_mbedtls_handshake

Is this a normal HTTPS request time-consuming? Or is there a problem with my configuration? Is there a way to optimize request time-consuming?

### What is the actual behavior?

Same as above

### Steps to reproduce.

Same as above

### Debug Logs.

```plain
I (5661) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:bad6:1aff:fe33:d0e4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (9161) esp_netif_handlers: example_connect: sta ip: 192.168.3.137, mask: 255.255.255.0, gw: 192.168.3.1
I (9161) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.3.137
I (9171) example_connect: Connected to example_connect: sta
I (9171) example_connect: - IPv4 address: 192.168.3.137
I (9181) example_connect: - IPv6 address: fe80:0000:0000:0000:bad6:1aff:fe33:d0e4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (9191) example: Start https_request example
I (9201) example: https_request using crt bundle
E (9881) esp-tls-mbedtls: -------start  = 9270ms esp_mbedtls_handshake
I (10691) esp-x509-crt-bundle: Certificate validated
E (15741) esp-tls-mbedtls: -------end =15130ms esp_mbedtls_handshake
I (15741) example: Connection established...
I (15751) example: 106 bytes written
I (15751) example: Reading HTTP response...
HTTP/1.1 200 OK
Content-Length: 2091
Access-Control-Allow-Origin: *
Connection: close
Content-Type: application/json
Date: Wed, 11 Jan 2023 06:15:16 GMT
Strict-Transport-Security: max-age=631138519; includeSubdomains; preload
...
I (42881) example: https_request using global ca_store
E (44281) esp-tls-mbedtls: -------start  = 43670ms esp_mbedtls_handshake
E (47381) esp-tls-mbedtls: -------end =46770ms esp_mbedtls_handshake
I (47391) example: Connection established...
I (47401) example: 106 bytes written
I (47401) example: Reading HTTP response...
HTTP/1.1 200 OK
Content-Length: 2091
Access-Control-Allow-Origin: *
Connection: close
Content-Type: application/json
Date: Wed, 11 Jan 2023 06:15:48 GMT
Strict-Transport-Security: max-age=631138519; includeSubdomains; preload
```

LeoKemp223, 2023-01-16:
The cause has been identified as network and server-side reasons,
and the current test HTTPS request is about 1.3 seconds

closed 2023-01-16 by LeoKemp223

laukik-hase, 2023-01-16:
Hello, @LeoKemp223!

Thank you for sending the issue report. A similar issue was filed - see https://github.com/espressif/esp-idf/issues/8852. The TLS Session Resumption feature might help you in reducing connection handshake times.

You can refer the `https_request` example for implementation details - see [here](https://github.com/espressif/esp-idf/blob/8153bfe4125e6a608abccf1561fd10285016c90a/examples/protocols/https_request/main/https_request_example_main.c#L183).