espressif/esp-idf #10524: after opening PSRAM,the part of app can't load.

casitelwithdean, 2023-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using esp32-wroom-s3,when I close PSRAM, it works well.When i open  PSRAM,it can't load the part of app,and it will restart because of RTCWATCHDOG. What I select  is Octal Mode (I am sure it is right,because if I select Quad Mode,it will show me the error)
Here is the log.THANKS!
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x460
load:0x403c9700,len:0xc20
load:0x403cc700,len:0x2ddc
entry 0x403c98f0
I (33) opi psram: vendor id : 0x0d (AP)
I (33) opi psram: dev id    : 0x02 (generation 3)
I (33) opi psram: density   : 0x03 (64 Mbit)
I (36) opi psram: good-die  : 0x01 (Pass)
I (41) opi psram: Latency   : 0x01 (Fixed)
I (46) opi psram: VCC       : 0x01 (3V)
I (50) opi psram: SRF       : 0x01 (Fast Refresh)
I (56) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (61) opi psram: BurstLen  : 0x01 (32 Byte)
I (66) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (72) opi psram: DriveStrength: 0x00 (1/1)
W (77) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (87) MSPI Timing: tuning fail, best point is fallen back to index 1
I (94) spiram: Found 64MBit SPI RAM device
I (99) spiram: SPI RAM mode: sram 80m
I (103) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (110) cpu_start: Pro cpu up.
I (114) cpu_start: Starting app cpu, entry point is 0x40379434
0x40379434: call_start_cpu1 at D:/desktop/esp-4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x460
load:0x403c9700,len:0xc20
load:0x403cc700,len:0x2ddc
entry 0x403c98f0
I (34) opi psram: vendor id : 0x0d (AP)
I (34) opi psram: dev id    : 0x02 (generation 3)
I (34) opi psram: density   : 0x03 (64 Mbit)
I (37) opi psram: good-die  : 0x01 (Pass)
I (42) opi psram: Latency   : 0x01 (Fixed)
I (47) opi psram: VCC       : 0x01 (3V)
I (51) opi psram: SRF       : 0x01 (Fast Refresh)
I (56) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (62) opi psram: BurstLen  : 0x01 (32 Byte)
I (67) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (73) opi psram: DriveStrength: 0x00 (1/1)
W (77) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (88) MSPI Timing: tuning fail, best point is fallen back to index 1
I (95) spiram: Found 64MBit SPI RAM device
I (99) spiram: SPI RAM mode: sram 80m
I (104) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (111) cpu_start: Pro cpu up.
I (114) cpu_start: Starting app cpu, entry point is 0x40379434
0x40379434: call_start_cpu1 at D:/desktop/esp-4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x460
load:0x403c9700,len:0xc20
load:0x403cc700,len:0x2ddc
entry 0x403c98f0
I (34) opi psram: vendor id : 0x0d (AP)
I (34) opi psram: dev id    : 0x02 (generation 3)
I (34) opi psram: density   : 0x03 (64 Mbit)
I (37) opi psram: good-die  : 0x01 (Pass)
I (42) opi psram: Latency   : 0x01 (Fixed)
I (47) opi psram: VCC       : 0x01 (3V)
I (51) opi psram: SRF       : 0x01 (Fast Refresh)
I (56) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (62) opi psram: BurstLen  : 0x01 (32 Byte)
I (67) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (73) opi psram: DriveStrength: 0x00 (1/1)
W (77) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (88) MSPI Timing: tuning fail, best point is fallen back to index 1
I (95) spiram: Found 64MBit SPI RAM device
I (99) spiram: SPI RAM mode: sram 80m
I (104) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (111) cpu_start: Pro cpu up.
I (114) cpu_start: Starting app cpu, entry point is 0x40379434
0x40379434: call_start_cpu1 at D:/desktop/esp-4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.

Icarus113, 2023-01-12:
@casitelwithdean 

> What I select is Octal Mode (I am sure it is right,because if I select Quad Mode,it will show me the error)

Yes, this is correct. The `ESP32-S3-WROOM-1-N8R8` is with an Octal PSRAM.

Could you please share your IDF version, and the sdkconfig you're using?

Besides, I see the `W (88) MSPI Timing: tuning fail, best point is fallen back to index 1` log, under what temperature you're using your module? What if you enable [this config](https://github.com/espressif/esp-idf/blob/master/components/esp_psram/Kconfig.spiram.common#L111)?

casitelwithdean, 2023-01-12:
> @casitelwithdean
>
>
> Yes, this is correct. The `ESP32-S3-WROOM-1-N8R8` is with an Octal PSRAM.
>
> Could you please share your IDF version, and the sdkconfig you're using?
>
> Besides, I see the `W (88) [...]

Hi,thanks for replying.
I'm using 4.4.2 IDF-version.
Temperature here is 15°C
I guess because of my IDF-version,I didn't find this config in my sdkconfig.
by the way ,I used IO 35 36 37 on my PCB.Will this have an impact？
Thanks again!

Icarus113, 2023-01-12:
> by the way ,I used IO 35 36 37 on my PCB.Will this have an impact？

Ah, that's the reason. IO 35 36 37 are MSPI D6, D7 and DQS pin respectively. As it's an octal PSRAM, these pins will be used by MSPI to communicate with the PSRAM.

casitelwithdean, 2023-01-12:
> > by the way ,I used IO 35 36 37 on my PCB.Will this have an impact？
> 
> Ah, that's the reason. IO 35 36 37 are MSPI D6, D7 and DQS pin respectively. As it's an octal PSRAM, these pins will be used by MSPI to communicate with the PSRAM.

Thanks a lot!

closed 2023-01-12 by casitelwithdean