espressif/esp-idf #10525: BLE Bluedroid: No ESP_GATTC_SEARCH_RES_EVT for second peripheral

vdm97, 2023-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

While connected to a first peripheral and connecting to a second peripheral via BLE and using `esp_ble_gattc_search_service` to find a service the `ESP_GATTC_SEARCH_RES_EVT` event should be raised.

### What is the actual behavior?

The event `ESP_GATTC_SEARCH_RES_EVT` never gets raised.

### Steps to reproduce.

This is my procedure for connecting to a peripheral and subscribing to a heartrate monitor.
1. Use `esp_ble_gatt_set_local_mtu(ESP_GATT_MAX_MTU_SIZE)` to allow the maximum MTU.
2. Register GATT-client with `esp_ble_gattc_app_register`.
3. In the event `ESP_GATTC_REG_EVT` use `esp_ble_gattc_open` to open the connection. BDA is already known so no scanning necessary.
4. The event `ESP_GATTC_CONNECT_EVT` is not used.
5. In the event `ESP_GATTC_OPEN_EVT` use `esp_ble_gattc_send_mtu_req` to request higher MTU. Gets confirmed in event `ESP_GATTC_CFG_MTU_EVT` with a value of 498.
6. In the event `ESP_GATTC_DIS_SRVC_CMPL_EVT` use `esp_ble_gattc_search_service` to search for known service UUID. The start and end handles are saved in the event `ESP_GATTC_SEARCH_RES_EVT`. 
7. In the event `ESP_GATTC_SEARCH_CMPL_EVT` use `esp_ble_gattc_get_attr_count`, then `esp_ble_gattc_get_char_by_uuid`, then `esp_ble_gattc_register_for_notify` to register for notify on a known characteristic.
8. In the event `ESP_GATTC_REG_FOR_NOTIFY_EVT` use `esp_ble_gattc_get_attr_count`, then `esp_ble_gattc_get_descr_by_char_handle`, then `esp_ble_gattc_write_char_descr` to activate the notify descriptor.
9. In the event `ESP_GATTC_WRITE_DESCR_EVT` the connection procedure is finished and the event `ESP_GATTC_NOTIFY_EVT` delivers the data from the peripheral. 
10. In the event `ESP_GATTC_CLOSE_EVT` use `esp_ble_gap_disconnect` to close the physical connection. 
11. In the event `ESP_GATTC_DISCONNECT_EVT` use `esp_ble_gattc_app_unregister` to remove the GATT-client.

### More Information.

I run the ESP32 in WiFi and Bluetooth coexistence mode because i need both functionalities.
If i run the connection procedure for the first peripheral, then disconnect from the first peripheral after which i run the connection procedure for the second peripheral, the connection gets established successfully. When i disconnect from the second peripheral i can run the connection procedure for the first peripheral successfully and so on.. 
But running the connection procedure while connected to the first peripheral fails because the event `ESP_GATTC_SEARCH_RES_EVT` never gets raised. Swapping the first and second peripheral does not help. The peripheral is a [nordic nrf52840 DK](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk) with a simple [heartrate peripheral sample](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.2.0/zephyr/samples/bluetooth/peripheral_hr/README.html) ([source](https://github.com/nrfconnect/sdk-zephyr/tree/main/samples/bluetooth/peripheral_hr)) with a [removed battery service](https://github.com/nrfconnect/sdk-zephyr/blob/bcc88b7fe7a63ba8428bb57cc25e11cd0588e530/samples/bluetooth/peripheral_hr/src/main.c#L130) so i do not suspect the error there. When the [sending of the heartrate values is deactivated](https://github.com/nrfconnect/sdk-zephyr/blob/bcc88b7fe7a63ba8428bb57cc25e11cd0588e530/samples/bluetooth/peripheral_hr/src/main.c#L127) the event `ESP_GATTC_SEARCH_RES_EVT` just never gets raised. If the sending of the heartrate values is present an error is present in the ESP32 logs after which both connection are terminated.
I raised all `BT DEBUG LOG LEVEL`s to `VERBOSE` which does not help me but maybe does provide some information to the maintainers.  Because of these changes i had to raise the `CONFIG_BTU_TASK_STACK_SIZE` from `4096` to `8192` and also raised the `CONFIG_BTC_TASK_STACK_SIZE` from `3072` to `6144` just to be sure enough stack is provided.
If needed i can translate the german parts in the logs to english...

vdm97, 2023-01-11:
Logs were too long...

<details>
<summary>sdkconfig</summary>
<br>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
[... 2538 lines cut ...]
I (19768) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0004  BDA: fd55f36f3610
W (19778) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
I (19778) BT_GATT: GATT_GetConnIdIfConnected status=0

W (19788) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
W (19788) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16
I (19798) BT_GATT: GATT_Deregister gatt_if=3
I (19798) BT_GATT: GATT_Listen gatt_if=3
I (19808) BT_BTM: BTM_ReadConnectability

W (19808) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (19818) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (19828) BT_GATT: GATT_Deregister gatt_if=4
I (19828) BT_GATT: GATT_Listen gatt_if=4
I (19838) BT_BTM: BTM_ReadConnectability

E (19838) esp_gattc_cb: ESP_GATTC_CLOSE_EVT interface_id: 3
E (19848) esp_gattc_cb: Die zu schliessende BLE-Verbindung zu einem peripheral mit interface_id 3 konnte nicht gefunden werden.
E (19858) esp_gattc_cb: ESP_GATTC_UNREG_EVT interface_id: 3
E (19858) esp_gattc_cb: ESP_GATTC_DISCONNECT_EVT interface_id: 4
E (19868) esp_gattc_cb: Die zu beendende BLE-Verbindung zu einem peripheral mit interface_id 4 konnte nicht gefunden werden.
E (19878) esp_gattc_cb: ESP_GATTC_UNREG_EVT interface_id: 4
```

</details>

xiewenxiang, 2023-01-30:
Hi @vdm97, 
Please remove the reproduce steps 11, after the first connection has been disconnected, don't call the `esp_ble_gattc_app_unregister ` API.

vdm97, 2023-01-30:
Hi, thank you for your time. Where should i put the call to `esp_ble_gattc_app_unregister` then? After the clients disconnects i do not need the registered GATT-client anymore.
Also, this does not fix my error because that part of the code never gets called when i try to connect to a second peripheral while there is already a connection present with a first peripheral. Maybe i have missed something?

xiewenxiang, 2023-01-31:
In fact, you don't need to call `esp_ble_gattc_app_unregister `, If you no longer need to use it, you can close the entire stack.

1. esp_bluedroid_disable
2. esp_bluedroid_deint
3. esp_bt_controller_disable
4. esp_bt_controller_deinit

Please try using `esp_ble_gattc_close` to terminated the connection. 
But I not sure if this will solve your problem, if not, I'll give you an example.

vdm97, 2023-02-01:
When i want to connect to a peripheral the first thing i do is to register the GATT-client (`esp_ble_gattc_app_register`). After a while i want to close the connection so i use `esp_ble_gattc_close` and then `esp_ble_gap_disconnect`. Because i do not need the GATT-client anymore my thinking was to free the internal resources by calling `esp_ble_gattc_app_unregister`. Maybe my reasoning is wrong and `esp_ble_gattc_close` or `esp_ble_gap_disconnect` already free the internal resources so a call to `esp_ble_gattc_app_unregister` is not needed?
I hope this makes clear that i cannot just deinitialize the whole BLE-stack because it is just 1 client that i do not need and i want to connect to other peripherals (with their own call to `esp_ble_gattc_app_register` etc.).
And to make things more clear. My problem does not occur when i disconnect from a peripheral, but when i want to connect to a second peripheral while still connected to a first peripheral. In order to achieve that i run the procedure for connecting to a peripheral (steps 2 to 9) for the first peripheral which succeeds and then i run the steps 2 to 9 again (with different data) in order to connect to the second peripheral. This does not succeed because the event `ESP_GATTC_SEARCH_RES_EVT` never gets raised for the second peripheral.

vdm97, 2023-05-25:
As it turned out, this error was a side effect of corrupted memory... Everything works great after fixing that error.

closed 2023-05-25 by vdm97