espressif/esp-idf #10527: xRingbufferSendAcquireFromIsr

malachib, 2023-01-12:
### Is your feature request related to a problem?

My ISR handler needs to queue objects which must be made in place/not copied.  In other non-ISR scenarios I do a placement new into the buffer from `xRingbufferSendAcquire` and it works quite well.

### Describe the solution you'd like.

I'd like an ISR-compatible set of functions which effectively are exactly `xRingbufferSendAcquire` and `xRingbufferSendComplete`

### Describe alternatives you've considered.

- doing a smart move/copy or in other words brute forcing the not-copyable object to be copied.  It's a bit of legwork and rubs me the wrong way especially in an ISR
- Make a 2nd queue which feeds the ring buffer.  But at that point I'm kind of writing xRingbufferSendAcquireFromIsr myself.  Which I am actually interested in participating in, but it seems I should consult the experts first

### Additional context.

This is to make a "delegate queue" which is a set of no-heap std::functions - kind of a more c++11 spin on something like `xTimerPendFunctionCall` and a more embedded friendly spin on `asio`

malachib, 2023-02-04:
Question - are there any specific architectural stumbling blocks to doing such a thing, or would the team say it's more a matter of legwork?  If it's the latter, I could try a fork and combine what I see `xRingbufferSendFromISR` and `xRingbufferSendAcquire` doing.  If it's the former, I'd rather not go down the rabbit hole :)

Dazza0, 2023-02-13:
@malachib I don't see any technical blockers to implementing a `xRingbufferSendAcquireFromISR()` and `xRingbufferSendCompleteFromISR()` versions of the API. This feature request has been added to our internal todo list.

AxelLin, 2023-08-21:
> @malachib I don't see any technical blockers to implementing a `xRingbufferSendAcquireFromISR()` and `xRingbufferSendCompleteFromISR()` versions of the API. This feature request has been added to our internal todo list.

Any update? It has been 6 months since your last reply.

malachib, 2023-08-21:
> Any update? It has been 6 months since your last reply.

@AxelLin not from me.  It seemed Espressif internally was taking point on this.  That being the case, I didn't want to duplicate / not coordinate efforts with them.  If there's a proper channel for me to do so, I'd be happy to pursue it.

malachib, 2023-08-29:
My first step in this journey is to run `esp_ringbuf`'s unit test.  This has confounded me.  After an `idf.py flash monitor` I end up with a "blank" unit test app which appears to have some corrupted text for bonus fun, i.e.

```
I (319) app_start: Starting scheduler on CPU0
I (324) 0_s2aIt: 2tarting_task:uler oed on C[0m
I (334) main_task: Calling app_main()
```

I never normally see "line noise".  I've spent some quality time with:

1.  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/index.html
2. https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32/contribute/esp-idf-tests-with-pytest.html
 
I have been using unit tests for years.  Still, number 2 was particularly confusing.  Should I be looking somewhere else?