espressif/esp-idf #10528: I generate sdkconfig lost any field

WangShuoran, 2023-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am use ESP-IDF v4.4.3-release, windows10 with idf-installer. My developer board is ESP32C3 with Vscode IDE with espressif extension.

When I use idf.py menuconfig and open Component config -> Power Management. I find the menu lost config CONFIG_PM_SLP_DISABLE_GPIO in document https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.3/esp32c3/api-reference/kconfig.html.

But I look the example power_manage the sdkconfig.default is use CONFIG_PM_SLP_DISABLE_GPIO config. And the components folder use CONFIG_PM_SLP_DISABLE_GPIO at any position:
[image]

So I want to know does this config lost in sdkconfig?

igrr, 2023-01-12:
@WangShuoran That option `depends on CONFIG_FREERTOS_USE_TICKLESS_IDLE`, perhaps you don't have that one enabled? Because of that, `CONFIG_PM_SLP_DISABLE_GPIO` isn't visible.

WangShuoran, 2023-01-12:
Hello, I just found out that I am preparing for feedback. I want you introduce in document. Thank you!

closed 2023-01-12 by WangShuoran