espressif/esp-idf #10529: driver/uart.h not found in component

Kampi, 2023-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.0

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The program compiles without issues

### What is the actual behavior?

I have a custom component that is using the UART driver. I get the compilation error `fatal error: driver/uart.h: No such file or directory` after upgrading the esp-idf to version 5.0.0 and using the 4.1 branch of my RAK library (https://github.com/Kampi/ESP32-RAK3172/tree/4.1.0_Dev).

### Steps to reproduce.

1. Using the upstream branch from PlatformIO that supports idf v5 (https://github.com/platformio/platform-espressif32.git)
2. Compile the application code

### Build or installation Logs.

```plain
Processing debug (board: esp32-c3-devkitm-1; platform: https://github.com/platformio/platform-espressif32.git; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.3.0+sha.6660f2c) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1

...
Compiling .pio\build\debug\components\rak3172\src\Modes\RF\rak3172_rf.o
In file included from components\rak3172\include/rak3172.h:23,
                 from components\rak3172\src\Commands\rak3172_commands_rui3.cpp:24:
components\rak3172\include\Definitions/rak3172_defs.h:23:10: fatal error: driver/uart.h: No such file or directory
   23 | #include <driver/uart.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
In file included from components\rak3172\include/rak3172.h:23,
                 from components\rak3172\src\Commands\rak3172_commands.cpp:22:
components\rak3172\include\Definitions/rak3172_defs.h:23:10: fatal error: driver/uart.h: No such file or directory
   23 | #include <driver/uart.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\debug\soc\dport_access_common.o
*** [.pio\build\debug\components\rak3172\src\Commands\rak3172_commands_rui3.o] Error 1
*** [.pio\build\debug\components\rak3172\src\Commands\rak3172_commands.o] Error 1
In file included from components\rak3172\include/rak3172.h:23,
                 from components\rak3172\src\rak3172.cpp:26:
components\rak3172\include\Definitions/rak3172_defs.h:23:10: fatal error: driver/uart.h: No such file or directory
   23 | #include <driver/uart.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\debug\components\rak3172\src\rak3172.o] Error 1
```

ESP-YJM, 2023-01-12:
@Kampi You may add "include/driver" to `PRIV_INCLUDE_DIRS` or "driver" to `PRIV_REQUIRES` in your `rak3172` components CMakeLists.txt file.

Kampi, 2023-01-12:
Hi @ESP-YJM,

weird, because I have tried it before writing this ticket. Anyway, it was the correct tip. Thank you.

closed 2023-01-12 by Kampi