espressif/esp-idf #10531: [v4.4.3][SD Card] sdmmc mount sometimes hits infinite loop in sdmmc_host_clock_update_command()

chipweinberger, 2023-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-215-g9f3f82f54b

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

1. esp_vfs_fat_sdmmc_mount() should succeed

### What is the actual behavior?

~50% of the time, for some applications,  esp_vfs_fat_sdmmc_mount() hits an infinite loop.

```
static void sdmmc_host_clock_update_command(int slot)
{
    // Clock update command (not a real command; just updates CIU registers)
    sdmmc_hw_cmd_t cmd_val = {
        .card_num = slot,
        .update_clk_reg = 1,
        .wait_complete = 1
    };
    bool repeat = true;
    while(repeat) {
        sdmmc_host_start_command(slot, cmd_val, 0);
        while (true) {
            // Sending clock update command to the CIU can generate HLE error.
            // According to the manual, this is okay and we must retry the command.
            if (SDMMC.rintsts.hle) {
                SDMMC.rintsts.hle = 1;
                repeat = true;
                break;
            }
            // When the command is accepted by CIU, start_command bit will be
            // cleared in SDMMC.cmd register.
            if (SDMMC.cmd.start_command == 0) {
                repeat = false;
                break;
            }
        }
    }
}

```

### Steps to reproduce.

1. mount a sd card with sdmmc
2. there are probably other necessary steps here...

### Debug Logs.

```plain
I (5079) pd sd card: Mounting SD card. format_if_mount_failed? NO
D (5079) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
D (5079) intr_alloc: Connected src 30 to int 17 (cpu 0)
D (5079) sdmmc_periph: using GPIO11 as clk pin
I (5079) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5079) sdmmc_periph: using GPIO12 as cmd pin
I (5089) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5089) sdmmc_periph: using GPIO10 as d0 pin
I (5089) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5089) sdmmc_periph: using GPIO9 as d1 pin
I (5089) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5099) sdmmc_periph: using GPIO14 as d2 pin
I (5099) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5099) sdmmc_periph: Force D3 high
I (5099) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (5099) sdmmc_periph: esp_rom_gpio_connect_in_signal() card_int
I (5099) sdmmc_periph: esp_rom_gpio_connect_in_signal() card_detect
I (5109) sdmmc_periph: esp_rom_gpio_connect_in_signal() write_protect
I (5109) sdmmc_periph: sdmmc_host_set_card_clk()
```

chipweinberger, 2023-01-12:
There should probably be a reasonable timeout....

The sd card might be in a bad state. 

It seems to happen more frequently after an esp_restart() into another app.

chipweinberger, 2023-01-12:
@igrr , would you recommend SPI or SDMMC if the sole consideration is reliability?

chipweinberger, 2023-01-12:
**Created a PR:** https://github.com/espressif/esp-idf/pull/10532

igrr, 2023-01-12:
Possibly related to https://github.com/espressif/esp-idf/issues/2986. There was some draft fix for that, we'll take a look.

chipweinberger, 2023-01-12:
https://github.com/espressif/esp-idf/issues/2986#issuecomment-486386499

"Resetting the peripheral before initialization seems like an acceptable work around."

`periph_module_reset(PERIPH_SDMMC_MODULE);`

In my PR, should I add this line somewhere?

chipweinberger, 2023-01-12:
I've confirmed the PR does fix this issue! 😀

**We now just log the error, and continue execution!**

```
I (5067) pd sd card: Mounting SD card. format_if_mount_failed? NO
D (5077) sdmmc_periph: peripheral version 5342270a, hardware config 03c44c83
D (5077) intr_alloc: Connected src 30 to int 17 (cpu 0)
D (5077) sdmmc_periph: using GPIO11 as clk pin
I (5077) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5077) sdmmc_periph: using GPIO12 as cmd pin
I (5077) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5087) sdmmc_periph: using GPIO10 as d0 pin
I (5087) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5087) sdmmc_periph: using GPIO9 as d1 pin
I (5087) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (5087) sdmmc_periph: using GPIO14 as d2 pin
I (5087) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5097) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (5097) sdmmc_periph: disable clk failed
E (5097) sdmmc_periph: sdmmc_host_set_card_clk() 400kHz failed
E (5097) vfs_fat_sdmmc: slot init failed (0x107).
I (5097) freertos utils: idle task_wdt enable
I (5097) sd update: on filesystem MOUNT FAILURE called
I (5107) pd notifications: notify issue
W (5107) pd issues: RAISE - [Sd Card] Mount Failure | None
E (5107) : e1 : [pd sd card] esp_vfs_fat_sdmmc_mount() failed. ESP_ERR_TIMEOUT
e2 : [pd sd card] pd_sd_mount_sdmmc() failed.
e3 : [sd update] pd_sd_mount() failed
```

adokitkat, 2023-04-13:
https://github.com/espressif/esp-idf/pull/10532 was merged to master so I will close this issue. Thank you for participating :)

closed 2023-04-13 by adokitkat