espressif/esp-idf #10534: I may work out "hello_world project cmake failed"

HBUTHUANGPX, 2023-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

realse-v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

when I try to

> idf.py set-target esp32s3

the cmd out is 

```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/lc/esp/hello_world/build' is empty. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/lc/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/lc/esp/hello_world"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at build/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/lc/esp/hello_world/build/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /home/lc/esp/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  /home/lc/esp/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  /home/lc/esp/esp-idf/tools/cmake/build.cmake:67 (git_describe)
  /home/lc/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/lc/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/lc/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/lc/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/lc/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/lc/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/lc/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /home/lc/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
[... 47 lines cut ...]
fatal: Needed a single revision
```

I follow the directory:` /home/lc/esp/hello_world/build/CMakeFiles/git-data/` and check where is` 'head-ref'`,and I only see a file named with `HEAD`

Then I follow the call stack,check the cmake file `tools/cmake/third_party/GetGitRevisionDescription.cmake`.I see `@GIT_DATA@/head-ref" COPYONLY`
then I replace the false cmake command @/head-ref with @/HEAD,and try to  idf.py set-target again,it works

### What is the actual behavior?

none

### Steps to reproduce.

none

### Build or installation Logs.

```plain
none
```

### More Information.

none

igrr, 2023-01-13:
Thanks for reporting this issue @HBUTHUANGPX.

It looks like something got broken in the git checkout of your ESP-IDF copy. To help us find a way to reproduce this, could you please add more information:

* What is the content of `/home/lc/esp/hello_world/build/CMakeFiles/git-data/HEAD` file?
* If the content looks like `ref: refs/heads/release/v5.0`, please check if you have either of the files:
  * /home/lc/esp/esp-idf/.git/refs/heads/release/v5.0
  * /home/lc/esp/esp-idf/.git/logs/refs/heads/release/v5.0
* If you run `git describe`, `git describe HEAD`, `git status` commands in /home/lc/esp/esp-idf/, what do they report?

HBUTHUANGPX, 2023-01-14:
1. I checked `/home/lc/esp/esp-idf/.git/refs/heads/release/v5.0` and `/home/lc/esp/esp-idf/.git/logs/refs/heads/release/v5.0`
,but nothing.
2. I run `git describe` and  `git describe HEAD`

     report `fatal:没有发现名称，无法描述任何东西`没有发现名称，无法描述任何东西`

     translate in English is `No name found, can't describe anything`
3. I run `git status`,it report 
 ```
   位于分支 master

   尚无提交
```

Dazza0, 2023-02-27:
@HBUTHUANGPX  好像你的 git repo 有异常，可能在 `git clone` 的过程中。请你再次完整的尝试 `git clone --recursive https://github.com/espressif/esp-idf.git`。如果继续有问题，请重开这个 issue。

closed 2023-02-27 by Dazza0