espressif/esp-idf #10535: Invalid header when using Secure Boot and Flash Encryption

davidallenmann, 2023-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-347-g9ee3c8337d

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Device to boot normally and perform flash encryption.

### What is the actual behavior?

After burning bootloader and firmware get invalid header.

### Steps to reproduce.

Following instructions from here: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/security/secure-boot-v2.html

Using secure boot signing key generated by espsecure.py generate_signing_key

1. Firmware works normally when not encrypted and when secure boot is not enabled.
2. Burning the signed bootloader and code produces invalid error header. I also burn certs to esp_secure_cert. I have done this both before and after burning the bootloader and still get the invalid header error.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048c3e
invalid header: 0x507bc083
invalid header: 0x507bc083
invalid header: 0x507bc083
```
3. I have tried implementing the changes suggested in #5771 and get the same results. See SDKconfig settings below.

SDKCONFIG
```
#
# Partition Table
#
# CONFIG_PARTITION_TABLE_SINGLE_APP is not set
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE is not set
# CONFIG_PARTITION_TABLE_TWO_OTA is not set
CONFIG_PARTITION_TABLE_CUSTOM=y
# CONFIG_PARTITION_TABLE_SINGLE_APP_ENCRYPTED_NVS is not set
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE_ENC_NVS is not set
# CONFIG_PARTITION_TABLE_TWO_OTA_ENCRYPTED_NVS is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota_haiku2.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_ota_haiku2.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xA000
CONFIG_PARTITION_TABLE_MD5=y
# end of Partition Table

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_ERROR=y
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=1
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=y
CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET=3
CONFIG_BOOTLOADER_FACTORY_RESET_PIN_LOW=y
# CONFIG_BOOTLOADER_FACTORY_RESET_PIN_HIGH is not set
CONFIG_BOOTLOADER_OTA_DATA_ERASE=y
CONFIG_BOOTLOADER_DATA_FACTORY_RESET="nvs"
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_HOLD_TIME_GPIO=5
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
# CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_SUPPORTS_RSA=y
CONFIG_SECURE_TARGET_HAS_SECURE_ROM_DL_MODE=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_AES128=y
# CONFIG_SECURE_FLASH_ENCRYPTION_AES256 is not set
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y

#
# Potentially insecure options
#
# CONFIG_SECURE_BOOT_ALLOW_JTAG is not set
# CONFIG_SECURE_BOOT_ALLOW_SHORT_APP_PARTITION is not set
# CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS is not set
# CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS is not set
# end of Potentially insecure options

CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
# CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE is not set
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
# end of Security features

```

Partition Table
```
esp_secure_cert, 0x3F,    ,     0xD000,   0x6000,   encrypted
nvs,             data, nvs,     0x13000,  0x6000,   
otadata,         data, ota,     0x19000,  0x2000,   encrypted
phy_init,        data, phy,     0x1b000,  0x1000,   encrypted
factory,         app,  factory,        ,      5M,   encrypted
ota_0,           app,  ota_0,          ,      5M,   encrypted
ota_1,           app,  ota_1,          ,      5M,   encrypted
storage,         data, nvs,            , 0x10000,   encrypted
nvs_key,         data, nvs_keys,       , 0x1000,    encrypted
```

mahavirj, 2023-01-13:
@davidallenmann 

It is not recommended to flash plain-text build artifacts once you enable flash encryption in release mode. I am afraid but this probably means that you have no way to recover this device now. Can you please confirm how you had generated the flash encryption key? Was it through bootloader or you had externally programmed it?

davidallenmann, 2023-01-13:
I don't believe I flashed any plain-text build artifacts after enabling flash encryption, unless one of these steps caused a system reset. The three steps below were performed without cycling power or resetting the ESP32-S3

The flash encryption key was generated by the ESP32-S3 I believe using the menuconfig settings. 

The secure boot signing key was generated by `espsecure.py generate_signing_key`. 

I performed three flashes usually in this order (on my last attempt I also tried with the secure_certs flash step last instead of first--in case they were getting written over). 

1. secure_certs: 
```
python3 configure_esp_secure_cert.py -p /dev/cu.usbserial-110 --keep_ds_data_on_host --ca-cert root_cert_auth.pem --device-cert $1/cert.pem --private-key $1/private.key --target_chip esp32s3 --secure_cert_type cust_flash
esptool.py --no-stub --port /dev/cu.usbserial-110 write_flash 0xD000 $1/esp_ds_data/esp_secure_cert.bin
```
2. encrypted bootloader: 
`esptool.py write_flash` 
 using command suggested by` idf.py bootloader`

3. binary: 
Note the command suggested by the `idf.py` for flashing generated an error on my second attempt at this. This error is below.
So, flashed with:` idf.py flash` on each of my 3 attempts (I used new modules each time).

I tried this with 3 modules making some changes each time
Attempt 1: IDF v4.4 
Attempt 2: Updated IDF to latest 4.4.3
Attempt 3: Set to Allow potentially insecure options. And enable ROM download mode.  Flash bootloader. Flash binary. Flash certs (last).  Already showing invalid header without reset on serial monitor. 

```
/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp/esp-idf-v4.4.2/components/esptool_py/esptool/esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size keep --flash_freq 80m 0xa000 build/partition_table/partition-table.bin 0x19000 build/ota_data_initial.bin 0x20000 build/xxxxx.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:d8:7e:b0
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000a000 to 0x0000afff...
Flash will be erased from 0x00019000 to 0x0001afff...
Flash will be erased from 0x00020000 to 0x00480fff...
Erasing flash...
Took 0.08s to erase flash block
Wrote 3072 bytes at 0x0000a000 in 0.2 seconds (99.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.08s to erase flash block
Wrote 8192 bytes at 0x00019000 in 0.7 seconds (100.1 kbit/s)...
Hash of data verified.
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01066430: Operation or feature not supported)
```

mahavirj, 2023-01-14:
@davidallenmann 

Can you please check if you can get the efuse summary of the device using:

```
espefuse.py --chip esp32s3 summary
```

If this command does not work then most likely you have secure ROM DL mode enabled on the device, in that case please run following command:

```
esptool.py --no-stub --chip esp32s3 get_security_info
```

Please share the results here.

davidallenmann, 2023-01-14:
```
esptool.py --no-stub --chip esp32s3 --port  /dev/cu.usbserial-110 get_security_info
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3 in Secure Download Mode
Enabling default SPI flash mode...
Flags: 0x000006f5 (0b11011110101)
Flash_Crypt_Cnt: 0x0
Key_Purposes: (9, 4, 0, 0, 0, 0, 12)
Chip_ID: 9
Api_Version: 0
Hard resetting via RTS pin...
```

mahavirj, 2023-01-15:
@davidallenmann 

> Key_Purposes: (9, 4, 0, 0, 0, 0, 12)

As per this both flash enc key and secure boot digest has been programmed on the device. EFuse purpose codes for S3 can be found [here](https://github.com/espressif/esp-idf/blob/db9586c53f31a65abf0a51cac7072d7750f7b073/components/efuse/esp32s3/include/esp_efuse_chip.h#L63).

> Flash_Crypt_Cnt: 0x0

This suggests that flash encryption is not yet enabled on this device. Probably flash encryption sequence was not completed on this device. Please try flashing plaintext artifacts (bootloader, partition-table and application) again, and then flash encryption sequence should start again. Note: basic flash writes are supported in secure DL mode with `--no-stub` argument to `esptool`.

davidallenmann, 2023-01-16:
I am able to reflash the encyrpted bootloader.
I get the following error when flashing the boot table and binary.
I got this error the first time I tried writing the binary with esptool.py on one of my previous attempts

```
python esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size keep --flash_freq 80m 0xa000 build/partition_table/partition-table.bin 0x19000 build/ota_data_initial.bin 0x20000 build/xxxxxx.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000a000 to 0x0000afff...
Flash will be erased from 0x00019000 to 0x0001afff...
Flash will be erased from 0x00020000 to 0x00480fff...
Erasing flash...
Took 0.09s to erase flash block
Wrote 3072 bytes at 0x0000a000 in 0.3 seconds (96.4 kbit/s)...
Erasing flash...
Took 0.13s to erase flash block
Wrote 8192 bytes at 0x00019000 in 0.7 seconds (96.4 kbit/s)...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```

mahavirj, 2023-01-17:
@davidallenmann 

It is difficult to say why this issue occurred in your setup. However, you can enable both flash enc and secure boot simultaneously, there should not be a problem with that.

Regarding flash write error, please see my earlier comment about `--no-stub` argument to esptool command. Since secure ROM DL mode is already enabled, providing `--no-stub` is mandatory during flash write. Please refer to docs for this at https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/security.html#uart-dl-mode

davidallenmann, 2023-01-17:
@mahavirj
--no-stub is in the command that fails (see output above). It seems to successfully write the partition table but fails when writing the binary

mahavirj, 2023-01-17:
Aha, my bad! Indeed the command looks correct. I am not sure why it failed when it reached to the application partition erase stage. Would you mind trying this workflow on official S3 devkit once? 

> I am using my own boards and programming with a serial programmer that has DTR tied to GPIO0. EN pin is connected to RTS. Not sure if that could cause a reset at some point in the flashing process.

Recommend checking this against our official S3 devkits schematics available at https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html#hardware

CC @radimkarnis

davidallenmann, 2023-01-17:
I get the same error using the ESP32-S3-WROOM2 DEVKIT-C board (changed to octal flash mode in menuconfig and recompiled).

```
python  esptool.py --chip esp32s3 --port=/dev/cu.usbserial-110 --baud=460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader.bin               
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:3b:af:b4
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00009fff...
Erasing flash...
Flash params set to 0x024f
Took 0.29s to erase flash block
Wrote 40960 bytes at 0x00000000 in 3.3 seconds (99.1 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
python esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size keep --flash_freq 80m 0xa000 build/partition_table/partition-table.bin 0x20000 build/xxxxx.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:3b:af:b4
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000a000 to 0x0000afff...
Flash will be erased from 0x00020000 to 0x00480fff...
Erasing flash...
Took 0.08s to erase flash block
Wrote 3072 bytes at 0x0000a000 in 0.2 seconds (101.1 kbit/s)...
Hash of data verified.
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01063136: Operation or feature not supported)
davidmann@Davids-Macbook haiku2 % /Users/davidmann/.espressif/python_env/idf4.4_py3.9_env/bin/python  /Users/davidmann/esp/esp-idf-v4.4.2/components/esptool_py/esptool/esptool.py --chip esp32s3 --port=/dev/cu.usbserial-110 --baud=460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/bootloader.bin               
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:3b:af:b4
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00009fff...
Erasing flash...
Flash params set to 0x024f
Took 0.29s to erase flash block
Wrote 40960 bytes at 0x00000000 in 3.3 seconds (99.1 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
davidmann@Davids-Macbook haiku2 % /Users/davidmann/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp/esp-idf-v4.4.2/components/esptool_py/esptool/esptool.py -p /dev/cu.usbserial-110 -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size keep --flash_freq 80m 0xa000 build/partition_table/partition-table.bin 0x20000 build/haiku2.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:3b:af:b4
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000a000 to 0x0000afff...
Flash will be erased from 0x00020000 to 0x00480fff...
Erasing flash...
Took 0.08s to erase flash block
Wrote 3072 bytes at 0x0000a000 in 0.2 seconds (101.1 kbit/s)...
Hash of data verified.
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01063136: Operation or feature not supported)
```

Security Info
```
esptool.py --no-stub --chip esp32s3 --port  /dev/cu.usbserial-110 get_security_info
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:3b:af:b4
Enabling default SPI flash mode...
Flags: 0x00000680 (0b11010000000)
Flash_Crypt_Cnt: 0x7
Key_Purposes: (4, 0, 0, 0, 0, 0, 12)
Chip_ID: 9
Api_Version: 0
Hard resetting via RTS pin...
```

I also tried increasing the partition table offset just in case there needed to be space (as it was there was exact byte alignment)

davidallenmann, 2023-01-18:
A test on a new board works with only Flash encryption enabled and UART ROM download mode permanently disabled.
So, it seems to have something to do with Secure Boot.

KonstantinKondrashov, 2023-01-19:
Hi @davidallenmann!
I tested SB + FE  for ESP32-S3 on the IDF v4.4.3-347-g9ee3c8337d version, it works, no issue that you have faced.

Here https://github.com/espressif/esp-idf/issues/10535#issuecomment-1385642447, it is strange why do you flash the bootloader twice?

Could you describe all steps that you do? I do not fully understand it.

In a simple way you need:
-  have erased flash
- idf.py build with enabled options:
  - CONFIG_SECURE_BOOT=y
  - CONFIG_SECURE_BOOT_V2_ENABLED=y
  - CONFIG_SECURE_BOOT_SIGNING_KEY="test/secure_boot_signing_key.pem"
  - CONFIG_SECURE_FLASH_ENC_ENABLED=y
- idf.py bootloader-flash (flash only bootloader). SB and FE will not be enabled until the partition table and app be found.
- idf.py flash monitor. If you have an issue at this stage then you need to flash the partition table at the end. It will help you to prevent enabling FE and SB until everything will be completely flashed.

What is your connection to chip? Is it a built-in USB-UART into S3 chip or an external USB-UART?

Minor: about Partition Table. You need to remove all offsets in your partition_table.csv file and use only `CONFIG_PARTITION_TABLE_OFFSET=0xA000`, in this case, esp_secure_cert should go into 0xD000 and the following go right after the previous partition. You will get the same result, and the ability to move it, only change one Kconfig option (CONFIG_PARTITION_TABLE_OFFSET).

davidallenmann, 2023-01-19:
The following worked. I think the issue may have either been the order of operations or the partition table offset being wrong. Thanks so much for your help.

Using ESP32-S3-WROOM2 DevKit-C and USB cable connected to UART port.

Set partition table offset to 0xC000 and removed offsets in partition_table.csv

1. esptool.py  --port /dev/cu.usbserial-110 erase_flash
2. idf.py bootloader
3.  python  esptool.py --chip esp32s3 --port=/dev/cu.usbserial-110 --baud=460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dout --flash_freq 80m --flash_size 16MB 0x0 bootloader.bin
4. Flash security creds
5. idf.py build
6. idf.py --port /dev/cu.usbserial-110 flash -b 1500000 monitor

KonstantinKondrashov, 2023-01-20:
Summary:
The bootloader with SB + FE for ESP32-S3 takes 45056 bytes  [0x0000 - 0xAFFFF].
Your partition table was at offset 0xA000 (see CONFIG_PARTITION_TABLE_OFFSET in sdkconfig)
So it means that the last sector of the bootloader was overwritten when you flash the partition table. The last sector of the bootloader is the signature of the secure boot.

The build system should throw an error.
```
Error: Bootloader binary size 0xb000 bytes is too large for partition table offset 0xa000. Bootloader binary can be maximum 0xa000 (40960) bytes unless the partition table offset is increased in the Partition Table section of the project configuration menu.
```

But is still not clear to me how you get `invalid header: 0x507bc083`, seems, at some point, you overwrite the bootloader with a new one after flash encryption was already enabled.  Please expose your steps how did you reach this state? (it can help us to prevent it in the future for other users). Thanks.

davidallenmann, 2023-01-20:
Updating to latest secure cert library still gives the no bootable app partitions issue.

1. I first uploaded without Secure Boot enabled and without flash encryption to make sure the new secure cert library was working correctly, which it was.

2. Turned on Secure Boot and Flash Encryption.
3. Built and flashed bootloader
```
Generated signed binary image /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/bootloader.bin from /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/bootloader-unsigned.bin
[4/4] Completed 'bootloader'

Bootloader build complete.
davidmann@Davids-Macbook haiku2 %  /Users/davidmann/.espressif/python_env/idf4.4_py3.9_env/bin/python  /Users/davidmann/esp/esp-idf-v4.4.2/components/esptool_py/esptool/esptool.py --chip esp32s3 --port=/dev/cu.usbserial-110 --baud=460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dout --flash_freq 80m --flash_size 16MB 0x0 /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/bootloader.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:db:bb:94
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00009fff...
Erasing flash...
Flash params set to 0x034f
Took 0.36s to erase flash block
Wrote 40960 bytes at 0x00000000 in 3.3 seconds (98.1 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```
4. idf.py build
5. idf.py flash monitor
```

esptool.py v3.3.2
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/davidmann/w/haiku2-release/haiku2/build/haiku2-unsigned.bin
[1399/1400] Generating signed binary image
espsecure.py v3.3.2
1 signing key(s) found.
Signed 4521984 bytes of data from /Users/davidmann/w/haiku2-release/haiku2/build/haiku2-unsigned.bin. Signature sector now has 1 signature blocks.
Generated signed binary image /Users/davidmann/w/haiku2-release/haiku2/build/haiku2.bin from /Users/davidmann/w/haiku2-release/haiku2/build/haiku2-unsigned.bin
[1400/1400] cd /Users/davidmann/w/haiku2-release/haiku2/build/esp-idf/esptool_py && /Use...tion_table/partition-table.bin /Users/davidmann/w/haiku2-release/haiku2/build/haiku2.bin
haiku2.bin binary size 0x451000 bytes. Smallest app partition is 0x500000 bytes. 0xaf000 bytes (14%) free.

Project build complete. To flash, run this command:
/Users/davidmann/.espressif/python_env/idf4.4_py3.9_env/bin/python ../../../esp/esp-idf-v4.4.2/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset --chip esp32s3 --no-stub write_flash --flash_mode dio --flash_size keep --flash_freq 80m 0xc000 build/partition_table/partition-table.bin 0x19000 build/ota_data_initial.bin 0x20000 build/haiku2.bin
or run 'idf.py -p (PORT) flash'
davidmann@Davids-Macbook haiku2 % idf.py --port /dev/cu.usbserial-110 flash -b 1500000 monitor
Executing action: flash
Running ninja in directory /Users/davidmann/w/haiku2-release/haiku2/build
Executing "ninja flash"...
[1/5] cd /Users/davidmann/w/haiku2-release/haiku2/build/esp-idf/esptool_py && /Users/dav...tion_table/partition-table.bin /Users/davidmann/w/haiku2-release/haiku2/build/haiku2.bin
haiku2.bin binary size 0x451000 bytes. Smallest app partition is 0x500000 bytes. 0xaf000 bytes (14%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/esp-idf/esptool_py && /Users/davidmann/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/davidmann/esp/esp-idf-v4.4.2/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x0 /Users/davidmann/w/haiku2-release/haiku2/build/bootloader/bootloader.bin
Bootloader binary size 0xa000 bytes. 0x2000 bytes (17%) free.
[2/3] cd /Users/davidmann/esp/esp-idf-v4.4.2/components/esptool_py && /opt/homebrew/Cell...uild" -P /Users/davidmann/esp/esp-idf-v4.4.2/components/esptool_py/run_serial_tool.cmake
esptool.py esp32s3 -p /dev/cu.usbserial-110 -b 1500000 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x20000 haiku2.bin 0xc000 partition_table/partition-table.bin 0x19000 ota_data_initial.bin
esptool.py v3.3.2
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
[... 74 lines cut ...]
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3970,len:0x2604
load:0x403c9700,len:0xcc4
load:0x403cc700,len:0x4d84
SHA-256 comparison failed:
Calculated: 75267704ce9a3f91b5be85dcf5669a015759d3683ff7e3d6d86e11124b13aa70
Expected: f1354bfaeb76fdbdf87fcce2accbb10a850ba5ae76ada3661fba9e7a86a62a5b
Attempting to boot anyway...
entry 0x403c9928
Sig block 0 invalid: Image digest does not match
E (943) secure_boot_v2: Secure Boot V2 verification failed.
E (943) esp_image: Secure boot signature verification failed
E (952) esp_image: Image hash failed - image is corrupt
E (954) secure_boot_v2: bootloader image appears invalid! error 8194
E (962) boot: Secure Boot v2 failed (8194)
E (966) boot: Factory app partition is not bootable
E (972) esp_image: image at 0x520000 has invalid magic byte (nothing flashed here?)
E (980) boot: OTA app partition slot 0 is not bootable
E (986) esp_image: image at 0xa20000 has invalid magic byte (nothing flashed here?)
E (994) boot: OTA app partition slot 1 is not bootable
E (1000) boot: No bootable app partitions in the partition table
```

KonstantinKondrashov, 2023-01-23:
Hi @davidallenmann!

You get the issue `SHA-256 comparison failed` because your settings in sdkconfig and  `esptool.py write_flash  ....bootloader.bin` are different. The bootloader image is overwritten by the given options (--flash_mode dout --flash_freq 80m --flash_size 16MB) during writing (note that, the bootloader file is not changed). Please make sure that ESPTOOLPY_FLASHMODE ESPTOOLPY_FLASHFREQ ESPTOOLPY_FLASHSIZE are set as required. 

This log is from ROM bootloader (it says that the 2nd stage bootloader image was changed):
```
SHA-256 comparison failed:
Calculated: 75267704ce9a3f91b5be85dcf5669a015759d3683ff7e3d6d86e11124b13aa70
Expected: f1354bfaeb76fdbdf87fcce2accbb10a850ba5ae76ada3661fba9e7a86a62a5b
```

When your sdkconfig options are not consistent with --flash_... options then the esptool overwrites the header of the bootloader but does not update the sha256 at the end of the image and signatures. So it makes secure boot not workable.
```
python  esptool.py --chip esp32s3 --port=/dev/cu.usbserial-110 --baud=460800 \
--before=default_reset  --after=no_reset --no-stub \
write_flash \
--flash_mode dout \
--flash_freq 80m \
--flash_size 16MB \
0x0 bootloader.bin
```

When the image flash configs are overwritten then you see this log: `Flash params set to 0x034f`

KonstantinKondrashov, 2023-01-24:
I am glad to hear that you managed to get it workable.
(I suspect that the original issue was that the bootloader was overwritten by the plain-text image when FE was already enabled.)

closed 2023-01-24 by KonstantinKondrashov