espressif/esp-idf #10539: HTTP requests are exhausting heap memory.

gasaimaple, 2023-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

No heap size difference before making http request and after esp_http_client_cleanup.

### What is the actual behavior?

Free heap size gets smaller and smaller after every http request.
Like LOG below.
I (5434) HTTP_CLIENT: Connected to AP, begin http example
I (7534) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (7544) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (7544) HTTP_CLIENT: Free space: 234940
I (8974) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (8974) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (8974) HTTP_CLIENT: Free space: 234728
I (9784) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (9794) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (9794) HTTP_CLIENT: Free space: 234512
I (10614) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (10614) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (10614) HTTP_CLIENT: Free space: 234308
I (11324) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (11324) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (11324) HTTP_CLIENT: Free space: 233900
I (13064) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (13074) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (13074) HTTP_CLIENT: Free space: 233884
I (13784) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (13784) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (13784) HTTP_CLIENT: Free space: 233464
I (14604) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (14604) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (14604) HTTP_CLIENT: Free space: 233460
I (18904) HTTP_CLIENT: HTTP GET Status = 200, content_length = 291
I (18904) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (18904) HTTP_CLIENT: Free space: 233048

### Steps to reproduce.

1. Create a new project with esp_http_client_example.
2. Delete every method after the GET method in http_rest_with_url.
3. Loop http_rest_with_url in http_test_task and log free heap size after every call.

igrr, 2023-01-13:
@gasaimaple could you check, does the heap size stabilize after running the example for some time, or does it keep declining even after long time? If you break out of the loop and just print the heap size every now and then without doing more requests, does the heap size recover?

If yes, then this might indicate that the heap usage you are seeing is related to sockets left in TIME_WAIT state. By default, MSL is set to 60 seconds, and the socket is kept in the TIME_WAIT state by the TCP stack for 2xMSL interval.

gasaimaple, 2023-01-14:
> @gasaimaple could you check, does the heap size stabilize after running the example for some time, or does it keep declining even after long time? If you break out of the loop and just print the [...]

Print out heap size does not affect heap size. Heap size do recover after 120 seconds.        
But There's still one question: Why is the heap usage per request so random?

closed 2023-01-14 by gasaimaple