espressif/esp-idf #10540: IP101PHY loss of ethernet packets when using internal clock sourced from GPIO0

NielsDeMoor, 2023-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-Ethernet-Kit V1.2 

### Power Supply used.

USB

### What is the expected behavior?

Since 2 years we use the ESP32 (custom board) with an IP101GRI as an ethernet phy to act as a smarthome-gateway. At that time we were at ESP-IDF v4.2.4. No issues occured. 
The 50MHz clock is sourced internally via GPIO0. 
When we for example execute ping commands to the ESP, it often fails and raises a time-out. 

To be sure it wasn't a hardware issue, we tested the same program with the ESP32-Ethernet-Kit V1.2, same issue occured here. 
Also the basic ethernet example, provided by ESP-IDF raised the same problem. 

I think the issue might be related to this commit: [esp_eth: IEEE 802.3 PHY MII Management Interface functionality][(https://github.com/espressif/esp-idf/commit/43f3904304c3761cebb6e145645b7a3c7371a871) 43f3904304c3761cebb6e145645b7a3c7371a871 ](https://github.com/espressif/esp-idf/commit/43f3904304c3761cebb6e145645b7a3c7371a871)

### What is the actual behavior?

The ESP doesn't seem to receive every ethernet packet. In whireshark we see a lot of retransmissions. 
The serial output doesn't show errors. When we enable LWIP debug, the messages don't appaer. I think this means it's an issue, deeper inside the libraries. 

### Steps to reproduce.

1. Use ESP-IDF V5.0 or higher.
2. Use ethernet example board or custom board with IP101 and ESP32.
3. Make sure the PHY clock is sourced internally via GPIO0.
4. Program the board with the basic ethernet example.
5. Once it is running and got an IP-address, try to ping the IP.

### Debug Logs.

```plain
No logs
```

### More Information.

/

kostaond, 2023-01-16:
@NielsDeMoor thanks for reporting potential issue.
Could you please share more about your HW configuration:

> Make sure the PHY clock is sourced internally via GPIO0

Just for sure, does it mean to generate  RMII CLK in the ESP32 and output it via GPIO0, right? If so, did you modify the ESP32-Ethernet-Kit as stated [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html#rmii-clock-sourced-internally-from-esp32-s-apll)? It means:
 * X1 to GND, 
 * de-solder R6 and R9, 
 * solder U2, R4

I've already observed similar behavior. I tested multiple Ethernet boards with different PHYs and different CLK configurations. At certain point, I incorrectly configured RMII CLK to be internally generated by ESP32 but used Ethernet board where the CLK was also sourced externally (via PYH's CLKO pin). The setup kind of worked, it got the IP but there was high packet loss when ran ping. Exactly as you described. It worked only by coincident when the two separate CLK sources got synced time to time.

The commit you linked was non-functional update. Only structure of PHY driver was refactored to reuse common code and so make potential updates very simple (e.g. when adding support for new PHY chip).

NielsDeMoor, 2023-01-23:
@kostaond Thanks for answering to my question. 

Yes, the RMII CLK is sourced from the ESP32 via GPIO0. 

I did not modify the ethernet kit as it was working with the previous ESP-IDF Framework (4.2.47). 
Since upgrading to the newest 5.1 version, both my hardware and the development kit stopped working (packet loss). 

Are you able to dig deeper into the problem or let me know if I could help with something? 

Many thanks in advance,

Niels

kostaond, 2023-01-23:
> I did not modify the ethernet kit as it was working with the previous ESP-IDF Framework (4.2.47).

It worked only by luck as I mentioned above.

> Are you able to dig deeper into the problem or let me know if I could help with something?

This seems to be HW issue. Could you please modify the Ethernet kit and double check the design of your custom board? As I said, I've already observed exactly the same behavior when RMII CLK was not properly handled.

NielsDeMoor, 2023-01-23:
Ok I understand. I will try and change the HW (both on DK and own hardware). 

We are aware of the possible HW problem, so we will try to fix that.
But, I still think there might be a change in FW that made this issue rise up. Since 1 year we manufactured 2000pcs with the 4.2.47 version and none of them had this issue. Since upgrading the framework every board has the problem. 

Maybe a drive strength change of the GPIO's used for the PHY? Only trying to find the underlaying reason of the sudden issue.

kostaond, 2023-01-23:
Frankly speaking, I don't know why it used to work. However, note that using GPIO0 as output of RMII REF CLK is experimental feature as is stated in the Ethernet configuration:

```
GPIO0 can be set to output a pre-divided PLL clock (test only!).
Enabling this option will configure GPIO0 to output a 50MHz clock.
In fact this clock doesn't have directly relationship with EMAC peripheral.
Sometimes this clock won't work well with your PHY chip. You might need to
add some extra devices after GPIO0 (e.g. inverter).
Note that outputting RMII clock on GPIO0 is an experimental practice.
If you want the Ethernet to work with WiFi, don't select GPIO0 output mode for stability.
```

oywino, 2023-01-31:
Could this be a related issue; When flashing ESPHome v12.x firmware to a GL-S10 ESP32 based v2.x board, we experience severe packet losses on the IP101GRI based Ethernet port. This happens whether the Arduino or the IDF Framework is used.

kostaond, 2023-02-01:
@oywino interesting... Does Ethernet work correctly with different ESP-IDF versions? For example, it seems the ESPHome v12 recommended IDF version is [4.4.2](https://github.com/esphome/esphome/blob/2022.12.8/esphome/components/esp32/__init__.py#L141). Could you please try different version, e.g. 4.2?  Would it be also possible to flash `ethernet\basic` example to the GL-S10 and use different ESP-IDF versions? It seems some users have the [same issue](https://github.com/esphome/bluetooth-proxies/issues/21#issuecomment-1253893065).

oywino, 2023-02-01:
@kostaond Regretfully I need [How-To] guidance to build my own bin files using different ESP-IDF versions, including how to select `ethernet/basic` 
I can confirm that many users have this problems. I've found them on several places here at GitHUB and on Reddit.

NielsDeMoor, 2023-02-01:
In the meantime I have looked further into the issue and found out that the issue disappeared when I disconnected the programmer. In the ESP/IP101 reference designs, the RTS line is shared with the GPIO0/50MHzClock. 
When the programmer is disconnected, the GPIO0 can source the clock from internal without any problem. 
Whenever the programmer is connected and the serial monitor is opened via Visual Code, the RTS line is not in tri-state but pulled low I think. This obstructs the ESP from generating a decent 25MHz clock. 

To make it work with my current hardware I made following changes to the file: _**../esp-idf/tools/idf_monitor_base/serial_reader.py**_

line 63: made sure both RTS and DTR are low after reset

`def run(self):
        #  type: () -> None
        if not self.serial.is_open:
            self.serial.baudrate = self.baud
            # We can come to this thread at startup or from external application line GDB.
            # If we come from GDB we would like to continue to run without reset.

            high = False
            low = True

            self.serial.dtr = low      # Non reset state
            self.serial.rts = high     # IO0=HIGH
            self.serial.dtr = self.serial.dtr   # usbser.sys workaround
            # Current state not reset the target!
            self.serial.open()
            if not self.gdb_exit and self.reset:
                self.serial.dtr = high     # Set dtr to reset state (affected by rts)
                self.serial.rts = low      # Set rts/dtr to the reset state
                self.serial.dtr = self.serial.dtr   # usbser.sys workaround

                # Add a delay to meet the requirements of minimal EN low time (2ms for ESP32-C3)
                time.sleep(MINIMAL_EN_LOW_DELAY)
            elif not self.reset:
                self.serial.setDTR(high)  # IO0=HIGH, default state
            self.gdb_exit = False
            self.serial.rts = low             # Set rts/dtr to the working state --> to low
            self.serial.dtr = low   # usbser.sys workaround --> to low `

At this moment, I don't know how this is related to the upgrade ESP-IDF framework.. but wanted to share the solution for our hardware.

kostaond, 2023-02-02:
@NielsDeMoor nice observation. At this point I would like to also share one of my similar observations. One could find it useful... It is related to boards which use external programmer for flashing. If you have such configuration, Ethernet REF RMII CLK at GPIO0 and you connect the programmer via longer wires, it can affect impedance of CLK PCB paths and in turn break signal integrity of the clock. The result is the Ethernet does not work reliably or not at all.

kostaond, 2023-02-03:
> At this moment, I don't know how this is related to the upgrade ESP-IDF framework.. but wanted to share the solution for our hardware.

If you check the history of  `serial_reader.py`, there were some updates related do `rst` and `dtr` handling. However, it work properly with our reference dev boards. Is your HW the same in this manner or do you use some specific configuration?

Alvin1Zhang, 2023-02-15:
@NielsDeMoor ​Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

kostaond, 2024-09-30:
Closing the issue since we were not able to reproduce. Feel free to reopen or open a new issue if you have any more detailed update.

closed 2024-09-30 by kostaond