espressif/esp-idf #10541: flash storage - Porting to IDF V5.0

AchimPieters, 2023-01-13:
Hello all,

I'm trying to port a piece of code to IDF 5.0. It, works kind of, but after compiling I get the error "Failed to read" and "Failed to write" And I'm a little bit lost, does anyone have the solution to my problem?

Old code:
```

#include <string.h>
#include <ctype.h>
#include "constants.h"
#include "debug.h"
#include "crypto.h"
#include "pairing.h"
#include "port.h"

#include "storage.h"

#pragma GCC diagnostic ignored "-Wunused-value"

#ifndef SPIFLASH_BASE_ADDR
#define SPIFLASH_BASE_ADDR 0x200000
#endif

#define MAGIC_OFFSET           0
#define ACCESSORY_ID_OFFSET    4
#define ACCESSORY_KEY_OFFSET   32
#define PAIRINGS_OFFSET        128

#define MAGIC_ADDR           (SPIFLASH_BASE_ADDR + MAGIC_OFFSET)
#define ACCESSORY_ID_ADDR    (SPIFLASH_BASE_ADDR + ACCESSORY_ID_OFFSET)
#define ACCESSORY_KEY_ADDR   (SPIFLASH_BASE_ADDR + ACCESSORY_KEY_OFFSET)
#define PAIRINGS_ADDR        (SPIFLASH_BASE_ADDR + PAIRINGS_OFFSET)

#define MAX_PAIRINGS 16

#define ACCESSORY_KEY_SIZE  64

const char magic1[] = "HAP";

int homekit_storage_init() {
    char magic[sizeof(magic1)];
    memset(magic, 0, sizeof(magic));

    if (!spiflash_read(MAGIC_ADDR, (byte *)magic, sizeof(magic))) {
        ERROR("Failed to read HomeKit storage magic");
    }

    if (strncmp(magic, magic1, sizeof(magic1))) {
        INFO("Formatting HomeKit storage at 0x%x", SPIFLASH_BASE_ADDR);
        if (!spiflash_erase_sector(SPIFLASH_BASE_ADDR)) {
            ERROR("Failed to erase HomeKit storage");
            return -1;
        }

        strncpy(magic, magic1, sizeof(magic));
        if (!spiflash_write(MAGIC_ADDR, (byte *)magic, sizeof(magic))) {
            ERROR("Failed to write HomeKit storage magic");
            return -1;
        }

        return 1;
    }

    return 0;
}

int homekit_storage_reset() {
    byte blank[sizeof(magic1)];
    memset(blank, 0, sizeof(blank));

[... 692 lines cut ...]
    return -1;
}
```

After compiling I get this error:

```
>>> HomeKit: Starting server
!!! HomeKit: Failed to read HomeKit storage magic
>>> HomeKit: Formatting HomeKit storage at 0x200000
!!! HomeKit: Failed to write HomeKit storage magic
>>> HomeKit: Resetting HomeKit storage
!!! HomeKit: Failed to reset HomeKit storage
>>> HomeKit: Generated new accessory ID: 29:40:D3:4A:98:53
!!! HomeKit: Failed to write accessory ID to HomeKit storage
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to HomeKit storage
>>> HomeKit: Configuring mDNS
```

Does anyone have a solution for this problem? I would be really appreciated!

Total Source code: https://github.com/AchimPieters/esp32-homekit-demo

Thanks in advance!

igrr, 2023-01-13:
@AchimPieters here are a couple things to check:

1. Please try modifying the code to log the error code, like this. That will help understand where the issue might be.
   ```c++
    esp_err_t err = esp_flash_read(NULL,(byte *)magic, MAGIC_ADDR, sizeof(magic));
    if (err != ESP_OK) {
        ERROR("Failed to read HomeKit storage magic, error %s (0x%x)", esp_err_to_name(err), err);
    }
   ```

2. What is the flash size selected in menuconfig, under Serial Flasher Config menu? The default value is 2 MB, but it looks like you are trying to access data at ESPFLASH_BASE_ADDR = 0x200000, i.e. above the 2MB boundary. That should result in `ESP_ERR_INVALID_ARG` return code from `esp_flash_read`.

AchimPieters, 2023-01-14:
@igrr Foremost, thank you very much for your comment! Sometimes you are so busy with something that you can no longer see the wood for the trees. And then it is nice that someone is watching with you who is still clear-headed.

I tried both options, the 2 MB boundary was not the solution. In addition, the adjustment of the code also did not provide more insight, this is what I got after adjustment:

```
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x200000
!!! HomeKit: Failed to write HomeKit storage magic
>>> HomeKit: Resetting HomeKit storage
!!! HomeKit: Failed to reset HomeKit storage
>>> HomeKit: Generated new accessory ID: FA:83:A4:C1:BE:8D
!!! HomeKit: Failed to write accessory ID to HomeKit storage
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to HomeKit storage
>>> HomeKit: Configuring mDNS
```

igrr, 2023-01-14:
> Failed to write accessory ID to HomeKit storage

What is the error code (esp_err_t) returned by `esp_flash*` API in this case? (See my previous comment on logging the error codes.)

AchimPieters, 2023-01-14:
@igrr What I did next is changed all the lines:

```
int homekit_storage_init() {
    char magic[sizeof(magic1)];
    memset(magic, 0, sizeof(magic));

    //esp_err_t esp_flash_read(esp_flash_t *chip, void *buffer, uint32_t address, uint32_t length)
    //if (!esp_flash_read(NULL,(byte *)magic, MAGIC_ADDR, sizeof(magic))) {
    //    ERROR("Failed to read HomeKit storage magic");
    //}
    esp_err_t err_read = esp_flash_read(NULL,(byte *)magic, MAGIC_ADDR, sizeof(magic));
     if (err_read != ESP_OK) {
         ERROR("Failed to read HomeKit storage magic, error %s (0x%x)", esp_err_to_name(err_read), err_read);
     }

    if (strncmp(magic, magic1, sizeof(magic1))) {
        INFO("Formatting HomeKit storage at 0x%x", ESPFLASH_BASE_ADDR);
        //esp_err_t esp_flash_erase_region(esp_flash_t *chip, uint32_t start, uint32_t len)
        //if (!esp_flash_erase_region(NULL, ESPFLASH_BASE_ADDR, sizeof(magic1))) {
        //    ERROR("Failed to erase HomeKit storage");
        esp_err_t err_erase = esp_flash_erase_region(NULL, ESPFLASH_BASE_ADDR, sizeof(magic));
         if (err_erase != ESP_OK) {
             ERROR("Failed to erase HomeKit storage, error %s (0x%x)", esp_err_to_name(err_erase), err_erase);
            return -1;
        }

        strncpy(magic, magic1, sizeof(magic));
        //esp_err_t esp_flash_write(esp_flash_t *chip, const void *buffer, uint32_t address, uint32_t length)
        //if (!esp_flash_write(NULL, (byte *)magic, MAGIC_ADDR, sizeof(magic))) {
        //    ERROR("Failed to write HomeKit storage magic");
        esp_err_t err_write = esp_flash_write(NULL,(byte *)magic, MAGIC_ADDR, sizeof(magic));
         if (err_write != ESP_OK) {
             ERROR("Failed to write HomeKit storage magic, error %s (0x%x)", esp_err_to_name(err_write), err_write);
            return -1;
        }

        return 1;
    }

    return 0;
}
```
 And this give me an error:
 
```
 >>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x200000
!!! HomeKit: Failed to erase HomeKit storage, error ERROR (0x1)
>>> HomeKit: Resetting HomeKit storage
!!! HomeKit: Failed to reset HomeKit storage
>>> HomeKit: Generated new accessory ID: B5:18:D7:94:1D:0B
!!! HomeKit: Failed to write accessory ID to HomeKit storage
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to HomeKit storage
>>> HomeKit: Configuring mDNS

```

Now looking into this **error ERROR (0x1)** to see what it means.

igrr, 2023-01-14:
That looks quite odd, 0x1 is not one of standard error codes defined in ESP-IDF. Could you please give the instructions how to reproduce this issue using your repository? It seems to have several projects, which one do we need to run and what are the steps to do this?

Also, please add the missing information from the issue template: which chip or module you are using, what is your compilation environment and what is your sdkconfig (if different from the default one).

AchimPieters, 2023-01-14:
@igrr all information can be found here: https://github.com/AchimPieters/esp32-homekit-demo

Reproduction steps are also included.

igrr, 2023-01-17:
@AchimPieters It seems that the esp32-homekit-demo repository still points to the esp32-homekit submodule which doesn't have the error checks around the esp_flash API calls ([here](https://github.com/AchimPieters/esp32-homekit/blob/72923ec061738cf5aa3fdb1023c164e72bee0e4b/src/storage.c#L39-L59)). Have you committed the changes from your comment above into the repository?

AchimPieters, 2023-01-21:
@igrr I took a deep dive and googled everything I could about the 0x1 error notification. The 0x1 is not one of the standard error codes defined in ESP-IDF. Have you become wiser in the meantime?

igrr, 2023-01-21:
I'm sorry @AchimPieters, i haven't looked into this in more detail. I have labelled this issue as "has enough information" and it should eventually get picked up by the team from the backlog.

(I don't have a clue where the 0x1 error code comes from, either! That's not something I have seen in IDF yet.)

AchimPieters, 2023-02-09:
fixed see https://github.com/AchimPieters/esp32-homekit

closed 2023-02-09 by AchimPieters