espressif/esp-idf #10542: Failed to initlialize/mount 32GB and 64GB sdcards

listout, 2023-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-DevKitM-2

### Power Supply used.

USB

### What is the expected behavior?

Mout/initialize sd card successfully.

### What is the actual behavior?

Cannout mout/initialize SD card. Initialization fails with `sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107`.

### Steps to reproduce.

1. From tag v5.0 build the sdspi example
2. Correct appt. pins to sdcard module
3. Observe failure

### Debug Logs.

```plain
I (30) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (30) boot: compile time 23:47:53
I (30) boot: chip revision: v3.0
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0c7e4h ( 51172) map
I (131) esp_image: segment 1: paddr=0001c80c vaddr=3ffb0000 size=02800h ( 10240) load
I (135) esp_image: segment 2: paddr=0001f014 vaddr=40080000 size=01004h (  4100) load
I (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=24a08h (150024) map
I (200) esp_image: segment 4: paddr=00044a30 vaddr=40081004 size=0c650h ( 50768) load
I (221) esp_image: segment 5: paddr=00051088 vaddr=50000000 size=00010h (    16) load
I (228) boot: Loaded app from partition at offset 0x10000
I (228) boot: Disabling RNG early entropy source...
I (241) cpu_start: Pro cpu up.
I (241) cpu_start: Starting app cpu, entry point is 0x40081194
0x40081194: call_start_cpu1 at /home/listout/esp/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (255) cpu_start: Pro cpu start user code
I (255) cpu_start: cpu freq: 160000000 Hz
I (255) cpu_start: Application information:
I (260) cpu_start: Project name:     sd_card
I (265) cpu_start: App version:      1
I (269) cpu_start: Compile time:     Jan 13 2023 23:56:38
I (275) cpu_start: ELF file SHA256:  bd976416abda1b1c...
I (281) cpu_start: ESP-IDF:          v5.0-dirty
I (287) heap_init: Initializing. RAM available for dynamic allocation:
I (294) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (300) heap_init: At 3FFB3188 len 0002CE78 (179 KiB): DRAM
I (306) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (312) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (319) heap_init: At 4008D654 len 000129AC (74 KiB): IRAM
I (326) spi_flash: detected chip: generic
I (330) spi_flash: flash io: dio
W (334) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (348) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (358) example: Initializing SD card
I (358) example: Using SPI peripheral
I (368) example: Mounting filesystem
I (368) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (418) sdspi_transaction: cmd=5, R1 response: command not supported
E (1418) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1418) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (1418) gpio: GPIO[20]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1428) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
```
```

### More Information.

SD card used was SanDisk Ultra microSDHC UHS-I card of 32 GB, same issue with 64 GB. Adapter was just a generic micro sdcard adapter found on Amazon.

![WhatsApp Image 2023-01-13 at 5 05 09 PM](https://user-images.githubusercontent.com/66715002/212394446-781e6213-1545-4de3-99cd-f026b666558d.jpeg)

igrr, 2023-01-13:
@listout Here are a few things you can do to troubleshoot this:

- Which GPIOs are you using to connect the card?
- Do any other SD cards work with the same software version and hardware setup?
- What is the log output if you set the log level to Verbose in menuconfig? (Please don't paste it inline, instead attach as a file or post a link to a gist.)
- You can check the data exchanged between the ESP and the card using a logic analyzer, check if the card is actually sending the response to the last command.

listout, 2023-01-13:
1. Here are the gpio defs for the dev board I'm using
```
#define PIN_NUM_MISO 12
#define PIN_NUM_MOSI 13
#define PIN_NUM_CLK 14
#define PIN_NUM_CS 20
```
2. Yes. I've other 16GB sdcards that work fine with same code. This the first 64 GB sdcard I'm trying
3. http://ix.io/4l6H
4. Alright, i'll try that

listout, 2023-01-14:
Hey @igrr any ideas/clues looking at the logs?

igrr, 2023-01-14:
There can be multiple reasons for the issue at this stage of card initialization process, most common ones are:
* timing and signal integrity — even under the same hardware setup, different cards have varying timing margins
* insufficient or unstable power supply (https://github.com/espressif/esp-idf/issues/6686#issuecomment-809938874)
* actual incompatibility between our sdmmc driver and the card (https://github.com/espressif/esp-idf/issues/6686)

To figure out which of these is the case we need to know whether
* the card gives some response, but the driver fails to read it properly
* or the card doesn't give any response

Usually the easiest way is to check that using a logic analyzer. If you have access to a known-good hardware (like ESP32-WROVER-KIT board), testing the card with that hardware can also provide a clue.

listout, 2023-01-21:
@igrr I've managed to get my hands on a Logic Analyzer. I'm not sure how to share the data with you. I'm using salae software where I can either export as csv or save the session. I'm attaching SS and save session for a working vs non-working card. Hope this helps in debugging.

Channel 1 was my MOSI and 2 was MISO

**Not working Card SS**

[image]

**Working Card SS**
[image]

Saved Sessions: [saves.zip](https://github.com/espressif/esp-idf/files/10472433/saves.zip)

ilxj, 2024-02-23:
**I also encountered the same problem. Any update about this?**

closed 2024-03-04 by espressif-bot