espressif/esp-idf #10544: Inefficient build with vscode ide,because it can't use all hyperthreading

Tilie, 2023-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4 5.0

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

it will successfully use all threading cores when using vscode with esp_idf expansion, building in terminal attach vscode

### What is the actual behavior?

only use the physical core but ignore the virtual core.
this phenomenon didn't occur in macOS terminal 

### Steps to reproduce.

use build command in vscode terminalthe

brianignacio5, 2023-01-16:
As described in the extension [basic use tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/3e8c1d04a698d43495cf69b0d7f70e597ab00e30/docs/tutorial/basic_use.md) step 7 you can modify the `idf.ninjaArgs` configuration setting in your project settings.json to use several jobs in parallel.

Tilie, 2023-01-16:
@brianignacio5 I have tried these args already, but it still doesn't work. actually, I check these args' usage, and it defaults to use all thread cores the PC has.

brianignacio5, 2023-01-17:
The thing is, I don't think the issue is related to vscode or any IDE on itself. If you notice, there is a cmake and ninja process running as a background process. In visual studio code, you can define which terminal executable to use in the IDE, so maybe the terminal you are using is different from your Mac terminal.

Tilie, 2023-01-17:
@brianignacio5 I check my setting of vs code, and confirm the  terminal executable to use in the IDE is macOS buildin terminal

brianignacio5, 2023-01-18:
Then could you show which arguments are you using when building from macOS terminal vs vscode terminal? The running code should be the same.

Tilie, 2023-01-18:
yes, I think it should be the same too, which argument do you need?

Tilie, 2023-01-18:
however, I don't think this is a big problem, just let someone like me who notices this phenomenon, getting a solution from the internet is enough

brianignacio5, 2023-01-20:
when you run the build from terminal, what command do you use ? `idf.py build` or directly with `cmake <args>` ?

Tilie, 2023-01-20:
in the system buildin terminal, I used `idf.py build`. in vscode you know like the tutorial does, press F1 and choose "ESP-IDF: Build your project"

closed 2025-04-09 by Alvin1Zhang