espressif/esp-idf #10548: Unable to compile normally after installing idf

yelvlab, 2023-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v4.4.3 & idf v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32s3 custom

### Power Supply used.

USB

### What is the expected behavior?

I used the same deployment method and the same board on other computers, but I went home from vacation recently, and I had some problems with my laptop. I tried to redeploy the environment, and a new problem appeared

### What is the actual behavior?

PS E:\esp-idf\blink> idf.py menuconfig
Executing action: menuconfig
Running cmake in directory e:\esp-idf\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 e:\esp-idf\blink"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
CMake Error at D:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/build.cmake:452 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "E:/esp-idf/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

### Steps to reproduce.

1. download ESP-IDF windows installer
2. instal it
3. copy sample project
4. idf.py menuconfig(some error)
...

jeckorf, 2023-02-09:
same problem,did you solved?

ligantx, 2023-03-06:
sometimes i have the same problem. Any solution?

Henry586, 2023-04-27:
need to set proxy for cmake, for exmaple:
export https_proxy=http://127.0.0.1:33210 http_proxy=http://127.0.0.1:33210 all_proxy=socks5://127.0.0.1:33211

zikalino, 2023-12-13:
This seems like connectivity error unrelated to ESP IDF, so I am closing this issue for now.

closed 2023-12-13 by zikalino