espressif/esp-idf #10549: esp32c6 uses the itwt example to run for half an hour and the watchdog triggers

dreamcmi, 2023-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2889-gdb9586c53f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

normal standby operation

### What is the actual behavior?

After running for half an hour, the idle task watchdog triggers

### Steps to reproduce.

1. idf.py flash
2. wait for half an hour
3. idle watchdog triggers

[image]

[itwt.zip](https://github.com/espressif/esp-idf/files/10420142/itwt.zip)

### Build or installation Logs.

```plain
[19:23:47.108] ESP-ROM:esp32c6-20220919
[19:23:47.108] Build:Sep 19 2022
[19:23:47.129] rst:0xc (SW_CPU),boo
[19:23:47.108] t:0xc (SPI_FAST_FLASH_BOOT)
[19:23:47.108] Saved PC:0x4001975a
[19:23:47.108] SPIWP:0xee
[19:23:47.129] mode:DIO, clock di
[19:23:47.108] v:1
[19:23:47.129] load:0x4086c610,le
[19:23:47.108] n:0xc64
[19:23:47.129] load:0x4086e610,le
[19:23:47.108] n:0x2ae4
[19:23:47.129] load:0x40875720,le
[19:23:47.108] n:0x1914
[19:23:47.108] entry 0x4086c610
[19:23:47.137] I (27) boot: ESP-IDF v5.1-dev-2889-gdb9586c53f 2nd stage bootloader
[19:23:47.163] I (28) boot: compile time Jan 15 2023 17:1
[19:23:47.137] 5:01
[19:23:47.163] I (28) boot: chip revisio
[19:23:47.137] n: v0.0
[19:23:47.137] I (32) qio_mode: Enabling default flash chip QIO
[19:23:47.163] I (37) boot.esp32c6: SPI Speed     
[19:23:47.137]  : 80MHz
[19:23:47.163] I (42) boot.esp32c6: SPI Mode      
[19:23:47.137]  : QIO
[19:23:47.184] I (47) boot.esp32c6: SPI Flash Size
[19:23:47.164]  : 8MB
[... 309 lines cut ...]
[19:57:12.270] g:
[19:57:12.303] E (2005157) task_wdt: CPU 
[19:57:12.270] 0: wifi
[19:57:12.270] E (2005157) task_wdt: Print CPU 0 (current core) registers
[19:57:12.346] Core 00 register dump:
[19:57:12.382] MEPC    : 0x40809116  RA      : 0x40809142  SP      : 0x408217a0  GP     
[19:57:12.346]  : 0x40810b20  
[19:57:12.382] TP      : 0x407fd444  T0      : 0x40022494  T1      : 0xffffffe0  T2     
[19:57:12.346]  : 0x00000007  
[19:57:12.382] S0/FP   : 0x4087c610  S1      : 0x40880000  A0      : 0x00000001  A1     
[19:57:12.346]  : 0x00000738  
[19:57:12.382] A2      : 0x00000000  A3      : 0x4087c644  A4      : 0x4081a000  A5     
[19:57:12.346]  : 0x20001000  
[19:57:12.382] A6      : 0x4004a590  A7      : 0x0000002a  S2      : 0x00000000  S3     
[19:57:12.346]  : 0x00000738  
[19:57:12.417] S4      : 0x000006a4  S5      : 0x00000000  S6      : 0x40880000  S7     
[19:57:12.384]  : 0x00000001  
[19:57:12.417] S8      : 0x40822bf0  S9      : 0x00000000  S10     : 0x00000000  S11    
[19:57:12.384]  : 0x00000000  
[19:57:12.417] T3      : 0x80000000  T4      : 0x01000406  T5      : 0x00080100  T6     
[19:57:12.384]  : 0x06080000  
[19:57:12.417] MSTATUS : 0x40821b08  MTVEC   : 0x40880000  MCAUSE  : 0x4087e9f4  MTVAL  
[19:57:12.384]  : 0x4080a84a  
[19:57:12.384] MHARTID : 0x40880000
```

ESP-iPENCIL, 2023-01-16:
The issue is caused by some system modules are not fullfilled.

https://gitlab.espressif.cn:6688/espressif/esp-idf/-/merge_requests/21985

This MR can fix the issue.

ESP-iPENCIL, 2023-01-16:
We are aware of the issue and the fix is already in progress.

dreamcmi, 2023-02-24:
Hi @ESP-iPENCIL ,is there any progress now?

xuxiao111, 2023-02-28:
Hi @dreamcmi, we have already fixed this issue but still need some time to commit the code changes.

AxelLin, 2023-03-19:
> Hi @dreamcmi, we have already fixed this issue but still need some time to commit the code changes.

@xuxiao111 @ESP-iPENCIL How is the status?

xuxiao111, 2023-04-26:
@AxelLin @dreamcmi This issue has been fixed and the code has been committed, please sync the latest code.

dreamcmi, 2023-04-27:
> @AxelLin @dreamcmi This issue has been fixed and the code has been committed, please sync the latest code.

I have tested it, it works fine.Thanks!

closed 2023-04-27 by dreamcmi