espressif/esp-idf #10550: cant flash and monitor with v5.0 base ubuntu 22.04

yelvlab, 2023-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

idf.py flash monitor -p /dev/ttyUSB0 
but some error about ninja ?

### What is the actual behavior?

hulk@hulk-virtual-machine:~/code/blink$ idf.py flash -p /dev/ttyUSB0
Executing action: flash
Running ninja in directory /home/hulk/code/blink/build
Executing "ninja flash"...
[1/5] cd /home/hulk/code/blink/build/esp-idf/esptool_py && /home/hulk/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/hulk/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/hulk/code/blink/build/partition_table/partition-table.bin /home/hulk/code/blink/build/blink.bin
blink.bin binary size 0x37cb0 bytes. Smallest app partition is 0x100000 bytes. 0xc8350 bytes (78%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/hulk/code/blink/build/bootloader/esp-idf/esptool_py && /home/hulk/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/hulk/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/hulk/code/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x5150 bytes. 0x2eb0 bytes (36%) free.
[2/3] cd /home/hulk/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/hulk/esp/esp-idf -D "SERIAL_TOOL=/home/hulk/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/hulk/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/hulk/code/blink/build -P /home/hulk/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.4
Serial port /dev/ttyUSB0

A fatal error occurred: Could not open /dev/ttyUSB0, the port doesn't exist
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/hulk/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/hulk/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash /home/hulk/code/blink/build/CMakeFiles/flash 
cd /home/hulk/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/hulk/esp/esp-idf -D "SERIAL_TOOL=/home/hulk/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/hulk/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/hulk/code/blink/build -P /home/hulk/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/hulk/code/blink/build/log/idf_py_stderr_output_18391 and /home/hulk/code/blink/build/log/idf_py_stdout_output_18391

I also encountered a similar problem under the windows system, and all of them were newly installed

### Steps to reproduce.

1. [install doc
](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0/esp32s3/get-started/linux-macos-setup.html)
2. install & copy example project
3. idf.py set-target esp32s3
4. idf.py menuconfig
5. idf.py build
6. idf.py flash -p /dev/ttyUSB0    (error)

0xjakob, 2023-01-16:
@yelvlab Can you ls

yelvlab, 2023-01-16:
> @yelvlab what is the output of `ls /dev/ttyUSB0`?

```
hulk@hulk-virtual-machine:~/esp/esp-idf$ ls /dev/ttyUSB0
/dev/ttyUSB0
```

device is OK,and
```
sudo usermod -a -G dialout hulk
```

dobairoland, 2023-01-16:
What if you don't specify the port and run just `idf.py flash`? Can it it find the port/device?

Are you sure you are connected through UART? Since this is ESP32-S3, it can be native USB as well. In that case the port will be `ttyACM` instead of `ttyUSB`.

dobairoland, 2023-01-16:
> hulk@hulk-virtual-machine:~/esp/esp-idf

If you are inside a virtual machine then maybe the port is not properly forwarded. Are you able to flash on the host machine? What OS are you using for as the host?

dobairoland, 2023-01-23:
I'm closing this for inactivity. Let me know if this needs further investigation.

closed 2023-01-23 by dobairoland

goog, 2023-07-18:
i met it, add it solved.
```bash
sudo chmod a+rw /dev/ttyUSB0
```