espressif/esp-idf #10552: light_sleep_enable=true provokes undesired GPIO glitches

mbastida123, 2023-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-devkit

### Power Supply used.

USB

### What is the expected behavior?

The GPIO should switch according to the code.

Using a logic analyzer if 

```
static esp_pm_config_esp32_t pm_config{
    .max_freq_mhz=80,
    .min_freq_mhz=40, 
    .light_sleep_enable=false,    
};
```

Then the GPIO looks like this:
[image]
You can apreciate some small glitches but this doesn't seem to affect the system.

### What is the actual behavior?

```
static esp_pm_config_esp32_t pm_config{
    .max_freq_mhz=80,
    .min_freq_mhz=40, 
    .light_sleep_enable=true,    
};
```

The waveform appears like this:
[image]

Moreover, in this case the GPIO drive a motor driver and so the motor is not working as expected.

### Steps to reproduce.

```
static esp_pm_config_esp32_t pm_config{
    .max_freq_mhz=80,
    .min_freq_mhz=40, 
    .light_sleep_enable=true,    
};
```

`esp_pm_configure(&pm_config);`

Then for GPIO config:
```
    gpio_pad_select_gpio(CONFIG_DRV8833_ENABLE_PIN);
    gpio_pad_select_gpio(CONFIG_DRV8833_IN1);
    gpio_pad_select_gpio(CONFIG_DRV8833_IN2);

    gpio_set_direction((gpio_num_t)CONFIG_DRV8833_ENABLE_PIN, GPIO_MODE_INPUT_OUTPUT);
    gpio_set_direction((gpio_num_t)CONFIG_DRV8833_IN1, GPIO_MODE_OUTPUT);
    gpio_set_direction((gpio_num_t)CONFIG_DRV8833_IN2, GPIO_MODE_OUTPUT);

    gpio_pulldown_dis((gpio_num_t)CONFIG_DRV8833_ENABLE_PIN);
    gpio_pulldown_dis((gpio_num_t)CONFIG_DRV8833_IN1);
    gpio_pulldown_dis((gpio_num_t)CONFIG_DRV8833_IN2);

    gpio_pullup_dis((gpio_num_t)CONFIG_DRV8833_ENABLE_PIN);
    gpio_pullup_dis((gpio_num_t)CONFIG_DRV8833_IN1);
    gpio_pullup_dis((gpio_num_t)CONFIG_DRV8833_IN2);
```
The pins are driven like so:
```
void DRV8833_forward(){
    DRV8833_enable();
    gpio_set_level((gpio_num_t)CONFIG_DRV8833_IN1, 0);
    gpio_set_level((gpio_num_t)CONFIG_DRV8833_IN2, 1);
    ESP_LOGD(TAG, "FORWARD");
}

void DRV8833_reverse(){
    DRV8833_enable();
    gpio_set_level((gpio_num_t)CONFIG_DRV8833_IN1, 1);
    gpio_set_level((gpio_num_t)CONFIG_DRV8833_IN2, 0);
    ESP_LOGD(TAG, "REVERSE");
}
```
The code continuously alternates between forward() and reverse(), it uses vTaskDelay(). I supose that in between these vTaskDelay() the CPU enters light sleep and this causes these glitches.

songruo, 2023-01-18:
Hi @mbastida123,

I wasn't able to reproduce. Does this phenomenon happen to all pins? Can you try to disconnect the motor from the pins, and then observe the signals with the pins only connecting to the logic analyzer?

songruo, 2023-01-18:
@mbastida123 I see. Please see the Kconfig option `PM_SLP_DISABLE_GPIO` (https://github.com/espressif/esp-idf/blob/v4.4.3/components/esp_pm/Kconfig#L80-91). I think either by disabling this option or calling `gpio_sleep_sel_dis` on the pins should solve the problem.

mbastida123, 2023-01-19:
> @mbastida123 I see. Please see the Kconfig option `PM_SLP_DISABLE_GPIO` (https://github.com/espressif/esp-idf/blob/v4.4.3/components/esp_pm/Kconfig#L80-91). I think either by disabling this option or calling `gpio_sleep_sel_dis` on the pins should solve the problem.

@songruo PM_SLP_DISABLE_GPIO was already disabled, so that was not the issue.
Using 
```
    gpio_sleep_sel_dis((gpio_num_t)CONFIG_DRV8833_ENABLE_PIN);
    gpio_sleep_sel_dis((gpio_num_t)CONFIG_DRV8833_IN1);
    gpio_sleep_sel_dis((gpio_num_t)CONFIG_DRV8833_IN2);
```
didn't change anything, it still doesn't switch properly.

mbastida123, 2023-01-19:
> @mbastida123 GPIO16 and GPIO17 are in the VDDSDIO power domain (See https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf pg.16). VDDSDIO power domain is default to be powered down during sleep. You can call `esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_ON)` to force the VDDSDIO domain to be on.
Thank you very much, that was the issue :)

closed 2023-01-19 by mbastida123