espressif/esp-idf #10553: In BLE gatt_server example, profile B's event handler being called/triggered when I'm connecting to profile A

listout, 2023-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-DevKitM-2

### Power Supply used.

USB

### What is the expected behavior?

Only currently used's profile's event being triggered.

I want to know which pofile my gatt client is connecting to, hence in the `ESP_GATTS_CONNECT_EVT` under both the profile's event handler I've put in a debug message printing "Connect event in profile A" and "Connect event in profile B". I'm seeing both the messages when I'm connecting to server's profile A. Shouldn't I be seeing only one of the messages, preferably only the "connect event in profile A" message?

I was trying NimBLe example, I can easy identify which profile I'm connecting to in connect event.

### What is the actual behavior?

Both the profile's event handler's connect event being triggered.

### Steps to reproduce.

1. Build gatt_server example from tag v4.4.3 with the follwing pathc/changes
```diff
diff --git a/main/gatts_demo.c b/main/gatts_demo.c
index ea140e8..fdd6e22 100644
--- a/main/gatts_demo.c
+++ b/main/gatts_demo.c
@@ -462,6 +462,7 @@ static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
     case ESP_GATTS_STOP_EVT:
         break;
     case ESP_GATTS_CONNECT_EVT: {
+        printf("connect event in profile A\n");
         esp_ble_conn_update_params_t conn_params = {0};
         memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
         /* For the IOS system, please reference the apple official documents about the ble connection parameters restrictions. */
@@ -619,6 +620,7 @@ static void gatts_profile_b_event_handler(esp_gatts_cb_event_t event, esp_gatt_i
     case ESP_GATTS_STOP_EVT:
         break;
     case ESP_GATTS_CONNECT_EVT:
+        printf("connect event in profile B\n");
         ESP_LOGI(GATTS_TAG, "CONNECT_EVT, conn_id %d, remote %02x:%02x:%02x:%02x:%02x:%02x:",
                  param->connect.conn_id,
                  param->connect.remote_bda[0], param->connect.remote_bda[1], param->connect.remote_bda[2],

```
2. Build and flash
3. Connect a ble client (phone or gatt_client example) to the ble server
4. Observe both the printf messages appear.

### Debug Logs.

```plain
I (30) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (30) boot: compile time 20:33:30
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=16a14h ( 92692) map
I (146) esp_image: segment 1: paddr=00026a3c vaddr=3ffbdb60 size=04708h ( 18184) load
I (153) esp_image: segment 2: paddr=0002b14c vaddr=40080000 size=04ecch ( 20172) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=5d8e4h (383204) map
I (300) esp_image: segment 4: paddr=0008d90c vaddr=40084ecc size=12204h ( 74244) load
I (331) esp_image: segment 5: paddr=0009fb18 vaddr=50000000 size=00010h (    16) load
I (343) boot: Loaded app from partition at offset 0x10000
I (343) boot: Disabling RNG early entropy source...
I (354) cpu_start: Pro cpu up.
I (355) cpu_start: Starting app cpu, entry point is 0x400811bc
0x400811bc: call_start_cpu1 at /home/listout/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (371) cpu_start: Pro cpu start user code
I (371) cpu_start: cpu freq: 160000000
I (371) cpu_start: Application information:
I (376) cpu_start: Project name:     gatt_server_demos
I (381) cpu_start: App version:      1
I (386) cpu_start: Compile time:     Jan 16 2023 20:33:21
I (392) cpu_start: ELF file SHA256:  656f4870f4d66e68...
I (398) cpu_start: ESP-IDF:          v4.4.3
I (403) heap_init: Initializing. RAM available for dynamic allocation:
I (410) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (416) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (422) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (428) heap_init: At 3FFC5F18 len 0001A0E8 (104 KiB): DRAM
I (434) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (441) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (447) heap_init: At 400970D0 len 00008F30 (35 KiB): IRAM
I (455) spi_flash: detected chip: generic
I (458) spi_flash: flash io: dio
W (462) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (476) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (506) BTDM_INIT: BT controller compile version [67ee4f0]
I (506) system_api: Base MAC address is not set
I (506) system_api: read default base MAC address from EFUSE
I (516) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (906) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (916) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (916) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (936) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (936) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (936) GATTS_DEMO: the gatts demo char length = 3

I (946) GATTS_DEMO: prf_char[0] =11

I (946) GATTS_DEMO: prf_char[1] =22

I (956) GATTS_DEMO: prf_char[2] =33

I (956) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (966) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (976) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (976) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (986) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

connect event in profile A
I (6966) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote ac:0b:fb:6c:f9:fe:
connect event in profile B
I (6966) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote ac:0b:fb:6c:f9:fe:
I (7136) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (8346) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (8346) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (8506) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 2, handle 43
I (8506) GATTS_DEMO: GATT_WRITE_EVT, value len 2, value :
I (8506) GATTS_DEMO: 01 00 
I (8506) GATTS_DEMO: notify enable
I (8516) GATTS_DEMO: ESP_GATTS_CONF_EVT, status 0 attr_handle 42
I (8586) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 3, handle 42
I (8586) GATTS_DEMO: GATT_WRITE_EVT, value len 35, value :
I (8586) GATTS_DEMO: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 
I (8596) GATTS_DEMO: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
I (8596) GATTS_DEMO: 20 21 22
```
```

xiewenxiang, 2023-01-30:
Currently in bluedroid, there is no distinction, both profile callbacks will be triggered. Can you make some distinctions in the application？

listout, 2023-01-31:
I'm not quite familiar with BLE or bluedroid, hence I'm not sure I understand how can perform the distinctions in application.

xiewenxiang, 2023-02-01:
Are you just trying to distinguish connection events? Each profile is accessible as long as a connection comes in, so the connection event is reported again.

For connection or disconnection events, you can focus on just one profile callback, Because they all carry the same parameters.

For read or write characteristic value operations, the only distinguishing condition is attribute handle.

listout, 2023-02-01:
> Are you just trying to distinguish connection events?

Kind of. What I want is having one ble peripheral and two central (phone and another ESP device). I want to do is know when my phone connection happens, and then when my ESP central connects (as I want to toggle a GPIO in this particular situation i.e. when the ESP Central connects).

xiewenxiang, 2023-02-02:
Is it possible to have each client write different data to the characteristic that distinguishes different device types when the connection is established?

listout, 2023-02-02:
I can definitely try that. I think this will be a temporary fix for me now, hence closing the issue. But it would very helpful a feature existed for bluedroid to distinguish between connections using.

closed 2023-02-02 by listout