espressif/esp-idf #10555: IllegalInstruction when connecting to Bluetooth

ftab, 2023-01-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-494-g490216a2ac

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LyraT

### Power Supply used.

USB

### What is the expected behavior?

Connect to Bluetooth reliably without crashing

### What is the actual behavior?

Occasionally crashes with IllegalInstruction

btdm_sleep_check_duration seems to be involved, despite having disabled Bluetooth modem sleep in the config.

### Steps to reproduce.

1. Select our fork of ESP-ADF which has commented out blufi (not working with the latest IDF) - https://github.com/radiosound-com/esp-adf - branch: idf-v5.0-494-g490216a2ac
2. cd ~/esp/esp-adf/examples/advanced_examples/wifi_bt_ble_coex
3. See attached sdkconfig.txt, save as sdkconfig in this folder
4. idf.py menuconfig
5. Put a valid SSID and password in example configuration
6. Build, flash, monitor (in this case I used ESP32-LyraT; issue also occurs on our custom boards with custom firmware)
7. In the phone's Bluetooth settings, connect and disconnect from ESP_ADF_COEX_EXAMPLE until the ESP32 crashes. (In the attached log, it took about 2.5 minutes. In some cases it happens more quickly)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10429095/sdkconfig.txt)

### Debug Logs.

```plain
[1277/1278] Generating binary image from built executableesptool.py v4.4
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Generated /Users/dennis/esp/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/build/wifi_bt_ble_coex.bin
[1278/1278] cd /Users/dennis/esp/esp-adf/examples/advanced_examples/wifi...f/examples/advanced_examples/wifi_bt_ble_coex/build/wifi_bt_ble_coex.binwifi_bt_ble_coex.bin binary size 0x1d5e60 bytes. Smallest app partition is 0x300000 bytes. 0x12a1a0 bytes (39%) free.
Executing action: flash
Serial port /dev/cu.wlan-debug
/dev/cu.wlan-debug failed to connect: Could not open /dev/cu.wlan-debug, the port doesn't exist
Serial port /dev/cu.usbserial-1120
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
Running ninja in directory /Users/dennis/esp/esp-adf/examples/advanced_examples/wifi_bt_ble_coex/build
[... 593 lines cut ...]
ASSERT_ERR(0), in ld_fm.c at line 559
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400927bc: 00f01d00 00004136 f01d0000
0x400927bc: btdm_sleep_check_duration at /Users/dennis/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:952

Core  1 register dump:
PC      : 0x400927c3  PS      : 0x00060034  A0      : 0x80085aad  A1      : 0x3ffc2028  
0x400927c3: r_assert at /Users/dennis/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1863

A2      : 0x3f429a54  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f42a19c  
A6      : 0x0000022f  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1f98  
A10     : 0x00000000  A11     : 0x3ffc1fbb  A12     : 0x3ffc1f67  A13     : 0x00000039  
A14     : 0x00000000  A15     : 0x3ffc1f6c  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085a21  LEND    : 0x40085a29  LCOUNT  : 0x00000000  
0x40085a21: r_assert_err at ??:?

0x40085a29: r_assert_err at ??:?

Backtrace: 0x400927c0:0x3ffc2028 |<-CORRUPTED
0x400927c0: r_assert at /Users/dennis/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1862

ELF file SHA256: edd385a32e9b55a0

CPU halted.
```

ftab, 2023-01-16:
Here is another log from another crash (this time built on Linux)

```
ASSERT_PARAM(131072 0), in rwbt.c at line 381
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092500: f01d092d 00004136 f01d0000
0x40092500: btdm_sleep_check_duration at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:951

Core  0 register dump:
PC      : 0x40092507  PS      : 0x00060834  A0      : 0x80085a29  A1      : 0x3ffc20e0  
0x40092507: r_assert at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1869

A2      : 0x00000000  A3      : 0x00020000  A4      : 0x00000000  A5      : 0x3f42406c  
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2050  
A10     : 0x00000000  A11     : 0x3ffc2073  A12     : 0x3ffc201f  A13     : 0x00000031  
A14     : 0x00000000  A15     : 0x3ffc2024  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085931  LEND    : 0x40085939  LCOUNT  : 0x00000000  
0x40085931: r_assert_param at ??:?

0x40085939: r_assert_param at ??:?

Backtrace: 0x40092504:0x3ffc20e0 0x40085a26:0x3ffc2100 0x4008be85:0x3ffc2120 0x4008c3eb:0x3ffc2140 0x40084511:0x3ffc2160 0x40084351:0x3ffc2180 0x40084333:0x00000000 |<-CORRUPTED
0x40092504: r_assert at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1868

0x40085a26: r_assert_param at ??:?

0x4008be85: r_rwbt_isr at ??:?

0x4008c3eb: r_rwbtdm_isr_wrapper at intc.c:?

0x40084511: hli_c_handler at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/hli_api.c:92

0x40084351: _highint4_stack_switch at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/hli_vectors.S:177

0x40084333: xt_highint4 at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/hli_vectors.S:161

ELF file SHA256: f1a0aff6678ee7ba

Rebooting...
```

xiongweichao, 2023-02-02:
Hi @ftab ,

Sorry, I didn't reproduce the issue. Please use the attached library(replace $IDF_PATH/components/bt/controller/lib_esp32/esp32/libbtdm_app.a) to reproduce the problem. 

Thanks
[libbtdm_app.a.txt](https://github.com/espressif/esp-idf/files/10565623/libbtdm_app.a.txt)

ftab, 2023-02-02:
I didn't see any new kinds of logs with this version of the library. Do I need to enable verbose logging?

It did take me some time to reproduce this time. Not sure if that's because of the updated library or just my luck.

Using a Galaxy A13 (SM-A135U1/DS).

```
W (3768917) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (3769578) BT_APPL: new conn_srvc id:27, app_id:1
W (3769580) BT_HCI: hci cmd send: sniff: hdl 0x81, intv(30 54)
I (3769583) HFP_STREAM: APP HFP event: CONNECTION_STATE_EVT
I (3769584) HFP_STREAM: --Connection state connected, peer feats 0x0, chld_feats 0x0
W (3769587) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 54, status 0x0
W (3769888) BT_HCI: hci cmd send: unsniff: hdl 0x81
W (3769973) BT_HCI: hcif mode change: hdl 0x81, mode 0, intv 0, status 0x0
I (3770105) HFP_STREAM: APP HFP event: CALL_IND_EVT
I (3770106) HFP_STREAM: --Call indicator NO call in progress
I (3770106) HFP_STREAM: APP HFP event: CALL_SEP_ND_EVT[0m
I (3770106) HFP_STREAM: --Call setup idicator NONE
I (3770107) HFP_STREAM: AP HFP vent: ETWRK_STATE_EVT
I (3770107) HFP_STREAM: --NETWORK state unaailable
I (3770107) HFP_STREAM: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (3770111) HFP_STREAM: --Signal strength: 0
I (3770111) HFP_STREAM: APP HFP event: ROAMING_STATUS_IND_EVT
I (3770111) HFP_STREAM: --ROAMING: inactive
I (3770111) HFP_STREAM: APP HFP event: BATTERY_LEVEL_IND_EVT
I (3770112) HFP_STREAM: --Battery level 4
I (3770112) HFP_STREAM: APP HFP event: CALL_HELD_IND_EVT
I (3770112) HFP_STREAM: --Call held indicator NONE held
I (3770385) HFP_STREAM: APP HFP event: CONNECTION_STATE_EVT
I (3770385) HFP_STREAM: --Connection state slc_connected, peer feats 0xfef, chld_feats 0x3f
I (3770385) HFP_STREAM: APP HFP event: INBAND_RING_TONE_EVT
I (3770386) HFP_STREAM: --Inband ring state Provided
W (3770508) BT_APPL: bta_hf_client_send_at: No service, skipping 11 command
E (3770594) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (3770595) COEX_EXAMPLE: A2DP sink user cb
I (3770595) COEX_EXAMPLE: A2DP connecting
I (3770595) A2DP_STREAM: A2DP bd address:, [9c:25:95:5c:66:4e]
W (3770623) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:1 state 2 closed: Peer connection failed (res: 16)
W (3770627) BT_HCI: hci cmd send: sniff: hdl 0x81, intv(30 54)
I (3770630) HFP_STREAM: APP HFP event: CONNECTION_STATE_EVT
I (3770630) HFP_STREAM: --Connection state disconnected, peer feats 0x0, chld_feats 0x0
W (3770694) BT_HCI: hcif mode change: hdl 0x81, mode 2, intv 54, status 0x0
W (3770697) BT_HCI: hci cmd send: unsniff: hdl 0x81
W (3770698) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x40 current:0x0
W (3770699) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x40
I (3770730) COEX_EXAMPLE: A2DP sink user cb
I (3770730) COEX_EXAMPLE: User cb A2DP event: 2
I (3770730) A2DP_STREAM: A2DP audio stream configuration, codec type 0
I (3770731) A2DP_STREAM: Bluetooth configured, sample rate=44100
D (3770731) AUDIO_ELEMENT: REPORT_INFO,[IN_aadp]evt out cmd:9,
W (3770731) AUDIO_EVT: Thereis no spae in external queue
I (3770732) COEX_EXAMPLE A2DP sik user cb
I (3770732) COEXEXAMPLE: Usercb A2D event: 5
I (3770732) A2DP_STREAM: Unhandled A2DP event: 5
W (3770748) BT_HCI: hcif mode change: hdl 0x81, mode 0, intv 0, status 0x0
ASSERT_PARAM(131072 0), in rwbt.c at line 381
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092558: 00f01d00 00004136 f01d0000
0x40092558: btdm_sleep_check_duration at /Users/dennis/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:951

Core  1 register dump:
PC      : 0x4009255f  PS      : 0x00060234  A0      : 0x80085dc5  A1      : 0x3ffc2000  
0x4009255f: r_assert at /Users/dennis/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1869

A2      : 0x00000000  A3      : 0x00020000  A4      : 0x00000000  A5      : 0x3f442638  
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1f70  
A10     : 0x00000000  A11     : 0x3ffc1f93  A12     : 0x3ffc1f3f  A13     : 0x00000031  
A14     : 0x00000000  A15     : 0x3ffc1f44  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085ccd  LEND    : 0x40085cd5  LCOUNT  : 0x00000000  
[... 33 lines cut ...]
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffcddb0  A13     : 0x00000048  
A14     : 0x3ffcc5b0  A15     : 0x3ffb8360  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0014833c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40015785:0x3ffcf320 0x4017d93a:0x3ffcf350 0x401803b3:0x3ffcf3c0 0x401835e1:0x3ffcf3f0 0x40183c3a:0x3ffcf450 0x40180cc3:0x3ffcf4a0 0x4017dee2:0x3ffcf4f0 0x40019d11:0x3ffcf530 0x40055b4d:0x3ffcf550 0x4017404f:0x3ffcf570 0x401746c1:0x3ffcf590
0x4017d93a: r_lld_evt_move_to_slave at ??:?

0x401803b3: r_lld_move_to_slave_hack at ??:?

0x401835e1: r_llm_con_req_ind at ??:?

0x40183c3a: r_llm_pdu_defer at ??:?

0x40180cc3: r_lld_pdu_check at ??:?

0x4017dee2: r_lld_evt_deffered_elt_handler at ??:?

0x4017404f: r_rw_schedule at ??:?

0x401746c1: btdm_controller_task at ??:?

ELF file SHA256: 9e0a528ab9cf11d7

CPU halted.
```

xiongweichao, 2023-02-03:
Hi @ftab ,

Since the log is incomplete, I can't be sure whether you are using the library I provided. I have added some logs to this version.
``` 
I (9663) COEX_EXAMPLE: User cb A2DP event: 2
I (9663) COEX_EXAMPLE: A2DP sink user cb
I (9664) COEX_EXAMPLE: User cb A2DP event: 5
W (9691) BT_APPL: new conn_srvc id:19, app_id:0
I (9692) COEX_EXAMPLE: A2DP sink user cb
I (9692) COEX_EXAMPLE: A2DP connected
SniffAttempt = 4
SniffTimeout = 1
W (12896) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
lc_env_ptr->sniff.SniffAttempt = 4
W (12900) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0

```
[lib_d9a1785.zip](https://github.com/espressif/esp-idf/files/10576072/lib_d9a1785.zip)

Thanks

ftab, 2023-02-03:
Sorry about that. My terminal had lost its scrollback buffer due to the length of the log. I've increased my scrollback buffer size and captured the full log this time.

[2023-02-03-esp-idf-gh-issue-10555-crash-log.txt](https://github.com/espressif/esp-idf/files/10583240/2023-02-03-esp-idf-gh-issue-10555-crash-log.txt)

ftab, 2023-02-03:
Here is a crash log with all Bluetooth logs set to verbose

[2023-02-03-esp-idf-gh-issue-10555-crash-log-verbose.txt](https://github.com/espressif/esp-idf/files/10583309/2023-02-03-esp-idf-gh-issue-10555-crash-log-verbose.txt)

xiongweichao, 2023-02-06:
Hi @ftab ,

I added some logs to this library, please help reproduce the problem.
BTW, can all mobile phones reproduce the problem? 

Thanks

[libbtdm_4967a98.zip](https://github.com/espressif/esp-idf/files/10659663/libbtdm_4967a98.zip)

ftab, 2023-02-06:
Here's the latest log

[2023-02-06-esp-idf-gh-issue-10555-btdm-4967a98-crash-log-verbose.txt.zip](https://github.com/espressif/esp-idf/files/10668588/2023-02-06-esp-idf-gh-issue-10555-btdm-4967a98-crash-log-verbose.txt.zip)

The issue has occurred on at least 2 out of the 3 phones I've tried so far: Galaxy A13 LTE (SM-A135U1), iPhone 11 Pro Max. The third phone I've been trying, Google Pixel 4a 5g, I may've just not tried for long enough yet.

It has even happened while idle (connected but not doing anything). (This was captured with an older version of the library. I am trying to capture it again with the latest version)

[2023-02-03-esp-idf-gh-issue-10555-btdm-d9a1785-idle-crash.txt](https://github.com/espressif/esp-idf/files/10668682/2023-02-03-esp-idf-gh-issue-10555-btdm-d9a1785-idle-crash.txt)

xiongweichao, 2023-02-07:
Hi @ftab ,

Please verify that this library can solve the problem.

Thanks
[libbtdm_0e3d793.zip](https://github.com/espressif/esp-idf/files/10673510/libbtdm_0e3d793.zip)

ftab, 2023-02-07:
Unfortunately it still crashes. This time it looked a little weird, disconnecting from the device and then spitting out some cryptic FEX statements.. The phone was attempting to connect to it but failed, and then the esp32 crashed

[2023-02-07-esp-idf-gh-issue-10555-btdm-0e3d793-crash.txt](https://github.com/espressif/esp-idf/files/10678485/2023-02-07-esp-idf-gh-issue-10555-btdm-0e3d793-crash.txt)

xiongweichao, 2023-02-08:
Hi @ftab ,

Sorry, I still can't reproduce the problem. Do you have other modifications locally?

Please help to verify whether this library can solve the problem.
[libbtdm_75939d3.zip](https://github.com/espressif/esp-idf/files/10681294/libbtdm_75939d3.zip)

Thanks

ftab, 2023-02-08:
These are my only diffs:

```diff
build@build-ThinkPad-W530:~/esp/esp-adf/examples/advanced_examples/wifi_bt_ble_coex$ git diff
diff --git a/esp-idf b/esp-idf
--- a/esp-idf
+++ b/esp-idf
@@ -1 +1 @@
-Subproject commit 490216a2ace6dc3e1b9a3f50d265a80481b32f6d
+Subproject commit 490216a2ace6dc3e1b9a3f50d265a80481b32f6d-dirty
diff --git a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
index dfe446f..2d407c7 100644
--- a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
+++ b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
@@ -354,8 +354,8 @@ void app_main(void)
     tcpip_adapter_init();
 #endif
 
-    esp_log_level_set("*", ESP_LOG_WARN);
-    esp_log_level_set(TAG, ESP_LOG_DEBUG);
+    //esp_log_level_set("*", ESP_LOG_WARN);
+    //esp_log_level_set(TAG, ESP_LOG_DEBUG);
     
     ESP_LOGI(TAG, "[ 1 ] Create coex handle for a2dp-gatt-wifi");
     g_coex_handle = (coex_handle_t *)audio_malloc(sizeof(coex_handle_t));
build@build-ThinkPad-W530:~/esp/esp-adf/examples/advanced_examples/wifi_bt_ble_coex$ cd ~/esp/esp-adf/esp-idf
build@build-ThinkPad-W530:~/esp/esp-adf/esp-idf$ git diff
diff --git a/components/bt/controller/lib_esp32 b/components/bt/controller/lib_esp32
--- a/components/bt/controller/lib_esp32
+++ b/components/bt/controller/lib_esp32
@@ -1 +1 @@
-Subproject commit d4a224c5d682d6b5a76542c6918b6a59dd0b2f8c
+Subproject commit d4a224c5d682d6b5a76542c6918b6a59dd0b2f8c-dirty
```

I can't imagine that would cause it, though.

Here is the crash with the latest library:

[2023-02-08-esp-idf-gh-issue-10555-btdm-75939d3-crash.txt](https://github.com/espressif/esp-idf/files/10686834/2023-02-08-esp-idf-gh-issue-10555-btdm-75939d3-crash.txt)

(Scroll to line 10,000 - my terminal buffer was cut short on this machine so I included the startup logs from the next reboot)

ftab, 2023-02-08:
[2023-02-08-esp-idf-gh-issue-10555-btdm-75939d3-iphone-crash.txt](https://github.com/espressif/esp-idf/files/10686995/2023-02-08-esp-idf-gh-issue-10555-btdm-75939d3-iphone-crash.txt)

Interesting. This crash seems different.

ftab, 2023-02-15:
I wish I could take a look at the code so I could help better. My "shot in the dark" theory is it has something to do with the WiFi coexistence code, because it seems to be exclusively when connected to WiFi, or at least, I haven't tried for long enough to reproduce without being connected to WiFi.

Sometimes it took quite a while, half an hour or an hour of repeated disconnects and reconnects to crash. Even when crashing at idle (connected to wifi and bluetooth, not streaming), it feels like a random amount of time. The phone could remain connected to the board for days with no issue, but then some other time crash within minutes.

ftab, 2023-02-17:
I was wrong. This time I was not connected to WiFi, and it crashed while idle (though I had tried earlier in the log to crash it with repeated disconnects & reconnects)

[2023-02-16-esp-idf-gh-issue-10555-btdm-75939d3-idle-crash-no-wifi.txt](https://github.com/espressif/esp-idf/files/10768662/2023-02-16-esp-idf-gh-issue-10555-btdm-75939d3-idle-crash-no-wifi.txt)

xiongweichao, 2023-04-14:
@ftab Sorry to reply you so late. 

1. This [commit](https://github.com/espressif/esp-idf/commit/f944418ca4dd227a11a896a5a8529ad227c79673) fixes the connection failure due to memory leaks.
2. Please try this new lib again.
[esp32_20230414.zip](https://github.com/espressif/esp-idf/files/11232021/esp32_20230414.zip)

ftab, 2023-04-14:
Thanks for the update. I have pulled the latest release/v5.0 branch of esp-idf and installed the attached library. Beginning my tests now. I will follow up with my findings.

ftab, 2023-04-14:
This is probably a different crash (heap corruption). I am still trying to reproduce the original crash (I will leave it streaming over the weekend) but figured I'd go ahead and give you this log:

```
I (27) boot: ESP-IDF v5.0.1-464-gef4b1b7704-dirty 2nd stage bootloader
I (27) boot: compile time 09:10:16
I (27) boot: chip revision: v1.0
I (27) boot.esp32: SPI Speed      : 80MHz
I (28) boot.esp32: SPI Mode       : DIO
I (28) boot.esp32: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (29) boot: Partition Table:
I (29) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (30) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (30) boot:  2 factory          factory app      00 00 00010000 00300000
I (30) boot: End of partition table
I (31) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=71bd8h (465880) map
I (172) esp_image: segment 1: paddr=00081c00 vaddr=3ffbdb60 size=05978h ( 22904) load
I (180) esp_image: segment 2: paddr=00087580 vaddr=40080000 size=08a98h ( 35480) load
I (192) esp_image: segment 3: paddr=00090020 vaddr=400d0020 size=177e70h (1539696) map
I (656) esp_image: segment 4: paddr=00207e98 vaddr=40088a98 size=16b08h ( 92936) load
I (705) boot: Loaded app from partition at offset 0x10000
I (705) boot: Disabling RNG early entropy source...
I (706) quad_psram: This chip is ESP32-D0WD
I (706) esp_psram: Found 8MB PSRAM device
I (706) esp_psram: Speed: 80MHz
I (706) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (707) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (707) cpu_start: Pro cpu up.
I (707) cpu_start: Starting app cpu, entry point is 0x4008184c
0x4008184c: call_start_cpu1 at /Users/dennis/esp/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (1172) esp_psram: SPI SRAM memory test OK
I (1181) cpu_start: Pro cpu start user code
I (1181) cpu_start: cpu freq: 240000000 Hz
I (1181) cpu_start: Application information:
I (1181) cpu_start: Project name:     wifi_bt_ble_coex
I (1181) cpu_start: App version:      v2.3-194-g0eafda3-dirty
I (1182) cpu_start: Compile time:     Apr 14 2023 09:10:04
I (1182) cpu_start: ELF file SHA256:  ef618c263c5ccbe4...
I (1182) cpu_start: ESP-IDF:          v5.0.1-464-gef4b1b7704-dirty
I (1183) cpu_start: Min chip rev:     v1.0
I (1183) cpu_start: Max chip rev:     v3.99 
I (1183) cpu_start: Chip rev:         v1.0
I (1184) heap_init: Initializing. RAM available for dynamic allocation:
I (1184) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1184) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1185) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1185) heap_init: At 3FFC5578 len 0001AA88 (106 KiB): DRAM
I (1185) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1186) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1186) heap_init: At 4009F5A0 len 00000A60 (2 KiB): IRAM
I (1469) esp_psram: Adding pool of 4082K of PSRAM memory to heap allocator
I (1471) spi_flash: detected chip: generic
I (1471) spi_flash: flash io: dio
W (1472) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
W (1472) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
I (1473) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1490) COEX_EXAMPLE: [ 1 ] Create coex handle for a2dp-gatt-wifi
I (1490) COEX_EXAMPLE: [ 2 ] Initialize peripherals
I (1491) COEX_EXAMPLE: [ 3 ] create and start input key service
I (1492) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1492) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1493) COEX_EXAMPLE: [ 4 ] Start a2dp and blufi network
I (1493) COEX_EXAMPLE: [4.1] Init Bluetooth
I (1493) BTDM_INIT: BT controller compile version [ed98430]
I (1495) system_api: Base MAC address is not set
I (1495) system_api: read default base MAC address from EFUSE
[... 164 lines cut ...]
0x40169e0b: bta_av_stream_data_cback at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/bta/av/bta_av_aact.c:586

0x401572bd: avdt_scb_hdl_pkt_no_frag at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/avdt/avdt_scb_act.c:300

0x40157b64: avdt_scb_hdl_pkt at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/avdt/avdt_scb_act.c:642

0x40156e11: avdt_scb_event at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/avdt/avdt_scb.c:561

0x4017446d: avdt_ad_tc_data_ind at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/avdt/avdt_ad.c:451

0x401556df: avdt_l2c_data_ind_cback at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/avdt/avdt_l2c.c:495

0x40124699: l2c_csm_open at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_csm.c:972
 (inlined by) l2c_csm_execute at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_csm.c:94

0x4012958a: l2c_rcv_acl_data at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_main.c:309

0x40113c2a: btu_hci_msg_process at /Users/dennis/esp/esp-adf/esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c:144

0x4013d317: osi_thread_run at /Users/dennis/esp/esp-adf/esp-idf/components/bt/common/osi/thread.c:165

ELF file SHA256: ef618c263c5ccbe4

CPU halted.
```

xiongweichao, 2023-04-17:
Hi @ftab ,

Sorry, I still can't reproduce your problem. Could you please help enable this [log](https://github.com/espressif/esp-idf/blob/f404fe96b17692e3f1de536a3d73a180cdb53b42/components/bt/host/bluedroid/Kconfig.in#L794)?

Thanks

ftab, 2023-04-18:
I have enabled BT_APPL debug logs, but I have not had the heap corruption issue aside from that one time on Friday. I must've gotten (un)lucky. I also have not been able to reproduce the IllegalInstruction crash at all with your latest library. I've been streaming for days, and hammering on the repeated disconnect/reconnect scenario for a while, and it hasn't crashed.

xiongweichao, 2023-04-19:
Hi @ftab ,

Could you please provide a test log? Because I added some printing in the new lib. For example `fifo empty`.

Thanks

ftab, 2023-04-19:
Here is the full log from my last test. It was almost a full day of streaming, followed by repeated disconnect/reconnect attempts, and has been sitting idle (but connected) for almost another full day since my last attempt (the last log entry). However, I don't see any prints of `fifo empty`

[full-log.txt](https://github.com/espressif/esp-idf/files/11277439/full-log.txt)

xiongweichao, 2023-04-20:
@ftab Maybe you can use this lib for the time being. Please let me know when the print of `fifo empty` appears. thanks.

ftab, 2023-04-28:
I regret to report that there was an IllegalInstruction crash. All I got was the backtrace because I did it on a machine with a short (10,000 lines) scrollback buffer and the initial boot was scrolled out by noisy WiFi warnings over a period of about a day and a half.

```
2023-04-26 01:25:41.349533 ASSERT_PARAM(1024 0), in rwbt.c at line 381
2023-04-26 01:25:42.740425 Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
2023-04-26 01:25:42.741095 Memory dump at 0x40093078: f01d092d 00004136 f01d0000
2023-04-26 01:25:42.741173 0x40093078: btdm_sleep_check_duration at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:952
2023-04-26 01:25:42.808939 
Core  1 register dump:
2023-04-26 01:25:42.811064 PC      : 0x4009307f  PS      : 0x00060634  A0      : 0x80085c9a  A1      : 0x3ffc2304  
2023-04-26 01:25:42.811290 0x4009307f: r_assert at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1863
2023-04-26 01:25:42.832693 
A2      : 0x00000000  A3      : 0x00000400  A4      : 0x00000000  A5      : 0x3f4e8f6c  
2023-04-26 01:25:42.833405 A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2274  
2023-04-26 01:25:42.833494 A10     : 0x00000000  A11     : 0x3ffc2297  A12     : 0x3ffc2243  A13     : 0x00000031  
2023-04-26 01:25:42.833538 A14     : 0x00000000  A15     : 0x3ffc2248  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
2023-04-26 01:25:42.833568 EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
2023-04-26 01:25:42.833599 
2023-04-26 01:25:42.833735 
2023-04-26 01:25:42.833775 Backtrace: 0x4009307c:0x3ffc2304 |<-CORRUPTED
2023-04-26 01:25:42.833801 0x4009307c: r_assert at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1862
2023-04-26 01:25:42.853804 
2023-04-26 01:25:42.853932 
2023-04-26 01:25:42.853987 
2023-04-26 01:25:42.854015 
2023-04-26 01:25:42.854039 ELF file SHA256: 27c2ac0dc717a31d
2023-04-26 01:25:42.854062 
2023-04-26 01:25:42.889841 Rebooting...
```

I still have not seen `fifo empty`

I will be on vacation through May 8, so I will resume testing around May 9

rockys-spindance, 2023-06-06:
@ftab or @xiongweichao 
I'm curious if there has been any more progress on this issue?

We are seeing something very similar using the latest esp-idf  release/v5.0 head ( v5.0.2-279-ga6a962d5f8 ) on a DevkitC-ESP32-WROVER-E.

The panic occurs sometimes during BLE connection or disconnection with a Windows 11 laptop (ThinkPad T14).  And sometimes occurs randomly during a long-established BLE connection with the laptop.  

In particular, we see the same **IllegalInstruction** panic occurring inside **btdm_sleep_check_duration** even though CONFIG_BTDM_CTRL_MODEM_SLEEP is disabled.  The only difference we know of with the results earlier in this ticket is that we are using the nimBLE host not bluedroid.

Here's a log snippet showing that the panic looks very similar to earlier reports above:

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 369
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092a60: f01d020c 00004136 f01d0000
0x40092a60: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.0/components/bt/controller/esp32/bt.c:950

Core  0 register dump:
PC      : 0x40092a67  PS      : 0x00060634  A0      : 0x80086561  A1      : 0x3ffdd4e0
0x40092a67: r_assert at C:/Espressif/frameworks/esp-idf-v5.0/components/bt/controller/esp32/bt.c:1890

A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f41099e
A6      : 0x00000171  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffdd450
A10     : 0x00000000  A11     : 0x3ffdd473  A12     : 0x3ffdd41f  A13     : 0x00000039
A14     : 0x00000000  A15     : 0x3ffdd424  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40086469  LEND    : 0x40086471  LCOUNT  : 0x00000000
0x40086469: r_assert_param at ??:?

0x40086471: r_assert_param at ??:?

Backtrace: 0x40092a64:0x3ffdd4e0 0x4008655e:0x3ffdd500 0x400ffb81:0x3ffdd520 0x40019fb5:0x3ffdd540 0x4001a1f2:0x3ffdd570 0x40019471:0x3ffdd590 0x400175a7:0x3ffdd5d0 0x40019d11:0x3ffdd5f0 0x40055b4d:0x3ffdd610 0x400fff67:0x3ffdd630 0x401005c5:0x3ffdd650 0x40098711:0x3ffdd680
0x40092a64: r_assert at C:/Espressif/frameworks/esp-idf-v5.0/components/bt/controller/esp32/bt.c:1889

0x4008655e: r_assert_param at ??:?

0x400ffb81: r_platform_reset at ??:?

0x400fff67: r_rw_schedule at ??:?

0x401005c5: btdm_controller_task at ??:?

0x40098711: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

ftab, 2023-06-06:
No change on our end. The latest library (from https://github.com/espressif/esp-idf/issues/10555#issuecomment-1508326921 ) still exhibits the issue, but doesn't seem to log anything new or interesting along the way. However, it has overall gotten harder to reproduce on demand, albeit still non-zero chance of occurring. (Sometimes I will have several crashes one right after the other, of course this is when I can't have a console attached so I don't know for sure what happened--other times, I will be able to go weeks without a crash, and it doesn't produce any new information)

xiongweichao, 2023-06-07:
Hi @rockys-spindance ,

Maybe you can create a new issue.

rockys-spindance, 2023-06-07:
@ftab and @xiongweichao  Thanks so much for the quick response.
Per @xiongweichao's suggestion, I will create a new issue with the full details that we are seeing.

rockys-spindance, 2023-06-08:
Per yesterday's thread, I have created a new issue here
https://github.com/espressif/esp-idf/issues/11624

The encouraging news is that we found a way to generate exactly the same failure backtrace in a quickly reproducible way.  Fingers crossed that the reproducible proximate failure has the same root cause as the intermittent crash we were all seeing.

ftab, 2023-07-20:
I have updated to esp-idf release/v5.0 to bring in the updated Bluetooth library from the fix for #11624, and made a small edit to bring in the fix for #11197 so that I could compile again. So, I am now using this version of esp-adf: https://github.com/radiosound-com/esp-adf/tree/radiosound-modded-v2.5-66-g49e63cd5 (updated to a recent copy of esp-adf master with a few minor changes, instead of my old fork which hacked up a bunch of stuff to work with idf5 and may have had other changes that could have caused an issue)

I was hoping that with the fix to #11624 this issue would be fixed as well, but I just got the same crash in the middle of streaming as I have been getting. So unfortunately this is still happening to us.

```
I (29) boot: ESP-IDF v5.0.3-147-gdc9a3c43ea 2nd stage bootloader
I (29) boot: compile time 10:13:53
I (29) boot: Multicore bootloader
I (29) boot: chip revision: v3.1
I (29) boot.esp32: SPI Speed      : 80MHz
I (29) boot.esp32: SPI Mode       : DIO
I (30) boot.esp32: SPI Flash Size : 8MB
I (30) boot: Enabling RNG early entropy source...
I (30) boot: Partition Table:
I (30) boot: ## Label            Usage          Type ST Offset   Length
I (31) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (31) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (32) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (32) boot:  3 ota_0            OTA app          00 10 00010000 00380000
I (33) boot:  4 ota_1            OTA app          00 11 00390000 00380000
I (33) boot:  5 storage          Unknown data     01 82 00710000 000c0000
I (34) boot: End of partition table
I (34) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=137150h (1274192) map
I (418) esp_image: segment 1: paddr=00147178 vaddr=3ff80063 size=00008h (     8) load
I (419) esp_image: segment 2: paddr=00147188 vaddr=3ffbdb60 size=04858h ( 18520) load
I (425) esp_image: segment 3: paddr=0014b9e8 vaddr=40080000 size=04630h ( 17968) load
I (432) esp_image: segment 4: paddr=00150020 vaddr=400d0020 size=1821f4h (1581556) map
I (908) esp_image: segment 5: paddr=002d221c vaddr=40084630 size=1a390h (107408) load
I (946) esp_image: segment 6: paddr=002ec5b4 vaddr=400c0000 size=00064h (   100) load
I (962) boot: Loaded app from partition at offset 0x10000
I (962) boot: Disabling RNG early entropy source...
I (963) cpu_start: Multicore app
I (963) quad_psram: This chip is ESP32-D0WD
I (963) esp_psram: Found 8MB PSRAM device
I (963) esp_psram: Speed: 80MHz
I (963) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (964) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (964) cpu_start: Pro cpu up.
I (964) cpu_start: Starting app cpu, entry point is 0x400818d8
0x400818d8: call_start_cpu1 at /home/build/esp/esp-adf/esp-idf/components/esp_system/port/cpu_start.c:143

I (950) cpu_start: App cpu up.
I (1432) esp_psram: SPI SRAM memory test OK
I (1441) cpu_start: Pro cpu start user code
I (1441) cpu_start: cpu freq: 240000000 Hz
I (1441) cpu_start: Application information:
I (1441) cpu_start: Project name:     ...
I (1441) cpu_start: App version:      008c14f-dirty
I (1442) cpu_start: Compile time:     Jul 20 2023 11:16:35
I (1442) cpu_start: ELF file SHA256:  33756a4c2bbae400...
I (1442) cpu_start: ESP-IDF:          v5.0.3-147-gdc9a3c43ea
I (1443) cpu_start: Min chip rev:     v3.0
I (1443) cpu_start: Max chip rev:     v3.99 
I (1443) cpu_start: Chip rev:         v3.1
I (1444) heap_init: Initializing. RAM available for dynamic allocation:
I (1444) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1444) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1445) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1445) heap_init: At 3FFC9AB8 len 00016548 (89 KiB): DRAM
I (1445) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1446) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1446) heap_init: At 4009E9C0 len 00001640 (5 KiB): IRAM
I (1447) esp_psram: Adding pool of 4084K of PSRAM memory to heap allocator
I (1448) spi_flash: detected chip: gd
I (1448) spi_flash: flash io: dio
W (1448) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
W (1449) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
W (1449) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (1451) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 21 lines cut ...]

Backtrace: 0x40092e8c:0x3ffc05a0 0x40085913:0x3ffc05c0 0x4008c52d:0x3ffc05e0 0x4008ca93:0x3ffc0600 0x40083a45:0x3ffc0620 0x40093083:0x3ffbcd50 0x400d4632:0x3ffbcd70 0x4009744f:0x3ffbcd90
0x40092e8c: r_assert at /home/build/esp/esp-adf/esp-idf/components/bt/controller/esp32/bt.c:1889

0x40085913: r_assert_param at ??:?

0x4008c52d: r_rwbt_isr at ??:?

0x4008ca93: r_rwbtdm_isr_wrapper at intc.c:?

0x40083a45: _xt_lowint1 at /home/build/esp/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118

0x40093083: xt_utils_wait_for_intr at /home/build/esp/esp-adf/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/build/esp/esp-adf/esp-idf/components/esp_hw_support/cpu.c:101

0x400d4632: esp_vApplicationIdleHook at /home/build/esp/esp-adf/esp-idf/components/esp_system/freertos_hooks.c:59

0x4009744f: prvIdleTask at /home/build/esp/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4157

ELF file SHA256: 33756a4c2bbae400

Rebooting...
```

The full log has stuff about our custom board which isn't relevant here. I will try to find some time to get back to reproducing this on LyraT in the esp-adf example I was using previously.

xiongweichao, 2023-08-11:
Hi @ftab ,

Set to non-discoverable non-connectable mode after the [connection is complete](https://github.com/radiosound-com/esp-adf/blob/49e63cd5367572c8f74d018a15069e088b9743ae/components/bluetooth_service/bluetooth_service.c#L181). Please let me know the test results.

```
esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
```

Thanks

ftab, 2023-08-11:
I have applied the suggested change, as well as the equivalent change to main/wifi_bt_ble_coex_example.c since that doesn't use bluetooth_service (it uses a2dp_stream). I also had to apply an esp-idf patch from esp-adf in order to get back to using the example ([required by audio_thread](https://github.com/espressif/esp-adf/blob/master/components/audio_sal/audio_thread.c#L43)). To recap, here are my git statuses and diffs:

```diff
build@dennis-2021:~/esp/esp-adf$ git status
On branch radiosound-modded-v2.5-66-g49e63cd5
Your branch is up to date with 'radiosound/radiosound-modded-v2.5-66-g49e63cd5'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   components/bluetooth_service/bluetooth_service.c
	modified:   esp-idf (modified content)
	modified:   examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c

no changes added to commit (use "git add" and/or "git commit -a")
build@dennis-2021:~/esp/esp-adf$ git diff
diff --git a/components/bluetooth_service/bluetooth_service.c b/components/bluetooth_service/bluetooth_service.c
index 594c04c5..2a0999fa 100644
--- a/components/bluetooth_service/bluetooth_service.c
+++ b/components/bluetooth_service/bluetooth_service.c
@@ -178,6 +178,11 @@ static void bt_a2d_sink_cb(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *p_param
 
             if (g_bt_service->connection_state == ESP_A2D_CONNECTION_STATE_DISCONNECTED
                 && a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_CONNECTED) {
+#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0))
+                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
+#else
+                esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE);
+#endif
                 memcpy(&g_bt_service->remote_bda, &a2d->conn_stat.remote_bda, sizeof(esp_bd_addr_t));
                 g_bt_service->connection_state = a2d->conn_stat.state;
                 ESP_LOGD(TAG, "A2DP connection state = CONNECTED");
@@ -188,6 +193,11 @@ static void bt_a2d_sink_cb(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *p_param
             if (memcmp(&g_bt_service->remote_bda, &a2d->conn_stat.remote_bda, sizeof(esp_bd_addr_t)) == 0
                 && g_bt_service->connection_state == ESP_A2D_CONNECTION_STATE_CONNECTED
                 && a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
+#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0))
+                esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
+#else
+                esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);
+#endif
                 memset(&g_bt_service->remote_bda, 0, sizeof(esp_bd_addr_t));
                 g_bt_service->connection_state = ESP_A2D_CONNECTION_STATE_DISCONNECTED;
                 ESP_LOGD(TAG, "A2DP connection state =  DISCONNECTED");
diff --git a/esp-idf b/esp-idf
--- a/esp-idf
+++ b/esp-idf
@@ -1 +1 @@
-Subproject commit dc9a3c43ea24007115382f7f194d03882031b091
+Subproject commit dc9a3c43ea24007115382f7f194d03882031b091-dirty
diff --git a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
index 0a90b917..526cc410 100644
--- a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
+++ b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
@@ -237,9 +237,11 @@ static void user_a2dp_sink_cb(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *para
         case ESP_A2D_CONNECTION_STATE_EVT:
             if (param->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
                 ESP_LOGI(TAG, "A2DP disconnected");
+                esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                 g_a2dp_connect_state = false;
             } else if (param->conn_stat.state == ESP_A2D_CONNECTION_STATE_CONNECTED) {
                 ESP_LOGI(TAG, "A2DP connected");
+                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
                 g_a2dp_connect_state = true;
             }
             break;
build@dennis-2021:~/esp/esp-adf$ cd esp-idf
build@dennis-2021:~/esp/esp-adf/esp-idf$ git status
On branch release/v5.0+0634bb9
[... 222 lines cut ...]

E (409561) BT_BTM: tBTM_SEC_DEV:0x3fff9c70 rs_disc_pending=0

W (409701) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
E (409771) BT_APPL: av scb not available for avdt connection
E (409801) BT_APPL: rcb[0].lidx=3, lcb.conn_msk=x1
E (409801) BT_BTC: Got RC OPEN in connected state, Connected RC: 1                 and Current RC: 0
W (409801) BT_AVCT: avct_lcb_last_ccb
W (409801) BT_AVCT: 0: aloc:1, lcb:0x3ffce5d4/0x3ffce5d4, ccb:0x3ffce61c/0x3ffe61
W (409801) BT_AVC: 1: aloc:1, lcb0x3ffc5b4/0xffce54, ccb:0x3ffce63/0x3ffce61c
W (409801) BT_AVCT: 2: aloc:0, lcb:0x0/0x3ffce5d4, ccb:0x3ffce64c/0x3ffce61c
E (409831) BT_BTC: Got disconnect of unknown device
W (409851) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x43 current:0x0
W (409851) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x43
W (410511) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 32000
W (411481) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
ELx400
PR
I (1456741) COEX_EXAMPLE: [ * ] input key id is 4, key type is 1
I (1456891) COEX_EXAMPLE: [ * ] input key id is 4, key type is 2
I (1456891) COEX_EXAMPLE: [ * ] [mode]
I (1456891) COEX_EXAMPLE: [ * ] Enter BT mode
```

I have 3 test setups I will leave streaming over the weekend to try to catch a crash. When I get back on Monday I will try the other method I had of reproducing, that is repeatedly disconnecting and reconnecting.

ftab, 2023-08-14:
In the hopes of capturing something useful, I had debug logging turned on, but that also includes a lot of noise from i2s as well as BT_APPL for every packet.

The log is too big to attach here (73.4 MB gzipped, 882 MB uncompressed): https://nextcloud.apps.radiosound.com/s/D2iRP2f9tbJpoaT

It appears to have streamed for about 7 hours, then Bluetooth disconnected and it crashed very shortly thereafter.

```
ASSERT_PARAM(1024 0), in rwbt.c at line 381
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092d04: 00f01d00 00004136 f01d0000
Core  1 register dump:
PC      : 0x40092d0b  PS      : 0x00060134  A0      : 0x80085e0a  A1      : 0x3ffc2084  
A2      : 0x00000000  A3      : 0x00000400  A4      : 0x00000000  A5      : 0x3f444c04  
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1ff4  
A10     : 0x00000000  A11     : 0x3ffc2017  A12     : 0x3ffc1fc3  A13     : 0x00000031  
A14     : 0x00000000  A15     : 0x3ffc1fc8  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40092d08:0x3ffc2084 |<-CORRUPTED

ELF file SHA256: f4cd83c8699d8ccf

CPU halted.
```

```
$ xtensa-esp32-elf-addr2line -pfiaC -e build/wifi_bt_ble_coex.elf 0x40092d04
0x40092d04: btdm_sleep_check_duration at /COMPONENT_BT_DIR/controller/esp32/bt.c:955
$ xtensa-esp32-elf-addr2line -pfiaC -e build/wifi_bt_ble_coex.elf 0x40092d0b
0x40092d0b: r_assert at /COMPONENT_BT_DIR/controller/esp32/bt.c:1890
$ xtensa-esp32-elf-addr2line -pfiaC -e build/wifi_bt_ble_coex.elf 0x40092d08
0x40092d08: r_assert at /COMPONENT_BT_DIR/controller/esp32/bt.c:1889
$ xtensa-esp32-elf-addr2line -pfiaC -e build/wifi_bt_ble_coex.elf 0x3ffc2084
0x3ffc2084: _l4_intr_stack at hli_vectors.S.obj:?
```

Here is the sdkconfig that was used: [sdkconfig.txt](https://github.com/espressif/esp-idf/files/12335962/sdkconfig.txt)

The board in use: LyraT v4.3

The streaming device: iPhone 6 Plus running iOS 12.5.7

Additionally, one more diff, as I had to turn up the stack size for the input key service (otherwise it reaches stack canary watchpoint when the Mode button is pressed)

```diff
diff --git a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
index 0a90b917..81af6fd9 100644
--- a/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
+++ b/examples/advanced_examples/wifi_bt_ble_coex/main/wifi_bt_ble_coex_example.c
@@ -237,9 +237,11 @@ static void user_a2dp_sink_cb(esp_a2d_cb_event_t event, esp_a2d_cb_param_t *para
         case ESP_A2D_CONNECTION_STATE_EVT:
             if (param->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
                 ESP_LOGI(TAG, "A2DP disconnected");
+                esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                 g_a2dp_connect_state = false;
             } else if (param->conn_stat.state == ESP_A2D_CONNECTION_STATE_CONNECTED) {
                 ESP_LOGI(TAG, "A2DP connected");
+                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
                 g_a2dp_connect_state = true;
             }
             break;
@@ -354,8 +356,8 @@ void app_main(void)
     tcpip_adapter_init();
 #endif
 
-    esp_log_level_set("*", ESP_LOG_WARN);
-    esp_log_level_set(TAG, ESP_LOG_DEBUG);
+    //esp_log_level_set("*", ESP_LOG_WARN);
+    //esp_log_level_set(TAG, ESP_LOG_DEBUG);
     
     ESP_LOGI(TAG, "[ 1 ] Create coex handle for a2dp-gatt-wifi");
     g_coex_handle = (coex_handle_t *)audio_malloc(sizeof(coex_handle_t));
@@ -372,7 +374,7 @@ void app_main(void)
     input_key_service_info_t input_key_info[] = INPUT_KEY_DEFAULT_INFO();
     input_key_service_cfg_t input_cfg = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
     input_cfg.handle = g_coex_handle->set;
-    input_cfg.based_cfg.task_stack = 2048;
+    input_cfg.based_cfg.task_stack = 3072;
     input_cfg.based_cfg.extern_stack = true;
     periph_service_handle_t input_ser = input_key_service_create(&input_cfg);
     input_key_service_add_key(input_ser, input_key_info, INPUT_KEY_NUM);
```

ftab, 2023-08-21:
Again too big to attach directly to GitHub. This time now with a coredump file. Hope that helps.

https://nextcloud.apps.radiosound.com/s/L3JoSorGz4SaKm9

This one was captured with a different LyraT v4.3 board, with a Galaxy A13 phone running Android 12 / OneUI 4.1

I have tried updating to the latest esp-idf to get the latest Bluetooth fixes, but I ran into #12104. I will try with a different example that does not include BLE.

ftab, 2023-08-23:
@xiongweichao I have devised an alternate reproduction scenario. It still takes time and is random, but it's largely hands-off and uses another ESP32 as the source device instead of the uncontrollable whatever-phone.

I have a branch on my fork of esp-adf with all of the previous diffs applied, as well as a couple of small modifications to make two esp-adf examples into an automated test stressing the sink device out to cause a crash.

https://github.com/radiosound-com/esp-adf branch `btdm-crash-testing`

Source device: ESP32-LyraT v4.3 running **[esp-adf/examples/player/pipeline_a2dp_source_stream](https://github.com/radiosound-com/esp-adf/tree/btdm-crash-testing/examples/player/pipeline_a2dp_source_stream)** with a file "test.mp3" on the SD card. (Set dip switches for 1-wire SD card mode) When configuring menuconfig -> example configuration -> remote device name is "ESP-ADF-AUDIO" not "ESP-ADF-SPEAKER" as this is the name used by the other example

Target device: ESP32-LyraT v4.3 running **[esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp](https://github.com/radiosound-com/esp-adf/tree/btdm-crash-testing/examples/get-started/pipeline_a2dp_sink_and_hfp)** (specifically in my branch, it is configured _not_ to terminate the program upon Bluetooth disconnect, just continue running ready for the next Bluetooth connection). I also set the panic handler to halt instead of reboot and turned on coredump in my config. The DIP switches can be set for JTAG since this does not use the SD card. I was about ready to start hooking up a JTAG to it myself, but I would be kind of lost as to what to look for.

[pipeline_a2dp_source_stream](https://github.com/radiosound-com/esp-adf/tree/btdm-crash-testing/examples/player/pipeline_a2dp_source_stream) searches for the specified device name from the example configuration, connects to it, streams test.mp3 from the sd card, then disconnects. In my branch it was modified to `esp_restart()` at the end of the program instead of remaining in an idle state. Therefore, the shorter test.mp3 is, the quicker the connect/disconnect cycle.

(Incidentally, this example seems to crash half the time during its shutdown process--but that's another issue for another day, at least it will reboot itself after the panic)

I have also devised another optional example to serve as a continuous stream from the LyraT source device instead of the repeated connect-stream-disconnect "hammer" test. See **[esp-adf/examples/recorder/pipeline_raw_a2dp](https://github.com/radiosound-com/esp-adf/tree/btdm-crash-testing/examples/recorder/pipeline_raw_a2dp)** in my branch. This searches for the specified device name from the example configuration, connects to it, records from LyraT's line in, and streams it over A2DP.

Please let me know if these are able to help you reproduce. If you do not have another LyraT handy to act as the source device, let me know--I can whip up another quick example that flashes the mp3 file to a SPIFFS partition for a DevKitC or other ESP32 board.

I used a very very short test.mp3 in my case. My thought was, thus, the quicker we arrive at the crash, but, it still took over a day of running it for me to catch one (which I just caught as of starting to write this reply). I don't know if that log would be useful compared to the other logs I've sent, but [here it is](https://github.com/espressif/esp-idf/files/12423878/log.txt.zip). However, the coredump data appears to have been overwritten in this case for some reason, as the subsequent `idf.py coredump-info` output didn't make any sense (invalid exception cause, not the right PC from the panic backtrace, etc).

ftab, 2023-08-28:
I have had a test setup running in a separate location that has not crashed yet. I wonder if it is something specific to this building, or if I'm just getting lucky again.

ftab, 2023-08-30:
I have captured a crash on my custom board using ESP Insights and the backtrace looks a little different

```
Exception IllegalInstructionCause has occured in task IDLE.
Register Values
PC:0x40093318
A0:0x800858c6
A1:0x3ffc0a80
A2:0x3f470c90
A3:0x00000000
A4:0x00000000
A5:0x3f471389
A6:0x00000239
A7:0xfffffffc
A8:0x8000814b
A9:0x3ffc09f0
A10:0x00000000
A11:0x3ffc0a13
A12:0x3ffc09bf
A13:0x00000039
A14:0x00000000
A15:0x3ffc09c4
EXCCAUSE:0x00000000
EXCVADDR:0x00000000
Backtrace
0x40093318
/COMPONENT_BT_DIR/controller/esp32/bt.c:955 in btdm_sleep_check_duration
0x400858c3
r_assert_err
0x40087d45
r_ld_fm_rx_isr
0x4008c022
r_rwbt_isr
0x4008cbff
r_rwbtdm_isr_wrapper
0x40083b4d
/COMPONENT_FREERTOS_DIR/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118 in _xt_lowint1
0x4009350f
/COMPONENT_ESP_HW_SUPPORT_DIR/cpu.c:101 in esp_cpu_wait_for_intr
```

The crash occurred shortly after I connected to the device with my phone (iPhone 14 Pro Max)

xiongweichao, 2023-12-14:
@ftab I'm very sorry to reply to you so late. Please update to the latest v5.0 branch to verify whether the problem still exists.

ftab, 2023-12-15:
> @ftab I'm very sorry to reply to you so late. Please update to the latest v5.0 branch to verify whether the problem still exists.

@xiongweichao We are currently on a fork of v5.1.2 to get task stack in SPI RAM. Is that recent enough?

xiongweichao, 2023-12-18:
@ftab  Yes, this [commit](https://github.com/espressif/esp-idf/commit/22ab7ff325f5c8c5b8eef51896f6ecb5976602b7) should fix the problem.

ftab, 2023-12-19:
@xiongweichao thanks, we will hammer on them this week and see if we run into the crash

ftab, 2024-01-03:
I haven't seen this crash in a while. Tentatively calling it fixed. 🎉

ftab, 2024-01-05:
@xiongweichao I am afraid I spoke too soon 😞 

ASSERT_PARAM(1024 0), in rwbt.c at line 393

The full log is too large to post (came from several hours of test, and has some irrelevant logs from our application every 500ms)

This happened a few seconds after the Bluetooth stream suddenly stopped and the device apparently disconnected unexpectedly

I will dust off one of my previous automated tests on a pair of LyraT boards to continue trying to get a reliable reproduction

ftab, 2024-02-23:
The LyraT setup ran for a while and didn't crash yet, so this log is from my custom board (implements an SPP server, more complex audio pipeline, etc). I have snipped some repetitive errors/logs out for brevity but this is otherwise the full log

```
I (31) boot: ESP-IDF v5.1-1279-g4e88327130 2nd stage bootloader
I (31) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (31) boot.esp32: SPI Speed      : 80MHz
I (31) boot.esp32: SPI Mode       : DIO
I (31) boot.esp32: SPI Flash Size : 16MB
I (32) boot: Enabling RNG early entropy source...
I (32) boot: Partition Table:
I (32) boot: ## Label            Usage          Type ST Offset   Length
I (33) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (33) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (34) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (34) boot:  3 ota_0            OTA app          00 10 00010000 00380000
I (34) boot:  4 ota_1            OTA app          00 11 00390000 00380000
I (35) boot:  5 storage          Unknown data     01 82 00710000 00800000
I (35) boot:  6 fctry            WiFi data        01 02 00f10000 00006000
I (36) boot:  7 coredump         Unknown data     01 03 00f16000 00013000
I (36) boot: End of partition table
I (72) esp_image: segment 0: paddr=00390020 vaddr=3f400020 size=7a7f0h (501744) map
I (224) esp_image: segment 1: paddr=0040a818 vaddr=3ff80063 size=00008h (     8) load
I (225) esp_image: segment 2: paddr=0040a828 vaddr=3ffbdb60 size=057f0h ( 22512) load
I (232) esp_image: segment 3: paddr=00410020 vaddr=400d0020 size=196f74h (1666932) map
I (736) esp_image: segment 4: paddr=005a6f9c vaddr=3ffc3350 size=00a3ch (  2620) load
I (737) esp_image: segment 5: paddr=005a79e0 vaddr=40080000 size=1bac8h (113352) load
I (777) esp_image: segment 6: paddr=005c34b0 vaddr=400c0000 size=00064h (   100) load
I (792) boot: Loaded app from partition at offset 0x390000
I (792) boot: Disabling RNG early entropy source...
I (793) cpu_start: Multicore app
I (793) quad_psram: This chip is ESP32-D0WD
I (793) esp_psram: Found 8MB PSRAM device
I (793) esp_psram: Speed: 80MHz
I (794) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (794) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (794) cpu_start: Pro cpu up.
I (794) cpu_start: Starting app cpu, entry point is 0x400816a8
0x400816a8: call_start_cpu1 at /COMPONENT_ESP_SYSTEM_DIR/port/cpu_start.c:157

I (779) cpu_start: App cpu up.
I (1268) esp_psram: SPI SRAM memory test OK
I (1278) cpu_start: Pro cpu start user code
I (1278) cpu_start: cpu freq: 240000000 Hz
I (1278) cpu_start: Application information:
I (1278) cpu_start: Project name:     Spokane
I (1279) cpu_start: App version:      v0.2.3
I (1279) cpu_start: ELF file SHA256:  fe5ff0de2775a570...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is fa8f887574c19321190a1b76bebf07d48e78b01c6353a2ffa32f21ef80f29895
I (1279) cpu_start: ESP-IDF:          v5.1-1279-g4e88327130
I (1280) cpu_start: Min chip rev:     v3.0
I (1280) cpu_start: Max chip rev:     v3.99 
I (1280) cpu_start: Chip rev:         v3.1
I (1281) heap_init: Initializing. RAM available for dynamic allocation:
I (1281) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1281) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1282) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1282) heap_init: At 3FFC83C8 len 00017C38 (95 KiB): DRAM
I (1282) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1283) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1283) heap_init: At 4009BAC8 len 00004538 (17 KiB): IRAM
I (1284) esp_psram: Adding pool of 4062K of PSRAM memory to heap allocator
I (1285) spi_flash: detected chip: gd
I (1285) spi_flash: flash io: dio
W (1286) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
W (1286) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (1292) coexist: coex firmware version: b6d5e8c
I (1294) app_start: Starting scheduler on CPU0
I (1294) main_task: Started on CPU0
I (1294) app_start: Starting scheduler on CPU1
[... 302 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400936bc: f01d092d 00004136 f01d0000
0x400936bc: btdm_sleep_check_duration at /COMPONENT_BT_DIR/controller/esp32/bt.c:914

Core  0 register dump:
PC      : 0x400936c3  PS      : 0x00060b34  A0      : 0x80085c8a  A1      : 0x3ffbedf4  
0x400936c3: r_assert at /COMPONENT_BT_DIR/controller/esp32/bt.c:1849

A2      : 0x00000000  A3      : 0x00200000  A4      : 0x00000000  A5      : 0x3f43125c  
A6      : 0x00000189  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffbed64  
A10     : 0x00000000  A11     : 0x3ffbed87  A12     : 0x3ffbed33  A13     : 0x00000033  
A14     : 0x00000000  A15     : 0x3ffbed38  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
0x4000c2e0: memcpy in ROM

0x4000c2f6: memcpy in ROM

Backtrace: 0x400936c0:0x3ffbedf4 |<-CORRUPTED
0x400936c0: r_assert at /COMPONENT_BT_DIR/controller/esp32/bt.c:1848

ELF file SHA256: fe5ff0de2775a570
Warning: checksum mismatch between flashed and built applications. Checksum of built application is fa8f887574c19321190a1b76bebf07d48e78b01c6353a2ffa32f21ef80f29895

Rebooting...
```

ftab, 2024-02-23:
Here is a very long log with only a small snippet taken out in the middle of the session as I had exited the monitor temporarily. This time the ELF SHA matches.

Of particular note, I observed FEX 1 8 being printed several times just before the crash.

[log-esp-idf-issue-10555-long.txt](https://github.com/espressif/esp-idf/files/14387285/log-esp-idf-issue-10555-long.txt)

xiongweichao, 2024-02-26:
@ftab Please try if this lib can solve the problem. Please let me know the test results, thanks.
[libbtdm_app_6511e45.a.txt](https://github.com/espressif/esp-idf/files/14403979/libbtdm_app_6511e45.a.txt)

ftab, 2024-02-27:
Thanks. I have installed the library and am checking with both a LyraT and my custom board. I will report back with my findings.

ftab, 2024-03-13:
I have not had the crash in the last couple of weeks, but I also have not had the crash with the production builds either (using the original library in v5.1.2 as per the log a few weeks ago). Continuing to test.

xiongweichao, 2024-07-09:
> @xiongweichao I am afraid I spoke too soon 😞
>
> ASSERT_PARAM(1024 0), in rwbt.c at line 393
>
> The full log is too large to post (came from several hours of test, and has some irrelevant logs from our [...]

This [commit](https://github.com/espressif/esp-idf/commit/8071b02318e6d13143f5ccb4e37ebe1d2b81bb7d) fixes this issue.

closed 2025-03-12 by Alvin1Zhang