espressif/esp-idf #10557: TinyUSB MCS device

Halabdelkrim, 2023-01-17:


esp-saurabhbansal, 2023-01-21:
@Halabdelkrim 
Thanks for posting the query.
To answer your question - Yes, you can use the example examples/peripherals/usb/device)/tusb_msc/ for eMMC also.
Just recently, this example is extended to support sdmmc card. The configuration that is used for using sdmmc card is also applicable to eMMC also. Please refer to the PIN assignments in the README of this example.

Thanks very much.

Halabdelkrim, 2023-01-21:
@esp-saurabhbansal 

Thank you so much for the answer.

Could you please let me know if we can use eMMC 5.1 with the esp32 s3 in this example ?

Can we use 8-line mode to connect the ESP32-S3 with the eMMC card in this example ? 

Thank you.

esp-saurabhbansal, 2023-01-24:
@Halabdelkrim 
Yes, eMMC 5.1 can be used with esp32s3 in this example.
Also, please refer to this link for Supported Speed Modes: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/sdmmc_host.html#supported-speed-modes

It should work.

I must also tell you that we are in the process of reviewing few issues with eMMC and S3:
1. https://github.com/espressif/esp-idf/issues/8257
2. https://github.com/espressif/esp-idf/issues/8521

Halabdelkrim, 2023-01-27:
@esp-saurabhbansal 

Thank you very for the answer.

If i use the ESP32-S3 as MSC device, how much will be the speed of the data transfer between the pc and the eMMC chip.

If i want to increase the bandwidth, should i add an external quad or octal SRAM so i can increase the buffer size? 

Or is it better to add a third party chip as a USB 2.0 eMMC 5.1 MSC bridge, and let the ESP32-S3 only handle the HTTP file serving.

The idea here is to handle the eMMC data online via http file serving using ESP32-S3 and offline via a PC.

Thank you.

esp-saurabhbansal, 2023-02-01:
@Halabdelkrim 
As I understand, you are working with two scenarios:

1. Get data via http to ESP32S3 and copy it in eMMC storage.
eMMC Chip <-> SDCard Adopter <-> ESP32-S3 <HTTP> Web
In this case, once you get the data in ESP32S3 via http, the speed depends on many factors such as line mode (1, 4 or 8), frequency(10MHz, 20MHz, 40MHz) etc.
For Ex - If using 16GB SDHC card with sector size of 512B, 4 line, 20MHz, write/read speed cab be > 7MB/s if writting/reading 64KB of data.
Similarly, with 1 line mode and other configuration remain same, write/read speed cab be > 2MB/s
The performance increases with the write block size. With block size as large as 64KB, 4 line mode and 40 MHz frequency, we could get up to 16 MBytes/s read speed. So, the limitation is card throughput.

2. Get data from PC via USB connection and copy it in eMMC storage.
eMMC Chip <-> SDCard Adopter <-> ESP32-S3 <USB Connection> PC
In this scenario, the speed is something around 355KB/s
It is limited by underlying tinyusb stack.

Halabdelkrim, 2023-02-01:
@esp-saurabhbansal 

Thank you very for the answer.

Yes, i'm designing a custom PCB for this project.

Which options is the best for my case?

Option 1:
PC <-> USB2642 <-> eMMC 5.1 chip <-> ESP32-S3 <-> HTTP

Option 2:
PC <-> ESP32-S3 + eMMC chip <-> HTTP

The idea is to make a device like a USB flash drive (USB stick) but can be accessed from the internet and, or from a PC or laptop.

Thank you.

closed 2023-02-09 by Halabdelkrim

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.

Halabdelkrim, 2023-04-16:
Hello,

Thank you very much for the recommendation!

I will try it asap, and let you know if i have any feedback.