espressif/esp-idf #10559: DHCP enforces classful network masks

xobs, 2023-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2926-g3df87a91a3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

An address such as 10.10.0.1/24 should be acceptable

### What is the actual behavior?

```
I (1188) wifi_manager: MESSAGE: ORDER_LOAD_AND_RESTORE_STA
I (1188) wifi_manager: No saved wifi found on startup. Starting access point.
I (1188) wifi_manager:
I (1198) wifi_manager: MESSAGE: ORDER_START_AP
dhcps: The subnet mask does not match the A address.
E (1208) esp_netif_lwip: DHCP server cannot be started
I (1218) wifi_manager: WIFI_EVENT_AP_START
```

### Steps to reproduce.

1. Start a DHCP server with a netmask of 255.255.255.0 and an IP address of 10.10.0.1

### Debug Logs.

```plain
I (1188) wifi_manager: MESSAGE: ORDER_LOAD_AND_RESTORE_STA
I (1188) wifi_manager: No saved wifi found on startup. Starting access point.
I (1188) wifi_manager:
I (1198) wifi_manager: MESSAGE: ORDER_START_AP
dhcps: The subnet mask does not match the A address.
E (1208) esp_netif_lwip: DHCP server cannot be started
I (1218) wifi_manager: WIFI_EVENT_AP_START
```
```

### More Information.

Classful networks were deprecated [in 1993](https://en.wikipedia.org/wiki/Classful_network). The DHCP server didn't used to care about this, but it has started caring as of a recent change.

xueyunfei998, 2023-01-17:
hi @xobs 

Now the ip address of the dhcp server needs to match the subnet mask.

10.10.0.1  It's a class a address

It is appropriate to set the subnet mask to 255.0.0.0.

xobs, 2023-01-17:
Hi @xueyunfei998,

CIDR-based routing was introduced in 1993 with RFC 1518 and RFC1519, and deprecated class-based networking. This is further reinforced by [RFC 4632 in section 3](https://www.rfc-editor.org/rfc/rfc4632#section-3)

IP addresses haven't had to match subnet masks in over thirty years, so it was weird to encounter this error all of a sudden.

AxelLin, 2023-02-22:
> Hi @xueyunfei998,
>
> CIDR-based routing was introduced in 1993 with RFC 1518 and RFC1519, and deprecated class-based networking. This is further reinforced by [RFC 4632 in section [...]

In https://github.com/espressif/esp-idf/issues/10024#issuecomment-1439398673 , it has below implementation:
```
+#define IS_INVALID_SUBNET_MASK(x)  (((x-1) | x) != 0xFFFFFFFF)
+/* Notes:
+*  CIDR eliminates the traditional Class A, Class B and Class C addresses.
+ */
+#define IP_CLASS_HOST_NUM(mask)            (0xffffffff & ~mask)
+#define DHCP_CHECK_SUBNET_MASK_IP(mask)                                                               \
+    do {                                                                                                  \
+        if (IS_INVALID_SUBNET_MASK(mask)) {                                                               \
+            DHCPS_LOG("dhcps: Illegal subnet mask.\n");                                                   \
+            return ERR_ARG;                                                                               \
+        }                                                                                                 \
+    } while (0)
+
```
BTW, v4.3 branch also needs fix.for cab17c713deb02a14743dea1f9ed3838e7cecdda.

closed 2023-03-03 by espressif-bot

AxelLin, 2023-03-30:
@xueyunfei998 
Unfortunately, v4.3.5 still has this regression even though I already told you v4.3 needs fix a Month ago https://github.com/espressif/esp-idf/issues/10559#issuecomment-1439479977.
Could you please fix it in v4.3 branch?

@igrr

xueyunfei998, 2023-03-30:
hi @AxelLin 
The internal  4.3 mr has been submitted and will be merged soon.