espressif/esp-idf #10560: [Crash][esp32s3][v4.4.3] assert failed: spinlock_release spinlock.h:158 (core_id == lock->owner) when config CPU 240Mhz

zbqxyz, 2023-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

i  config CPU 160Mhz,init wifi ok ,bug change 240Mhz ,when app init wifi ,crash ,

I (737) uart: queue fre
assert failed: spinlock_release spinlock.h:170 (core_id == lock->owner)

Backtrace:0x40379f12:0x3fcabd900x40382d8d:0x3fcabdb0 0x40389cae:0x3fcabdd0 0x40386937:0x3fcabef0 0x4038939d:0x3fcabf10 0x4037a1ce:0x3fcabf30 0x4038479c:0x3fcabf60 0x420ea315:0x3fcabfa0 0x420e9f2d:0x3fcabfd0 0x42016282:0x3fcac010 0x420155ef:0x3fcac030
0x40379f12: panic_abort at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/panic.c:402

0x40382d8d: esp_system_abort at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/esp_system.c:121

0x40389cae: __assert_func at C:/Espressif/frameworks/esp-idf-v4.4/components/newlib/assert.c:85

0x40386937: spinlock_release at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc/spinlock.h:170
 (inlined by) vPortExitCritical at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/port.c:310

0x4038939d: multi_heap_internal_unlock at C:/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:157
 (inlined by) multi_heap_malloc_impl at C:/Espressif/frameworks/esp-idf-v4.4/components/heap/multi_heap.c:204

0x4037a1ce: heap_caps_malloc at C:/Espressif/frameworks/esp-idf-v4.4/components/heap/heap_caps.c:145

0x4038479c: xTaskCreatePinnedToCore at C:/Espressif/frameworks/esp-idf-v4.4/components/freertos/tasks.c:861

0x420ea315: esp_event_loop_create at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_event/esp_event.c:493

0x420e9f2d: esp_event_loop_create_default at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_event/default_event_loop.c:118

0x42016282: initialise_wifi at f:\portable_power-masterv2_1226\build/../components/myhal/blufi_example_main.c:230
 (inlined by) wifi_init at f:\portable_power-masterv2_1226\build/../components/myhal/blufi_example_main.c:530

0x420155ef: net_task at f:\portable_power-masterv2_1226\build/../components/dataprocess/net.c:48

ESP-Marius, 2023-01-29:
Hi, sorry for the late reply!

This sounds like the spinlock structure might have been corrupted. You can try to print out the owner when it fails to see if it gives you any more information, i.e. add the following in `spinlock.h`  before the assert:

```
    if (core_id != lock->owner){
        ESP_EARLY_LOGI("","core_id %u, lock->owner %u", core_id, lock->owner);
    }
  assert(core_id == lock->owner); // This is a mutex we didn't lock, or it's corru
```

Which version of IDF are you running? There was a similar issue with PSRAM a while ago, but the fix for that was merged in 486316f222c97381ed3019603a5cf93d2f538f38

If it is memory corruption then I recommend checking out some tips on how to debug it [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-memory-debugging)

Alvin1Zhang, 2023-02-15:
@zbqxyz Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

ESP-Marius, 2023-02-27:
Closing this issue due to lack of updates, please feel free to re-open it if you are still facing the same issue and want to investigate it further.

closed 2023-02-27 by ESP-Marius