espressif/esp-idf #10561: Using ADC single mode with a Faster clock on esp32s3

5ami, 2023-01-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi All, 
Is there a possibility to use the ADC single conversion with the DIG ADC  
as only The  RTC controllers can be used for single conversion mode.  the RTC controller runs on slow clock which makes the ADC reading super slow.
I am using ESP-IDF v5.0 and esp32s3

suda-morris, 2023-01-18:
The TRMs says
[image]

We use the RTC controller to do the single conversions.

5ami, 2023-01-18:
Is there a way to do a single reading using the continuous ADC mode ? is there is resource from Espressif explaining how the continuous mode works, the example shown in IDF5.0 didn't give stable values , and I couldn't get the idea of using a circular buffer why cant be a simple as triggering the ADC measurement according to the pattern table and read value from a register.

5ami, 2023-02-13:
Hi @suda-morris  also Why is there settings related changing the digital Controllers. 

[esp-idf](https://github.com/espressif/esp-idf)/[components](https://github.com/espressif/esp-idf/tree/master/components)/[hal](https://github.com/espressif/esp-idf/tree/master/components/hal)/adc_oneshot_hal.c

closed 2023-02-16 by 5ami