espressif/esp-idf #10564: spi init fail in `esp32`

Coloryr, 2023-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-1006-g02605f1a31

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32-wroow

### Power Supply used.

USB

### What is the expected behavior?

spi init

### What is the actual behavior?

spi init fail

### Steps to reproduce.

```
spi_bus_config_t buscfg;
  buscfg.miso_io_num = GPIO_NUM_12;
  buscfg.mosi_io_num = GPIO_NUM_13;
  buscfg.sclk_io_num = GPIO_NUM_14;
  buscfg.quadwp_io_num = -1;
  buscfg.quadhd_io_num = -1;
  buscfg.max_transfer_sz = 64;

  // Initialize the SPI bus
  esp_err_t ret = spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO);
  ESP_ERROR_CHECK(ret);

  spi_device_interface_config_t devcfg;
  devcfg.clock_speed_hz = SPI_MASTER_FREQ_10M; // Clock out at 10 MHz,
  devcfg.mode = 0;    // SPI mode 0
  devcfg.cs_ena_pretrans = 0;
  devcfg.spics_io_num = -1;
  devcfg.queue_size = 7;

  ret = spi_bus_add_device(SPI2_HOST, &devcfg, &spi);
  ESP_ERROR_CHECK(ret);
```

### Debug Logs.

```plain
[0;31mE (372) spi_hal: spi_hal_cal_clock_conf(94): When work in full-duplex mode at frequency > 0.0MHz, device cannot read correct data.
Try to use IOMUX pins to increase the frequency limit, or use the half duplex mode.
Please note the SPI master can only work at divisors of 80MHz, and the driver always tries to find the closest frequency to your configuration.
Specify ``SPI_DEVICE_NO_DUMMY`` to ignore this checking. Then you can output data at higher speed, or read data at your own risk.[0m
[0;31mE (422) spi_master: spi_bus_add_device(372): assigned clock speed not supported[0m
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x400d6851
```
```

Coloryr, 2023-01-18:
```
[0;31mE (373) spi: spi_bus_initialize(766): intr flag not allowed[0m
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400d5c23
```
with
```
PS E:\workload\esp\esp-idf> git describe --tags
v5.1-dev-2926-g3df87a91a3
```

suda-morris, 2023-01-18:
> spi_bus_config_t buscfg;

I think this is where introduced the issue. `buscfg` is not initiated with proper "default" values. You can do that in this way:

```
spi_bus_config_t buscfg = {};
```

Coloryr, 2023-01-18:
> > spi_bus_config_t buscfg;
> 
> I think this is where introduced the issue. `buscfg` is not initiated with proper "default" values. You can do that in this way:
> 
> ```
> spi_bus_config_t buscfg = {};
> ```

yes, it work, thanks

closed 2023-01-18 by Coloryr