espressif/esp-idf #10565: `GPIO_NUM_12` can't set level, but `GPIO_NUM_16` can

Coloryr, 2023-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-494-g490216a2ac

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32_devkite_v4

### Power Supply used.

USB

### What is the expected behavior?

gpio can set level

### What is the actual behavior?

gpio set level fail

it is my code error?????????

### Steps to reproduce.

```
void app_main(void)
{
    ESP_LOGI(TAG, "Main start");
    //key_init();
    //beep_init();

    //beep_start();

    //xTaskCreate(task_key_run, "task_key", 2048, NULL, 3, NULL);

    ESP_LOGI(TAG, "Main init done");

    gpio_set_direction(GPIO_NUM_12, GPIO_MODE_OUTPUT);

    while (1)
    {
        gpio_set_level(GPIO_NUM_12, 1);
        ESP_LOGI(TAG, "LED On");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        gpio_set_level(GPIO_NUM_12, 0);
        ESP_LOGI(TAG, "LED Off");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

```

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7008
load:0x40078000,len:15408
load:0x40080400,len:3840
entry 0x4008064c
[0;33mW (83) rtc_clk_init: Potentially bogus XTAL frequency: 35 MHz, guessing 40 MHz[0m
[0;32mI (37) boot: ESP-IDF v5.0-494-g490216a2ac 2nd stage bootloader[0m
[0;32mI (37) boot: compile time 14:35:12[0m
[0;32mI (37) boot: chip revision: v1.0[0m
[0;32mI (41) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (48) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (53) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (58) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (62) boot: Enabling RNG early entropy source...[0m
[0;32mI (68) boot: Partition Table:[0m
[0;32mI (71) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (86) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (93) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (101) boot: End of partition table[0m
[0;32mI (105) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=085ach ( 34220) map[0m
[0;32mI (133) esp_image: segment 1: paddr=000185d4 vaddr=3ffb0000 size=01ec4h (  7876) load[0m
[0;32mI (137) esp_image: segment 2: paddr=0001a4a0 vaddr=40080000 size=05b78h ( 23416) load[0m
[0;32mI (150) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1601ch ( 90140) map[0m
[0;32mI (182) esp_image: segment 4: paddr=00036044 vaddr=40085b78 size=05c0ch ( 23564) load[0m
[0;32mI (198) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (198) boot: Disabling RNG early entropy source...[0m
[0;32mI (210) cpu_start: Pro cpu up.[0m
[0;32mI (210) cpu_start: Starting app cpu, entry point is 0x40081048[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (224) cpu_start: Pro cpu start user code[0m
[0;32mI (224) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (224) cpu_start: Application information:[0m
[0;32mI (229) cpu_start: Project name:     blink[0m
[0;32mI (234) cpu_start: App version:      1[0m
[0;32mI (238) cpu_start: Compile time:     Jan 18 2023 16:05:29[0m
[0;32mI (244) cpu_start: ELF file SHA256:  c8f8b555ea7599c2...[0m
[0;32mI (250) cpu_start: ESP-IDF:          v5.0-494-g490216a2ac[0m
[0;32mI (257) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (270) heap_init: At 3FFB27B8 len 0002D848 (182 KiB): DRAM[0m
[0;32mI (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (289) heap_init: At 4008B784 len 0001487C (82 KiB): IRAM[0m
[0;32mI (296) spi_flash: detected chip: generic[0m
[0;32mI (300) spi_flash: flash io: dio[0m
[0;33mW (303) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (317) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (327) main: Main start[0m
[0;32mI (327) main: Main init done[0m
[0;32mI (337) main: LED On[0m
[0;32mI (1337) main: LED Off[0m
[0;32mI (2337) main: LED On[0m
[0;32mI (3337) main: LED Off[0m
[0;32mI (4337) main: LED On[0m

```
```

Coloryr, 2023-01-18:
https://user-images.githubusercontent.com/26276037/213125150-49c498fe-bc33-45d3-bf75-9de4e95f240d.mp4

igrr, 2023-01-18:
By default, GPIO12 is configured as a JTAG pin. You need to [configure it](https://github.com/espressif/esp-idf/blob/8bf18b76ac18e80c4d49e945147ec95f31ad3d4f/examples/peripherals/gpio/generic_gpio/main/gpio_example_main.c#L67-L89) as a GPIO before setting the level, using `gpio_config` or `gpio_reset_pin` functions.

Coloryr, 2023-01-18:
> By default, GPIO12 is configured as a JTAG pin. You need to [configure [...]

thanks, it work

closed 2023-01-18 by Coloryr