espressif/esp-idf #10566: esp_http_client does not handle URL with space

AxelLin, 2023-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.4-332-g56ee43ec5e

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

esp_http_client should either provide API to do urlencode or handle it internally.

### What is the actual behavior?

If the URL has space, it shows error:
e.g.

E (5419) HTTP_CLIENT: Error parse url http://192.168.0.104/test space/test
E (5426) HTTP_CLIENT: Failed to set URL

### Steps to reproduce.

You can test using esp_http_client_example, just need to modify the URL setting to include space.

kriegste, 2023-01-18:
Spaces are no valid characters in URLs. They have to be encoded using %20

AxelLin, 2023-01-18:
> Spaces are no valid characters in URLs. They have to be encoded using %20

I know that.
maybe esp_http_client can provide API to encode it. (Otherwise people have to open-coded the same API)

laukik-hase, 2023-01-29:
Closed with https://github.com/espressif/esp-idf/commit/167618d6a4e2cfcaab967ba2ba920df64fb4795d

closed 2023-01-29 by laukik-hase