espressif/esp-idf #10567: ESP32 C3 clang15.0.0 build error with option "-g"

Jiu-xiao, 2023-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2926-g3df87a91a3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

我使用idf-as-lib的例程，但是将main.c更改为main.cpp。然后使用./build-esp32c3.sh编译，并且将脚本中的
`PARAM="-DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-${TARGET}.cmake -DTARGET=${TARGET} -GNinja"`
更改为
`PARAM="-DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-clang-${TARGET}.cmake -DTARGET=${TARGET} -GNinja"`，编译成功，并没有出现问题。

### What is the actual behavior?

但是我在我的CMakeLists.txt中加入`add_compile_options(-g)`之后，编译出现错误。我对比了两次编译生成的compile_commands.json文件，只有-g这一个参数有区别

### Steps to reproduce.

复制官方的idf-as-lib工程，将main.c重命名为main.cpp。为app_main函数添加extern “C”修饰，然后将build.sh中的`toolchain-${TARGET}.cmake`更改为`toolchain-clang-${TARGET}.cmake`。运行./build-esp32c3.sh进行编译

### Build or installation Logs.

```plain
╰─$ ./build-esp32c3.sh
-- The C compiler identification is Clang 15.0.0
-- The CXX compiler identification is Clang 15.0.0
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /home/xiao/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/xiao/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/xiao/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Component directory /home/xiao/esp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- Checking Python dependencies...
Python requirements are satisfied.
Constraint file: /home/xiao/.espressif/espidf.constraints.v5.1.txt
Requirement files:
 - /home/xiao/esp/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/xiao/.espressif/python_env/idf5.1_py3.10_env/bin/python
-- Project sdkconfig file /home/xiao/code/idf_as_lib/sdkconfig
CMake Warning at /home/xiao/esp/esp-idf/components/esp_common/project_include.cmake:14 (message):
  Building ESP-IDF with clang is an experimental feature and is not yet
  officially supported.
Call Stack (most recent call first):
  /home/xiao/esp/esp-idf/tools/cmake/build.cmake:400 (include)
  /home/xiao/esp/esp-idf/tools/cmake/build.cmake:622 (__build_process_project_includes)
[... 168 lines cut ...]
esptool.py v4.5.dev2
Creating esp32c3 image...
Merged 1 ELF section
Successfully created esp32c3 image.
Generated /home/xiao/code/idf_as_lib/build/bootloader/bootloader.bin
[94/94] cd /home/xiao/code/idf_as_lib/build/bootloader/esp-idf/esptool_py && python /home/xiao/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/xiao/code/idf_as_lib/build/bootloader/bootloader.bin
Bootloader binary size 0x4ef0 bytes. 0x3110 bytes (38%) free.
[431/434] Building CXX object CMakeFiles/idf_as_lib.elf.dir/main.cpp.obj
FAILED: CMakeFiles/idf_as_lib.elf.dir/main.cpp.obj 
/home/xiao/.espressif/tools/esp-clang/15.0.0-23786128ae/esp-clang/bin/clang++  -I/home/xiao/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/xiao/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/xiao/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/xiao/esp/esp-idf/components/freertos/esp_additions/include -I/home/xiao/code/idf_as_lib/build/config -I/home/xiao/esp/esp-idf/components/newlib/platform_include -I/home/xiao/esp/esp-idf/components/esp_hw_support/include -I/home/xiao/esp/esp-idf/components/esp_hw_support/include/soc -I/home/xiao/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/home/xiao/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/home/xiao/esp/esp-idf/components/heap/include -I/home/xiao/esp/esp-idf/components/log/include -I/home/xiao/esp/esp-idf/components/soc/include -I/home/xiao/esp/esp-idf/components/soc/esp32c3/. -I/home/xiao/esp/esp-idf/components/soc/esp32c3/include -I/home/xiao/esp/esp-idf/components/hal/esp32c3/include -I/home/xiao/esp/esp-idf/components/hal/include -I/home/xiao/esp/esp-idf/components/hal/platform_port/include -I/home/xiao/esp/esp-idf/components/esp_rom/include -I/home/xiao/esp/esp-idf/components/esp_rom/include/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_rom/esp32c3 -I/home/xiao/esp/esp-idf/components/esp_common/include -I/home/xiao/esp/esp-idf/components/esp_system/include -I/home/xiao/esp/esp-idf/components/esp_system/port/soc -I/home/xiao/esp/esp-idf/components/esp_system/port/include/riscv -I/home/xiao/esp/esp-idf/components/esp_system/port/include/private -I/home/xiao/esp/esp-idf/components/riscv/include -I/home/xiao/esp/esp-idf/components/spi_flash/include --target=riscv32-esp-elf -march=rv32imc -mabi=ilp32  -g -std=gnu++17 -MD -MT CMakeFiles/idf_as_lib.elf.dir/main.cpp.obj -MF CMakeFiles/idf_as_lib.elf.dir/main.cpp.obj.d -o CMakeFiles/idf_as_lib.elf.dir/main.cpp.obj -c /home/xiao/code/idf_as_lib/main.cpp
/tmp/main-c6744c.s: Assembler messages:
/tmp/main-c6744c.s:5: Error: file number less than one
/tmp/main-c6744c.s:6: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:7: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:8: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:9: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:10: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:11: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:12: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:13: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:14: Error: junk at end of line, first unrecognized character is `"'
/tmp/main-c6744c.s:15: Error: junk at end of line, first unrecognized character is `"'
clang-15: error: assembler command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

gerekon, 2023-01-26:
Hi @Jiu-xiao.

Could you explain in English how to reproduce the problem? What application do you build?
What does you script `./build-esp32c3.sh` actually do?

Jiu-xiao, 2023-01-26:
`main.cpp`

```cpp
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_system.h"
#include "esp_flash.h"

extern "C" void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    uint32_t flash_size;
    esp_chip_info(&chip_info);
    printf("This is ESP32 chip with %d CPU cores, WiFi%s%s, ",
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);
    if(esp_flash_get_size(NULL, &flash_size) != ESP_OK) {
        printf("Get flash size failed");
        return;
    }

    printf("%dMB %s flash\n", flash_size / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

```

Jiu-xiao, 2023-01-26:
`CMakeLists.txt`
```cpp
cmake_minimum_required(VERSION 3.16)

project(idf_as_lib LANGUAGES C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_C_STANDARD_REQUIRED ON)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

set(targets "esp32" "esp32s2" "esp32s3" "esp32c3" "esp32h2" "esp32c2")

add_compile_options(--target=riscv32-esp-elf -march=rv32imc -mabi=ilp32 -gdwarf-4)

if("${TARGET}" IN_LIST targets)
    # Include for ESP-IDF build system functions
    include($ENV{IDF_PATH}/tools/cmake/idf.cmake)
    # Create idf::{target} and idf::freertos static libraries
    idf_build_process("${TARGET}"
                    # try and trim the build; additional components
                    # will be included as needed based on dependency tree
                    #
                    # although esptool_py does not generate static library,
                    # processing the component is needed for flashing related
                    # targets and file generation
                    COMPONENTS freertos esptool_py
                    SDKCONFIG ${CMAKE_CURRENT_LIST_DIR}/sdkconfig
                    BUILD_DIR ${CMAKE_BINARY_DIR})
else()
    message(WARNING "Unknown target ${TARGET}, creating stubs for esp32 instead")
    # Create stubs for esp32 and freertos, stub::esp32 and stub::freertos
    add_subdirectory(stubs/esp32)
    add_subdirectory(stubs/freertos)
    add_subdirectory(stubs/spi_flash)
endif()

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(elf_file ${CMAKE_PROJECT_NAME}.elf)
add_executable(${elf_file} main.cpp)

# Link the static libraries to the executable
if("${TARGET}" IN_LIST targets)
    target_link_libraries(${elf_file} idf::freertos idf::spi_flash)
    # Attach additional targets to the executable file for flashing,
    # linker script generation, partition_table generation, etc.
    idf_build_executable(${elf_file})
else()
    target_link_libraries(${elf_file} stub::esp32 stub::freertos stub::spi_flash)
endif()
```

Jiu-xiao, 2023-01-28:
> Hi @Jiu-xiao.
> 
> Could you explain in English how to reproduce the problem? What application do you build? What does you script `./build-esp32c3.sh` actually do?

If I add `add_compile_options(-g)`  in CMakeLists.txt, an error will be reported when I use clang to compile.

gerekon, 2023-01-30:
Looks like you build `hello_world` example using your custom build script.
I recommend you to use `idf.py` for building your app.
```
IDF_TOOLCHAIN=clang idf.py set-target esp32c3
IDF_TOOLCHAIN=clang idf.py build
```
The commands above pick up compiler settings from https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-clang-esp32c3.cmake

gerekon, 2023-01-30:
I checked it. It has no problems with your `main.cpp`

gerekon, 2023-01-30:
> If I add add_compile_options(-g) in CMakeLists.txt, an error will be reported when I use clang to compile.

Unfortunately I can not build app with your `CMakefile.txt`. Looks like it does not work well with `idf.py build`. But when I use [the commands above](https://github.com/espressif/esp-idf/issues/10567#issuecomment-1409129697) with ` -gdwarf-4` added to [CMAKE_C_FLAGS](https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-clang-esp32c3.cmake#L13), [CMAKE_CXX_FLAGS](https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-clang-esp32c3.cmake#L20) and [CMAKE_ASM_FLAGS](https://github.com/espressif/esp-idf/blob/master/tools/cmake/toolchain-clang-esp32c3.cmake#L27) it works well. Adding `-g` alone does not work because Clang 15 uses DWARF5 by default by GNU `as` has problem with. Our current Clang toolchain uses `riscv32-esp-elf-as` as assembler, because intergrated one is not well tested yet with IDF code.

Jiu-xiao, 2023-01-31:
Fixed