espressif/esp-idf #10568: esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c:778:1: internal compiler error: Segmentation fault

babadue, 2023-01-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

compiled with no error.

### What is the actual behavior?

Instead it stopped and telling me to:

"Please submit a full bug report"

### Steps to reproduce.

Just build my project.  The error seemed to be in the esp-idf components:

C:/Users/uuumm/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c: In function 'rgb_panel_draw_bitmap':
C:/Users/uuumm/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c:778:1: internal compiler error: Segmentation fault

### Build or installation Logs.

```plain
[783/1544] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_rgb_panel.c.obj

idf/components/esp_rom/include/esp32s3 -IC:/Users/uuumm/esp-idf/components/esp_rom/esp32s3 -IC:/Users/uuumm/esp-idf/components/esp_common/include -IC:/Users/uuumm/esp-idf/components/esp_system/include -IC:/Users/uuumm/esp-idf/components/esp_system/port/soc -IC:/Users/uuumm/esp-idf/components/esp_system/port/include/private -IC:/Users/uuumm/esp-idf/components/xtensa/include -IC:/Users/uuumm/esp-idf/components/xtensa/esp32s3/include -IC:/Users/uuumm/esp-idf/components/lwip/include -IC:/Users/uuumm/esp-idf/components/lwip/include/apps -IC:/Users/uuumm/esp-idf/components/lwip/include/apps/sntp -IC:/Users/uuumm/esp-idf/components/lwip/lwip/src/include -IC:/Users/uuumm/esp-idf/components/lwip/port/esp32/include -IC:/Users/uuumm/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/uuumm/esp-idf/components/driver/include -IC:/Users/uuumm/esp-idf/components/driver/deprecated -IC:/Users/uuumm/esp-idf/components/driver/esp32s3/include -IC:/Users/uuumm/esp-idf/components/esp_pm/include -IC:/Users/uuumm/esp-idf/components/esp_ringbuf/include -IC:/Users/uuumm/esp-idf/components/esp_psram/include -mlongcalls  -fdiagnostics-color=always -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=C:/Users/uuumm/Projects_esp/my_project=. -fmacro-prefix-map=C:/Users/uuumm/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_rgb_panel.c.obj -MF esp-idf\esp_lcd\CMakeFiles\__idf_esp_lcd.dir\src\esp_lcd_rgb_panel.c.obj.d -o esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_rgb_panel.c.obj -c C:/Users/uuumm/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c
during RTL pass: ira
C:/Users/uuumm/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c: In function 'rgb_panel_draw_bitmap':
C:/Users/uuumm/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c:778:1: internal compiler error: Segmentation fault
  778 | }
      | ^
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
[785/1544] Building C object esp-idf/protobuf-c/CMakeFiles/__idf_protobuf-c.dir/protobuf-c/protobuf-c/protobuf-c.c.obj
ninja: build stopped: subcommand failed.
```

suda-morris, 2023-01-19:
This seems like not a bug in the driver, but an environment or toolchain error. I wonder how you run the build command? Are you using the `idf.py build` from your terminal?

closed 2023-01-20 by babadue

JeremiahGillis, 2023-08-04:
This issue should be reopened. I just started receiving this code today building in VS Code. I have never seen this issue prior to cleaning out my clearing out my ESP-IDF directory, ESP-IDF Tools directory, and installing ESP-IDF v5.1 and toolchain via Configure ESP-IDF extension in VS Code. I am building ESP32-S3 project using the `ESP-IDF Build Project` button in the `Espressif IDF (Extension)` bar.

I have to clean the project and try to build again. Sometimes the build will be successful, sometimes it will fail again for the same error. I see no consistent way to prevent the error, or make it trigger every build.

```
[834/1483] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_rgb.c.obj
FAILED: esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_rgb.c.obj 
C:\Users\jerem\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/jerem/Source/Repos/cl-touchscreen/touchscreen/build/config -IC:/Users/jerem/esp/esp-idf/components/esp_lcd/include -IC:/Users/jerem/esp/esp-idf/components/esp_lcd/interface -IC:/Users/jerem/esp/esp-idf/components/newlib/platform_include -IC:/Users/jerem/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/jerem/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/jerem/esp/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/jerem/esp/esp-idf/components/freertos/esp_additions/include -IC:/Users/jerem/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -IC:/Users/jerem/esp/esp-idf/components/esp_hw_support/include -IC:/Users/jerem/esp/esp-idf/components/esp_hw_support/include/soc -IC:/Users/jerem/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 
-IC:/Users/jerem/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/jerem/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -IC:/Users/jerem/esp/esp-idf/components/heap/include -IC:/Users/jerem/esp/esp-idf/components/log/include -IC:/Users/jerem/esp/esp-idf/components/soc/include -IC:/Users/jerem/esp/esp-idf/components/soc/esp32s3 -IC:/Users/jerem/esp/esp-idf/components/soc/esp32s3/include -IC:/Users/jerem/esp/esp-idf/components/hal/esp32s3/include -IC:/Users/jerem/esp/esp-idf/components/hal/include -IC:/Users/jerem/esp/esp-idf/components/hal/platform_port/include -IC:/Users/jerem/esp/esp-idf/components/esp_rom/include -IC:/Users/jerem/esp/esp-idf/components/esp_rom/include/esp32s3 -IC:/Users/jerem/esp/esp-idf/components/esp_rom/esp32s3 -IC:/Users/jerem/esp/esp-idf/components/esp_common/include -IC:/Users/jerem/esp/esp-idf/components/esp_system/include -IC:/Users/jerem/esp/esp-idf/components/esp_system/port/soc -IC:/Users/jerem/esp/esp-idf/components/esp_system/port/include/private -IC:/Users/jerem/esp/esp-idf/components/xtensa/include -IC:/Users/jerem/esp/esp-idf/components/xtensa/esp32s3/include -IC:/Users/jerem/esp/esp-idf/components/lwip/include -IC:/Users/jerem/esp/esp-idf/components/lwip/include/apps -IC:/Users/jerem/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/jerem/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/jerem/esp/esp-idf/components/lwip/port/include -IC:/Users/jerem/esp/esp-idf/components/lwip/port/freertos/include -IC:/Users/jerem/esp/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/jerem/esp/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/jerem/esp/esp-idf/components/driver/include -IC:/Users/jerem/esp/esp-idf/components/driver/deprecated -IC:/Users/jerem/esp/esp-idf/components/driver/analog_comparator/include -IC:/Users/jerem/esp/esp-idf/components/driver/dac/include -IC:/Users/jerem/esp/esp-idf/components/driver/gpio/include -IC:/Users/jerem/esp/esp-idf/components/driver/gptimer/include -IC:/Users/jerem/esp/esp-idf/components/driver/i2c/include -IC:/Users/jerem/esp/esp-idf/components/driver/i2s/include -IC:/Users/jerem/esp/esp-idf/components/driver/ledc/include -IC:/Users/jerem/esp/esp-idf/components/driver/mcpwm/include -IC:/Users/jerem/esp/esp-idf/components/driver/parlio/include -IC:/Users/jerem/esp/esp-idf/components/driver/pcnt/include -IC:/Users/jerem/esp/esp-idf/components/driver/rmt/include -IC:/Users/jerem/esp/esp-idf/components/driver/sdio_slave/include -IC:/Users/jerem/esp/esp-idf/components/driver/sdmmc/include -IC:/Users/jerem/esp/esp-idf/components/driver/sigma_delta/include -IC:/Users/jerem/esp/esp-idf/components/driver/spi/include -IC:/Users/jerem/esp/esp-idf/components/driver/temperature_sensor/include -IC:/Users/jerem/esp/esp-idf/components/driver/touch_sensor/include -IC:/Users/jerem/esp/esp-idf/components/driver/twai/include -IC:/Users/jerem/esp/esp-idf/components/driver/uart/include -IC:/Users/jerem/esp/esp-idf/components/driver/usb_serial_jtag/include -IC:/Users/jerem/esp/esp-idf/components/driver/touch_sensor/esp32s3/include -IC:/Users/jerem/esp/esp-idf/components/esp_pm/include -IC:/Users/jerem/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/jerem/esp/esp-idf/components/esp_mm/include -IC:/Users/jerem/esp/esp-idf/components/esp_psram/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum- [... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

```

JeremiahGillis, 2023-08-13:
Here is my config from another computer before I updated to ESP-IDF v5.1. I never encountered this issue with v5.0. I am going to update this laptop and see if the same issue happens.

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python311\Scripts\;C:\Python311\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files\WireGuard\;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;C:\Program Files\Go\bin;C:\Program Files\SEGGER\JLink;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\go\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.80.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\Source\Repos\Zenith\zti-touchscreen 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
[... cut ...]
Check Debug AdapterPython packages C:<HOMEPATH>\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

```

HamzaHajeir, 2023-11-03:
Any update/solution?

I'm encountering it, it seems similar to @JeremiahGillis's one: ESP32-S3 with IDF v5.1

TobiasUhmannVoss, 2023-11-09:
I encounter this error with IDF 5.0.3 since a few days. The problem usually disappears after some attempts to restart VS Code, clearing the build cache or opening the config in VS Code.

JeremiahGillis, 2023-11-09:
I have not encountered the error on my laptop.

I also did a fresh install of Windows 11 on my desktop, same computer as my post above, so I installed the ESP-IDF again which would have also installed the latest ESP tools. I have not encountered this issue again with v5.1.

Maybe you can try to update the ESP-IDF Tools to see if that changes your outcome. I not sure how you officially do it, I delete the folder and use the VS Code ESP-IDF config tool to install them again.

Dadigno, 2024-12-06:
Same development environment and same issue for me, but with IDF v5.3.0.
The same error you described appears seemingly at random. Sometimes it can be resolved with idf.py fullclean or by manually deleting the CMakeCache.txt file

brianignacio5, 2024-12-06:
You are using an old version of the esp-idf vscode extension. Could you try updating the extension and reinstall the esp-idf. You can use the `ESP-IDF: Configure ESP-IDF extension` to find existing esp-idf install in your system.

daleffe, 2024-12-30:
Hi, 

I'm using 5.3.0 version and solved it by including ```set(EXCLUDE_COMPONENTS esp_lcd)``` at root *CMakeLists.txt* file.

It's workaround since i don't use LCD.

hamidsh07, 2025-02-05:
Hi, please add the following line to your CMakeLists.txt file:

set_source_files_properties(${CMAKE_SOURCE_DIR}/components/esp_lcd/rgb/esp_lcd_panel_rgb.c PROPERTIES HEADER_FILE_ONLY TRUE)

daanpape, 2025-10-25:
For what it's worth, I had this on a clean build with ESP-IDF v5.4.3 on Windows 11 when using the VSCode extension to run the build command. Doing a clean build solved the issue but it returns at times.

beelsebob, 2025-12-04:
Hitting this on ESP-IDF 5.5.0.  Started occuring when I enabled -fkeep-inlined-functions.