espressif/esp-idf #10571: ESP32-S3 USB host transfer IN problem.

marbalon, 2023-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-427-g4532e6e0b2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board.

### Power Supply used.

USB

### What is the expected behavior?

Two ways communication with USB Thermal POS printer.

### What is the actual behavior?

Everything is described here:

https://www.esp32.com/viewtopic.php?f=2&t=31554&p=108456#p108456

But the most important is in the last Edit3.

Output transmission to printer something is working fine. I can print files with sizes 200-300kb easily.

But I need to query the printer for paper status or case open/close. I'm sending the correct commands (three bytes only 0x10, 0x04, 0x04) and I should get one byte as the response - 0x12. But transfer callback mostly comes back with actual_num_bytes.

Mostly means once per maybe 10 restarts I got it working as expected. The printer sends the answer to every query, but it is hard to gues when it is working or not. Not related to printer restart/plug/unplug.

What I found is that the printer receives these commands correctly and buffers this answer on the printer side, because when I restart the board and send a query for paper status I'm getting multiple buffered answers for frames sent before restart like 0x12 0x12 0x12 0x12 0x12 0x12 ...

So it looks like ESP32 transfer IN not working properly for some reason.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

I tried:
IDF 4.4
IDF 5.0
VisualCode + Platformio + Espressif 32 plugin 6.0

Always have the same scenario, sometimes a got it working, but most of the time it is not working and I have no idea how to make it work stable.

marbalon, 2023-02-14:
Thanks, answered on the forum.

Alvin1Zhang, 2023-02-15:
Thanks for sharing the updates, let us track on the forum.

closed 2023-02-15 by Alvin1Zhang