espressif/esp-idf #10573: Reading coredump information from flash is verbose

cbaechler, 2023-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-321-g38961260d8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

Verbosity of logs when executing `esp_core_dump_get_summary` should be configurable by setting the level with `esp_log_level_set`. 

Is there any way to silence the logs when reading the coredump partition? 

### What is the actual behavior?

Reading coredump partition produces a big amount of logs with TAGs `esp_core_dump_elf` and `esp_core_dump_port`. The output looks like an actual fault is just happening and can confuse the developer. 

### Steps to reproduce.

1. Enable coredump to flash feature and add corresponding partition
2. Trigger a hardfault condition to fill the coredump partition
3. Read the summary from the partition with `esp_core_dump_get_summary`
4. Observe log output

KonssnoK, 2023-01-19:
Adding example core dump
```
D (00:00:36.928) modem_comm: IP event 0
I (00:00:36.929) network: Network connected
I (00:00:36.934) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (00:00:38.468) mqtt_app: MQTT_EVENT_CONNECTED
I (00:00:38.475) mqtt_app: sent publish announce returned msg_id=7736
D (3741) esp_core_dump_elf: ELF ident 7f E L F
D (3741) esp_core_dump_elf: Ph_num 46 offset 34
D (3746) esp_core_dump_elf: PHDR type 4 off 5f4 vaddr 0 paddr 0 filesz 3440 memsz 3440 flags 6 align 0
D (3755) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3762) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3768) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3775) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3781) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3788) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3794) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3801) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3807) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3814) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3820) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3827) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3833) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3840) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3846) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3853) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3859) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3866) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3872) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3879) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3885) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3892) esp_core_dump_elf: Note NameSZ 8 DescSZ 24c Type 1 name CORE
D (3898) esp_core_dump_elf: PHDR type 1 off 3a34 vaddr 3fcbbea8 paddr 3fcbbea8 filesz 158 memsz 158 flags 6 align 0
D (3909) esp_core_dump_elf: PHDR type 1 off 3b8c vaddr 3fccecc0 paddr 3fccecc0 filesz 870 memsz 870 flags 6 align 0
D (3919) esp_core_dump_elf: PHDR type 1 off 43fc vaddr 3fcd20c0 paddr 3fcd20c0 filesz 158 memsz 158 flags 6 align 0
D (3930) esp_core_dump_elf: PHDR type 1 off 4554 vaddr 3fcd1a50 paddr 3fcd1a50 filesz 660 memsz 660 flags 6 align 0
D (3940) esp_core_dump_elf: PHDR type 1 off 4bb4 vaddr 3fcc93e4 paddr 3fcc93e4 filesz 158 memsz 158 flags 6 align 0
D (3951) esp_core_dump_elf: PHDR type 1 off 4d0c vaddr 3fcc90b0 paddr 3fcc90b0 filesz 320 memsz 320 flags 6 align 0
D (3961) esp_core_dump_elf: PHDR type 1 off 502c vaddr 3fcf40ac paddr 3fcf40ac filesz 158 memsz 158 flags 6 align 0
D (3972) esp_core_dump_elf: PHDR type 1 off 5184 vaddr 3fcf3d50 paddr 3fcf3d50 filesz 350 memsz 350 flags 6 align 0
D (3982) esp_core_dump_elf: PHDR type 1 off 54d4 vaddr 3fcf480c paddr 3fcf480c filesz 158 memsz 158 flags 6 align 0
D (3993) esp_core_dump_elf: PHDR type 1 off 562c vaddr 3fcf4520 paddr 3fcf4520 filesz 2e0 memsz 2e0 flags 6 align 0
D (4003) esp_core_dump_elf: PHDR type 1 off 590c vaddr 3fcf4f6c paddr 3fcf4f6c filesz 158 memsz 158 flags 6 align 0
D (4014) esp_core_dump_elf: PHDR type 1 off 5a64 vaddr 3fcf4ce0 paddr 3fcf4ce0 filesz 280 memsz 280 flags 6 align 0
D (4024) esp_core_dump_elf: PHDR type 1 off 5ce4 vaddr 3fcecfa0 paddr 3fcecfa0 filesz 158 memsz 158 flags 6 align 0
D (4035) esp_core_dump_elf: PHDR type 1 off 5e3c vaddr 3fceccb0 paddr 3fceccb0 filesz 2e0 memsz 2e0 flags 6 align 0
D (4045) esp_core_dump_elf: PHDR type 1 off 611c vaddr 3fcc30fc paddr 3fcc30fc filesz 158 memsz 158 flags 6 align 0
D (4056) esp_core_dump_elf: PHDR type 1 off 6274 vaddr 3fcc2e20 paddr 3fcc2e20 filesz 2d0 memsz 2d0 flags 6 align 0
D (4066) esp_core_dump_elf: PHDR type 1 off 6544 vaddr 3fcc4dd4 paddr 3fcc4dd4 filesz 158 memsz 158 flags 6 align 0
D (4077) esp_core_dump_elf: PHDR type 1 off 669c vaddr 3fcc49c0 paddr 3fcc49c0 filesz 400 memsz 400 flags 6 align 0
D (4087) esp_core_dump_elf: PHDR type 1 off 6a9c vaddr 3fcc6df4 paddr 3fcc6df4 filesz 158 memsz 158 flags 6 align 0
D (4098) esp_core_dump_elf: PHDR type 1 off 6bf4 vaddr 3fcc6b20 paddr 3fcc6b20 filesz 2c0 memsz 2c0 flags 6 align 0
D (4108) esp_core_dump_elf: PHDR type 1 off 6eb4 vaddr 3fceeae0 paddr 3fceeae0 filesz 158 memsz 158 flags 6 align 0
D (4119) esp_core_dump_elf: PHDR type 1 off 700c vaddr 3fcf5d70 paddr 3fcf5d70 filesz 380 memsz 380 flags 6 align 0
D (4129) esp_core_dump_elf: PHDR type 1 off 738c vaddr 3fcf1264 paddr 3fcf1264 filesz 158 memsz 158 flags 6 align 0
D (4140) esp_core_dump_elf: PHDR type 1 off 74e4 vaddr 3fcf0fd0 paddr 3fcf0fd0 filesz 280 memsz 280 flags 6 align 0
D (4150) esp_core_dump_elf: PHDR type 1 off 7764 vaddr 3fcc81ac paddr 3fcc81ac filesz 158 memsz 158 flags 6 align 0
D (4161) esp_core_dump_elf: PHDR type 1 off 78bc vaddr 3fcc7bf0 paddr 3fcc7bf0 filesz 5b0 memsz 5b0 flags 6 align 0
D (4171) esp_core_dump_elf: PHDR type 1 off 7e6c vaddr 3fcc5b34 paddr 3fcc5b34 filesz 158 memsz 158 flags 6 align 0
D (4182) esp_core_dump_elf: PHDR type 1 off 7fc4 vaddr 3fcc57c0 paddr 3fcc57c0 filesz 360 memsz 360 flags 6 align 0
D (4192) esp_core_dump_elf: PHDR type 1 off 8324 vaddr 3fcee67c paddr 3fcee67c filesz 158 memsz 158 flags 6 align 0
[... cut ...]
D (4482) esp_core_dump_port: A[15] 0x3fca22f2
D (4486) esp_core_dump_port: Crash Backtrace
D (4490) esp_core_dump_port:  0x420045fd
0x420045fd: frame_to_panic_info at panic_handler.c:?

D (4494) esp_core_dump_port:  0x40375b98
0x40375b98: xt_highint4 at C:/Users/DH6440/repos/v3/esp-idf/components/esp_system/port/soc/esp32s3/highint_hdl.S:119

W (08:33:14.053) coredump: Corrupted coredump found!
W (08:33:14.061) ntp: state change: [1 (ESP), 2]
I (08:33:14.063) mqtt_app: sent publish logs returned msg_id=51102
```

cbaechler, 2023-07-13:
We found that an option to fully disable logs for the coredump component was merged with https://github.com/espressif/esp-idf/commit/962621e5d76699bde807c1cd9f3eea35c4352cec. Since this is enough for our use-case, I am going to close this issue.

closed 2023-07-13 by cbaechler