espressif/esp-idf #10574: No Wake up from deep sleep after several minutes in deeep sleep

jrHernandezSio, 2023-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ENVIRONMENT:
espressif32@3.3.2 -> framework-espidf @ 3.40300.0 (4.3.0)
espressif32@5.2.0 ->framework-espidf @ 3.40402.0 (4.4.2) (also tried with this and same result)

DESCRIPTION:

If the wake up is programmed to happen each 30-60 seconds there is no problem. The ESP wakes up ok.
But if the wakeup is set to be 180 sec or longer the system never wakes up.

The bootloader shows the following:
```
[1/19 10:09:06.9]rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[1/19 10:09:06.9]flash read err, 1000
[1/19 10:09:06.9]ets_main.c 371
[1/19 10:09:07.6]ets Jun 8 2016 00:22:57
[1/19 10:09:07.6]
[1/19 10:09:07.6]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[1/19 10:09:07.6]flash read err, 1000
[1/19 10:09:07.6]ets_main.c 371
[1/19 10:09:08.2]ets Jun 8 2016 00:22:57
[1/19 10:09:08.2]
[1/19 10:09:08.2]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[1/19 10:09:08.2]flash read err, 1000
[1/19 10:09:08.2]ets_main.c 371
[1/19 10:09:08.9]ets Jun 8 2016 00:22:57
[1/19 10:09:08.9]
[1/19 10:09:08.9]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[1/19 10:09:08.9]flash read err, 1000
[1/19 10:09:08.9]ets_main.c 371
[1/19 10:09:09.5]ets Jun 8 2016 00:22:57
```

We have cheched for undetermined voltage in strapping pins, we have also checked the RTC FAST MEMORY is set to be on.

We are using FAST and SLOW RTC memory for placing some data for our program.
I am wondering about this setting in the bootloader config?
```[ ] Reserve RTC FAST memory for custom purposes```

Not sure if enable it. In other project I did not need to enable it and everything worked fine.

Now, we are out of ideas for catching the error. Any ideas?

Regards, Ramon.

AxelLin, 2023-03-19:
@jrHernandezSio 

1. Could you still reporudce the issue with latest stable releases?
   e.g. v4.3.4, v4.4.4, v5.0.1
2. Could you share you sdkconfig for reference?

jrHernandezSio, 2023-03-23:
Hi,

Eventualy, we found this in only one piece. No further investigaiton needed. Dead devices have drained the battery due to other HW design issue.

Thanks for your interest.

closed 2023-03-23 by jrHernandezSio