espressif/esp-idf #10578: Nimble : getting BLE_HS_ENOTCONN after connection established

QuentinFarizon, 2023-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I am calling peer_disc_all (which calls ble_gattc_disc_svc_by_uuid) after receiving BLE_GAP_EVENT_CONNECT.

I'm expecting that it starts discovering BLE services.

### What is the actual behavior?

Sometimes, I'm getting an error with status 7 (BLE_HS_ENOTCONN : No open connection with the specified handle)

### Steps to reproduce.

```
static int
blecent_gap_event(struct ble_gap_event *event, void *arg)
{
    case BLE_GAP_EVENT_CONNECT:
        if (event->connect.status == 0) {
            MODLOG_DFLT(INFO, "Connection established ");
            rc = peer_disc_all(event->connect.conn_handle,
                                        blecent_on_disc_complete, NULL);
            (...)
        }
        (...)
}

static void
blecent_on_disc_complete(const struct peer *peer, int status, void *arg)
{
    if (status != 0) {
        /* Service discovery failed.  Terminate the connection. */
        MODLOG_DFLT(ERROR, "Error: Service discovery failed; status=%s "
                    "conn_handle=%d\n", error_code_to_string(status), peer->conn_handle);
        (...)
    }
    (...)
}
```

### Debug Logs.

```plain
I (14:16:03.211) NimBLE: GAP procedure initiated: connect; 
I (14:16:03.217) NimBLE: peer_addr_type=1 peer_addr=
I (14:16:03.222) NimBLE: df:bd:04:a5:1a:02
I (14:16:03.227) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=250 itvl_max=600 latency=3 supervision_timeout=3200 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (14:16:03.242) NimBLE: 

I (14:16:03.809) NimBLE: Connection established 

I (14:16:03.813) NimBLE: GATT procedure initiated: discover service by uuid; uuid=
I (14:16:03.821) NimBLE: a7630001-f491-4f21-95ea-846ba586e361
I (14:16:03.827) NimBLE: 

E (14:16:09.003) NimBLE: Error: Service discovery failed; status=BLE_HS_ENOTCONN No open connection with the specified handle. conn_handle=1
```

QuentinFarizon, 2023-01-20:
I reproduced, followed by a crash this time :

```
I (08:27:32.303) NimBLE_BLE_CENT: Scan finished, RSSI -43, connecting...
I (08:27:32.311) NimBLE: GAP procedure initiated: connect; 
I (08:27:32.317) NimBLE: peer_addr_type=1 peer_addr=
I (08:27:32.322) NimBLE: df:bd:04:a5:1a:02
I (08:27:32.327) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=250 itvl_max=600 latency=3 supervision_timeout=3200 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (08:27:32.342) NimBLE: 

I (08:27:32.909) NimBLE: Connection established 
I (08:27:32.909) NimBLE: 

I (08:27:32.910) NimBLE_BLE_CENT: Added peer
I (08:27:32.912) NimBLE: GATT procedure initiated: discover service by uuid; uuid=
I (08:27:32.920) NimBLE: a7630001-f491-4f21-95ea-846ba586e361
I (08:27:32.926) NimBLE: 

E (08:27:37.421) NimBLE: Error: Service discovery failed; status=BLE_HS_ENOTCONN No open connection with the specified handle. conn_handle=1

I (08:27:37.424) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (08:27:37.464) NimBLE: GAP procedure initiated: connect; 
I (08:27:37.470) NimBLE: peer_addr_type=1 peer_addr=
I (08:27:37.476) NimBLE: df:bd:04:a5:1a:02
I (08:27:37.481) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=250 itvl_max=600 latency=3 supervision_timeout=3200 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (08:27:37.495) NimBLE: 

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40056eed  PS      : 0x00060030  A0      : 0x820152b6  A1      : 0x3fcbf760  
A2      : 0x3fc9f134  A3      : 0x00000007  A4      : 0x00000007  A5      : 0x3fc9f321  
A6      : 0x3fc9f31c  A7      : 0x00000001  A8      : 0x00000007  A9      : 0x00000001  
A10     : 0x00000001  A11     : 0x3fc9e79c  A12     : 0x00060820  A13     : 0x00060823  
A14     : 0x00000000  A15     : 0x0000abab  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000007  LBEG    : 0x420154fe  LEND    : 0x4201550c  LCOUNT  : 0x00000002  
0x420154fe: ble_hs_hci_evt_dispatch_find at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:150
 (inlined by) ble_hs_hci_evt_process at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:914

0x4201550c: ble_hs_hci_evt_process at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:917

Backtrace: 0x40056eea:0x3fcbf760 |<-CORRUPTED

ELF file SHA256: 11f10f0b8eae5952

Rebooting...
```

rahult-github, 2023-01-20:
Hi @QuentinFarizon, 

 Do you observe this behaviour if you use the bleprph example of IDF ? We tried quickly at our end, and didn't observe this issue.
 
 If this is a self developed application, can you possibly share your sample application code that we can try to use to reproduce the issue ?

Initial analysis looks like service discovery has failed which can be related to remote not sending any packet. The crash you have mentioned seems to be happening because a new connect request is sent from application when a previous disconnection attempt is still in progress.

QuentinFarizon, 2023-01-20:
Maybe the service discovery failed due to the remote, but I would expect to have a different error code in this case.

When I look at Nimble code, BLE_HS_ENOTCONN is thrown only when ble_hs_conn_find does not find the connection with given handle.
But as I discover service in BLE_GAP_EVENT_CONNECT, I'd expect connection handle to be set

rahult-github, 2023-02-08:
Hi @QuentinFarizon ,
 
  I have been trying to reproduce this issue for quite sometime, but not able to see the service discovery failure. I checked your logs in detail and the error print is in the service discovery callback function routine. At this point , it is expected to get some packets pertaining to gatt services. I think, an error OR disconnection can also cause the service discovery callback to be invoked and it happens to be with status 7. 
  
 In order to debug this further, may i request you to please help share one console log with attached patch . I have just introduced a dump to print the callstack when error happens to get more understanding for the code flow which will assist us to understand the failure scenario being observed at your end..   

[client_assert.txt](https://github.com/espressif/esp-idf/files/10686915/client_assert.txt)

Thanks,
 Rahul

Alvin1Zhang, 2023-02-15:
@QuentinFarizon Thanks for reporting, would you please help share if any updates for the issues? Thanks.

QuentinFarizon, 2023-02-22:
Hello,
Still getting this error.

See log below, I am getting BLE_HS_ENOTCONN from service discovery callback before having been called on BLE_GAP_EVENT_DISCONNECT

```
I (19253) NimBLE: Connection established; conn_handle=1
I (19253) NimBLE_BLE_CENT: BLE_GAP_EVENT_CONNECT setting conn_handle to 1
I (19253) NimBLE: 

I (19253) NimBLE_BLE_CENT: Added peer; conn_handle=1
I (19263) NimBLE: GATT procedure initiated: discover service by uuid; uuid=
I (19273) NimBLE: a7630001-f491-4f21-95ea-846ba586e361
I (19273) NimBLE: 

I (21093) NimBLE: mtu update event; conn_handle=1 cid=4 mtu=40

I (24843) NimBLE: GAP procedure initiated: 
I (24843) NimBLE: connection parameter update; conn_handle=1 itvl_min=12 itvl_max=120 latency=0 supervision_timeout=400 min_ce_len=0 max_ce_len=0
I (24843) NimBLE: 

I (28593) NimBLE: GATT procedure initiated: discover all characteristics; 
I (28593) NimBLE: start_handle=14 end_handle=22

I (51833) NimBLE: connection params update event; conn_handle=1 status=OK

I (51833) NimBLE: 

E (51843) NimBLE: Error: Service discovery failed; status=BLE_HS_ENOTCONN No open connection with the specified handle. conn_handle=1
```

rahult-github, 2023-02-23:
Hi @QuentinFarizon ,  

I assumed you are using IDF examples and had provided the patch. Can you please confirm that the peer.c file changes i provided in previous comment were applied ?  the callback on service discovery  completion looks to be called in your logs, but the backtrace information is not seen , which the previous patch , if applied, shows in my testing.

QuentinFarizon, 2023-02-24:
Here is the stacktrace with your assert :

```
I (285102) NimBLE_BLE_CENT: Scan finished, RSSI -53, connecting...
I (285112) NimBLE: GAP procedure initiated: connect; 
I (285112) NimBLE: peer_addr_type=1 peer_addr=
I (285122) NimBLE: cf:de:db:5e:7b:31
I (285122) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=250 itvl_max=600 latency=3 supervision_timeout=3200 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (285142) NimBLE: 

I (285392) NimBLE: Connection established; conn_handle=1
I (285392) NimBLE: 

I (285392) NimBLE_BLE_CENT: Added peer; conn_handle=1
I (285392) NimBLE: GATT procedure initiated: discover service by uuid; uuid=
I (285402) NimBLE: a7630001-f491-4f21-95ea-846ba586e361
I (285412) NimBLE: 

E (290132) NimBLE: disconnection error during service discovery ; rc=7

assert failed: peer_disc_complete peer.c:83 (0)

Backtrace: 0x40376036:0x3fcbf180 0x4038049d:0x3fcbf1a0 0x403867f5:0x3fcbf1c0 0x4200b0aa:0x3fcbf2e0 0x4200b2d1:0x3fcbf300 0x42015889:0x3fcbf320 0x420158bf:0x3fcbf340 0x42015fe9:0x3fcbf360 0x42016e11:0x3fcbf390 0x42019553:0x3fcbf3b0 0x4201534b:0x3fcbf410 0x42015561:0x3fcbf430 0x42014b51:0x3fcbf450 0x4037ea05:0x3fcbf470 0x40377dfd:0x3fcbf490 0x42009ade:0x3fcbf4b0
0x40376036: panic_abort at /home/quentin/esp/esp-idf/components/esp_system/panic.c:412

0x4038049d: esp_system_abort at /home/quentin/esp/esp-idf/components/esp_system/esp_system.c:135

0x403867f5: __assert_func at /home/quentin/esp/esp-idf/components/newlib/assert.c:78

0x4200b0aa: peer_disc_complete at /home/quentin/Code/Boks/boks-link/main/ble/peer.c:83
 (inlined by) peer_disc_complete at /home/quentin/Code/Boks/boks-link/main/ble/peer.c:77

0x4200b2d1: peer_svc_disced at /home/quentin/Code/Boks/boks-link/main/ble/peer.c:635

0x42015889: ble_gattc_disc_svc_uuid_cb at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:1585

0x420158bf: ble_gattc_disc_svc_uuid_err at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:1657

0x42015fe9: ble_gattc_fail_procs at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:1091

0x42016e11: ble_gattc_connection_broken at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4773

0x42019553: ble_gap_master_connect_reattempt at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1002

0x4201534b: ble_hs_hci_evt_disconn_complete at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:233
 (inlined by) ble_hs_hci_evt_disconn_complete at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:187

0x42015561: ble_hs_hci_evt_process at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:919

0x42014b51: ble_hs_event_rx_hci_ev at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:542

0x4037ea05: npl_freertos_event_run at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:450

0x40377dfd: ble_npl_event_run at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:217

0x42009ade: blecent_host_task at /home/quentin/Code/Boks/boks-link/main/ble/main_ble.c:572

ELF file SHA256: 61c6ab3d09679840

Rebooting...

```

QuentinFarizon, 2023-02-24:
I see that CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT is enabled, I was not aware of that. Could it be the cause of the issue ?

What I want is ti maintain a permanent connection, always scanning and always connecting when down, so I already manages to reconnect if disconnected. So I guess I should disable CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT and keep managing it myself ?

rahult-github, 2023-02-24:
Hi @QuentinFarizon ,

 The CONFIG_BT_NIMBLE_ENABLE_CONN_REATTEMPT macro would attempt connection thrice if the connection did not happen in the first attempt. This basically gives more chance to get a connection through, in case the first attempt failed with reason 3E.

You can disable this macro as per your use case.  

However, this still doesn't give input as to why disconnection is happening in the first place. Do you have any provision of taking a Over the Air log ?   Is this issue seen with all devices or seen with a particular remote device ? 

Thanks,
 Rahul

Alvin1Zhang, 2023-03-27:
@QuentinFarizon Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

rahult-github, 2023-08-07:
Closing issue due to no further response. Please feel free to reopen with logs , if needed.

closed 2023-08-07 by rahult-github