espressif/esp-idf #10579: CMAKE (Native cmake) is not supported.

phelter, 2023-01-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I would like to use native cmake with the esp_idf platform.

Right now the ESP-IDF environment states it supports cmake; however, I do not see it supporting cmake it supports some variation of cmake which is ESP-IDF specific and should be explicitly stated in the documentation that it DOES NOT SUPPORT NATIVE CMAKE.

Native CMake is not possible due to the python and other scripts that could be implemented in CMake but are implemented externally.

The fact that the esp-idf process enforces the use of `idf_component_register()` and a predefined directory structure for the scripts built on top of CMake means the following do not work.  An example of a native CMake `target` (similar to an `idf_component`) 

```cmake
add_library(adapter_foo STATIC)
add_library(adapter::foo ALIAS adapter_foo)

target_sources(adapter_foo
    PRIVATE
        api/adapter_foo.h
        src/adapter_foo.c
)

target_include_directories(adapter_foo
    PUBLIC
        api
)

target_compile_options(adapter_foo
    PRIVATE
        $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-declaration-after-statement>
        $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-padded>
        $<$<COMPILE_LANG_AND_ID:C,Clang>:-Wno-unused-variable>
)

target_link_libraries(adapter_foo
    PUBLIC 
        idf::esp_common
    PRIVATE
        idf::freertos
        idf::esp_hw_support
        idf::log
        idf::nvs_flash
)

# -------------------------------------------------------------------
add_subdirectory(mock)
add_subdirectory(test)
```

The following do not work because of the way cmake is being executed on the idf.
`idf.py build -DCMAKE_BUILD_TYPE:STRING=Debug`

This executes cmake with the `-P` option which means everything must be a script and this then means that the following cause the cmake initial pass to fail:

* `add_subdirectory()`
* `add_library()`
* `target_compile_definitions()`
And a bunch of other basic cmake functions do not work.

I'd also like to enforce the `target/component` hierarchy and use a different directory structure from what is expected by the idf.py scripts.

I'd also like to allow for multiple executables being generated (i.e. multiple *.elf files) all of which is not feasible due to the way the cmake is wrapped by the idf specific build environment.

I also use `GoogleTest` for testing and not the test environment that you have because I want to perform all of the testing natively and not on the device itself.  This is the reason for the `mock` and  `test` sub-directories.  The mock sub-directory is the implementation of the mock of the `target/component` and the test is the [CMAKE_HOST_SYSTEM_NAME](https://cmake.org/cmake/help/latest/variable/CMAKE_HOST_SYSTEM_NAME.html#variable:CMAKE_HOST_SYSTEM_NAME) test suite for that component.

I have tried using `idf_as_lib` - but this means I need to recreate all of the:
* menuconfig
* flash
* etc....
commands that could be native cmake using `add_custom_target()`.

### What is the actual behavior?

Fails - with the error - 
`This is not scriptable.`

### Steps to reproduce.

None

### Build or installation Logs.

```plain
None.
```

### More Information.

None

igrr, 2023-01-21:
While changing the way the way IDF build system works to fit the way you describe might not be viable without breaking changes, I do agree that the behavior you observe is inconsistent with the documentation.

Currently we have a documentation section which says that pure CMake libraries (or "native cmake", as you say) can be used as components without additional wrappers: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#writing-pure-cmake-components. This is obviously not the case, as you have found. I'll check why that is the case and will likely update the documentation to match the actual behavior, explicitly stating that standard CMake libraries need to be wrapped into a component.

higaski, 2023-01-22:
There are already multiple issues dressing this:
https://github.com/espressif/esp-idf/issues/9929
https://github.com/espressif/esp-idf/issues/10589

I would put up with the fact that this will probably not change anymore.

Funnily enough, I just wrote a blog entry about how I use GoogleTest with ESP32s just last week:
https://higaski.at/working-with-or-rather-around-the-esp32-build-system/
I think it addresses at least some of your problems and offers more or less reasonable workarounds.

phelter, 2023-01-22:
Yes - my workaround unfortunately is to have 2 completely different and parallel build systems - both using CMake.

This is a bit of trickery but using additional directories in the structure of the code I can skip over particular ones and build for both.

eg:
```
main/
   drivers/
        driverA/
              api/
              src/
              mock/
                  CMakeLists.txt - Native CMake
              test/
                  CMakeLists.txt - Native CMake
             CMakeLists.txt - Native CMake
         CMakeLists.txt - Native CMake - include all subdirs.
   tasks/
       taskZ/
           - Native CMake similar to driverA.
       CMakeLists.txt - Native CMake - include all subdirs.
    apps/
       app1/
          - Native CMake similar to driverA but no mock.
       CMakeLists.txt - Native CMake - include all subdirs.

main/CMakeLists.txt - ESP-IDF Selection of all src code into one monolithic app idf_component.
CMakeLists.txt - project cmake selects based on an option whether to include the main dir or the subdirs directly.
```

All of your assumptions on requiring a two stage approach are incorrect.  The KConfig could be made as a single target - eg `idf::sdkconfig` which performs all of the actions required to build the sdkconfig.h file.  All components that rely on an sdkconfig (most but not all) then depend on that `idf::sdkconfig` target and the build process handles things.

Also the `feature` of not having to request all the required components for building and writing your own parallel method for handling this is missing out on one of CMake's best features - dependency handling.  If all components had a clear dependency tree in the esp-idf and then the components consumed are also in the `target_link_libraries` of the components you have documentation on your dependency tree already defined.  A simple way to provide the `don't worry about dependencies` would be to create a monolithic target which includes all esp-idf targets and then only one thing would be needed to be linked to your apps.

Supporting the native cmake in a single step would also allow multiple apps to be built at the same time (assuming the sdkconfig definitions were the same) in the same repo.

All `features` that are being requested in the issues you've mentioned and are already supported in CMake but not in the ESP-IDF non-compliant cmake.

igrr, 2023-01-24:
Switching to a single stage build (without the component requirements expansion step) would indeed be a nice improvement, solving several existing issues along the way and making developers familiar with CMake feel more at home with IDF.

In the past few days I've looked through the list of reasons we have initially decided to have a two stage build instead of the simpler single stage one. I have also went through the differences between the existing approach and the potential single stage approach and noted the following issues I don't yet have a good solution for. I want to point out that I list these not for the reason of finding excuses to keep the existing two stage approach. It is mostly for the sake of the discussion. These issues are mostly related to ensuring compatibility with components and projects written for the currently supported versions of IDF.

1. In the current build system, the name of the Kconfig file inside the component can be customized in the `idf_component_register` call. If Kconfig processing happens without expansion of component CMakeLists.txt files in scripting mode, we can only support the default names (`Kconfig` or `Kconfig.projbuild`). It is probably going to be a breaking change to drop the customizable Kconfig file name support, not a major one though.
2. In the current build system, if a component is not included in the build (i.e. no other component lists it in REQUIRES or PRIV_REQUIRES), then its Kconfig file doesn't participate in sdkconfig file generation. This means that if a component is totally unused by the application, its options are not visible in menuconfig or in sdkconfig file. With the single stage build without CMakeLists.txt expansion in script mode, component dependencies (expressed by target_link_libraries calls) would only be known after sdkconfig generation is done. Therefore, we don't know which components will end up being unused, and all the components' options have to be present in sdkconfig. This has two implications:
   1. It might appear to the user that some menuconfig option exists and is editable, but in fact it won't have any impact on the program if the corresponding component is not linked. This can be particularly confusing for options which enable some "background" features that are not directly used in the code. One example is Core Dump. Currently you need to enable a Kconfig option to enable the Core Dump. In the single stage build, if espcoredump component ends up not being linked, it is still possible to enable the Core Dump in menuconfig. But since the component is not linked, the option will have no effect. Of course for the components within IDF project we can fix this, by adding a `target_link_libraries(... PRIVATE idf::espcoredump`) from another component if the corresponding Kconfig option is enabled. However for components not maintained within IDF, this would be a breaking change in the behavior.
   2. Currently in a few places we detect whether a component is included into the build by checking some Kconfig option of that component. If all the component's options are present in sdkconfig even if the component ends up not being linked, this logic stops working. Again, this is something we can fix for the components built into the IDF, but is a breaking change for components maintained elsewhere.
3. The approach for "trimming" the build (that is, removing unused components) becomes different with the single stage build. In the current build system, the project developer can add `set(COMPONENTS main)` to the project, which will result in `main` plus all of its dependencies and transitive dependencies to be included into the build. All other components not mentioned in REQUIRES or PRIV_REQUIRES of `main` and its transitive dependencies won't be included. This means that:
   - their Kconfig options will not appear in sdkconfig or menuconfig
   - their CMakeLists.txt files won't be added via add_subdirectory
   The nice part about this is that the project developer doesn't need to know which components _exactly_ will be included or excluded from the build; all they need to know is the root of the dependency tree. This helps IDF developers a lot, because we can add a new component to IDF, list it as PRIV_REQUIRES of another component, and it will get automatically included without the project developer having to update anything.
   With the single stage build, if we can't expect the project developer to list all the components to be included, then it looks like we must call `add_subdirectory` for all components in ESP-IDF.
   I don't see a nice way yet to provide compatibility for projects which used to do `set(COMPONENTS main)`. Maybe we can keep the requirements expansion machinery for compatibility, and invoke it only if the project CMakeLists.txt file happens to set the `COMPONENTS` variable like this.

phelter, 2023-01-24:
Thanks for the info and rationale for the 2 stage approach @igrr .

For some of your concerns:

```
1. In the current build system, the name of the Kconfig file inside the component can be customized in the idf_component_register call. If Kconfig processing happens without expansion of component CMakeLists.txt files in scripting mode, we can only support the default names (Kconfig or Kconfig.projbuild). It is probably going to be a breaking change to drop the customizable Kconfig file name support, not a major one though.
```

This could be a per `target` user defined Property which selects an additional file to search for inside the component directory and then use [find_file](https://cmake.org/cmake/help/latest/command/find_file.html) to select which file is used.
```cmake
get_target_property( _userDefinedKConfigFile ${COMPONENT_NAME} IDF_COMPONENT_KCONFIG_FILE)
find_file( _kconfigFile
   NAMES
      KConfig
      KConfig.projbuild
     ${_kconfigFile}
)
```

For the next two issues the duplication of the linkage and dependency tree in a two stage compile is problematic.

```
2. In the current build system, if a component is not included in the build (i.e. no other component lists it in REQUIRES or PRIV_REQUIRES), then its Kconfig file doesn't participate in sdkconfig file generation. This means that if a component is totally unused by the application, its options are not visible in menuconfig or in sdkconfig file. With the single stage build without CMakeLists.txt expansion in script mode, component dependencies (expressed by target_link_libraries calls) would only be known after sdkconfig generation is done. Therefore, we don't know which components will end up being unused, and all the components' options have to be present in sdkconfig. This has two implications:
```

My suggestion would be to look at how [Zephyr](https://docs.zephyrproject.org/3.2.0/build/kconfig/index.html) does the similar function.

Since this is a per-application setting - typically if you have multiple apps with different configurations you would have to rebuild.
The act of building cmake means the dependencies are already sorted out - so the setup for a menuconfig with limited set could be done with a cmake macro/function that executes based on the input application that is requested to be built, [recursively iterate through it's list of dependencies](https://stackoverflow.com/questions/32756195/recursive-list-of-link-libraries-in-cmake) (via target properties and only CMAKE properties - not IDF ones).  Determine the full list and then display that for that particular application.

I would also suggest making this it's own target as well - eg `idf::sdkconfig` so that if you want a single sdkconfig for multiple applications that all share most of the definition you can - and can compile them that way.  This target would then perform all of the kconfig selection and then build the sdkconfig - and then all esp_idf dependencies that consume idf::sdkconfig get recompiled and the ones that don't have any configuration are left as-is.

For multiple applications with different sdkconfig settings in the same project,  You can then enforce a per `cmake -G Ninja ....` process per application and the appropriate sdkconfig would then be setup via the menuconfig.

```
3. The approach for "trimming" the build (that is, removing unused components) becomes different with the single stage build. In the current build system, the project developer can add set(COMPONENTS main) to the project, which will result in main plus all of its dependencies and transitive dependencies to be included into the build. All other components not mentioned in REQUIRES or PRIV_REQUIRES of main and its transitive dependencies won't be included. This means that:

    their Kconfig options will not appear in sdkconfig or menuconfig
    their CMakeLists.txt files won't be added via add_subdirectory
```

This is done automatically with CMAKE.  The only thing you need to do is add: `EXCLUDE_FROM_ALL` to the target definition.  The difference is all of the CMakeLists.txt are processed and you can still compile the unused ones if you like but when you compile the application or the all and they are not used, they don't get built unless explicitly requested on the command line.
eg:
`add_library(log STATIC EXCLUDE_FROM_ALL)`

One more comment regarding:
```
The nice part about this is that the project developer doesn't need to know which components exactly will be included or excluded from the build; all they need to know is the root of the dependency tree. This helps IDF developers a lot, because we can add a new component to IDF, list it as PRIV_REQUIRES of another component, and it will get automatically included without the project developer having to update anything.
```

I consider this a deficiency in the current definition of your esp_idf components.  It is much easier to know which dependencies are needed when they are explicitly stated in all of the cmake `target_link_libraries`.  And identifies issues of where circular dependencies and other issues may arise - which actually lead to better API and dependency tree definition once explicitly indicated.  You can also review the dependency tree with cmake and graphviz to see where these issues arise.

If you want to support this `feature` still of not needing to know which components exactly will be included - then make a full blown `idf_all` target that includes each and every one of the libraries and then consume that in the user's main, and make sure to set this as `EXCLUDE_FROM_ALL`.  This however goes against the `only build and use what you need` option.

There are ways around the circular dependencies by supporting just an API specific target (INTERFACE ONLY) and a static library target that includes the interface.  - Once those specific ones are identified you can then deal with them later by reorganizing how the are linked.

igrr, 2023-01-25:
> This could be a per target user defined Property which selects an additional file to search for inside the component directory and then use [find_file](https://cmake.org/cmake/help/latest/command/find_file.html) to select which file is used.

Where would this property be set, though? If it is set in the component CMakeLists.txt file, then the property will be set too late. All Kconfig values should be known before a component subdirectory is added by add_subdirectory(), otherwise we won't be able to use them, for example, in conditional logic. This means that `sdkconfig` and `sdkconfig.cmake` files should be generated at CMake configuration time, before component subdirectories are processed. Therefore Kconfig files should be collected before the `sdkconfig` is generated, and before the component CMakeLists.txt file has a chance to set the target property `IDF_COMPONENT_KCONFIG_FILE`.

From your reply, it seems that you propose to do Kconfig generation at build time, not CMake configuration time, since you mention this `idf::sdkconfig` build target. I think this approach will mean that Kconfig option values can't be used in component CMakeLists.txt files, which make Kconfig options much less useful. (And be a major breaking change, besides that.)

Regarding your answer to my 2nd point, I'm sorry, I can't understand how the answer relates to the problem. In this problem, I am not concerned with building more than one app in the same build, for now. Let me try to explain the issue again by using an example:

Consider the case when there are 3 components in IDF: esp_a, esp_b, esp_c, each with a Kconfig file. `esp_a` has a target_link_libraries dependency on `esp_b`. Then there is the component defined in the user application, `main`, which has a dependency on `esp_a` only.

In the existing build system, in the first stage we will determine that esp_a, esp_b, main components should be included into the build. Kconfig generation step will use Kconfig files only of these components. There will be no options in `sdkconfig` from the Kconfig file of esp_c component.

In the single stage build, dependency tree information is only known after we have added all component subdirectories and each component CMakeLists.txt file has been evaluated. Since Kconfig processing has to happen earlier than that, target dependency information can't be used when creating `sdkconfig`, so we can only create `sdkconfig` with Kconfig options of all the components. In the example above, it would mean that esp_c component's Kconfig options will be visible in `sdkconfig` and as CMake variables, even though this component is actually not used in the application.

> The difference is all of the CMakeLists.txt are processed and you can still compile the unused ones if you like but when you compile the application or the all and they are not used, they don't get built unless explicitly requested on the command line.

This is exactly the difference I was trying to point out — CMakeLists.txt of components not used in the application (like esp_c) are going to be processed in the single stage build. And as mentioned above, Kconfig options of unused components will be processed and included into `sdkconfig` as well.

> I consider this a deficiency in the current definition of your esp_idf components. It is much easier to know which dependencies are needed when they are explicitly stated in all of the cmake target_link_libraries.

In fact, the existing build system's REQUIRES and PRIV_REQUIRES dependency specifications are eventually passed on as target_link_libraries calls, so this part is not so much different in the single stage and dual stage build. The only difference is that currently we are implicitly adding several "common" components as dependencies to every component in the build, but this is equivalent of users having to add some `idf::common` to `target_link_libraries` call. `idf:common` would in turn depend on those other "common" components. I am totally not against expressing the dependencies between components and using that information to reduce cyclic dependencies, improve the APIs and so on.

However, I don't think the developer of an application based on ESP-IDF is expected to know the list and the purpose of _all_ the ESP-IDF components. Let me illustrate that with another example:

Suppose IDF contains 3 libraries, http_client, http_server and http_parser. Both http_client and http_server have a PRIVATE dependency on http_parser. The user is writing an application that relies either on http_client or http_server. They will add http_client or http_server to target_link_libraries call in their `main` component. However, in order for the dependency on http_parser to be satisfied, that component has to be added to the build (via `add_subdirectory`).

Who should be responsible for calling `add_subdirectory(http_parser)`? In the single stage build approach, there are the following alternatives:
1. User application, i.e. the top-level project CMakeLists.txt file, calls add_subdirectory for all the components which should be included into the build. I would like to avoid this, because the dependency on the http_parser component is an internal implementation detail of http_client and http_server. It is not something that the user of the higher level http_client/http_server API is supposed to know or care about.
2. IDF itself calls add_subdirectory for all the components which are present in the components/ directory. This is also not ideal, since this adds a bunch of components (and their Kconfig options, as pointed out above) which aren't necessarily used by the application.
3. `http_client` and `http_server` CMakeLists.txt files will both try to call `add_subdirectory(../http_parser)`. I think this produces an error in the recent CMake versions, and requires "include guards" to be implemented around the add_subdirectory calls.

phelter, 2023-01-25:
I suggest you purchase a good book on cmake.  The one I regularly reference is:
[Professional CMake](https://crascit.com/professional-cmake/)

I have worked around the issues I have at this moment with the special idf build flow by providing duplicate build flows - one for native cmake and one for the idf version.

If you would like more consultation services, feel free to reach out to me and we can arrange a time and a contract.

I will answer the above questions and leave you with that.

> Q: Where would this property be set, though? 
**A:**  In the `CMakeLists.txt` - with a macro of your choosing or defined by the user.  No it is not too late. The KConfig can be a target that all other idf targets rely on and that target can be derived from an [add_custom_target](https://cmake.org/cmake/help/latest/command/add_custom_target.html?highlight=add_custom#command:add_custom_target) that first recursively iterates through the list of targets to build the app, identify all of the KConfig or other files, and then generates an output of an sdkconfig and sdkconfig.h file.  This can all be done after all targets have been identified in the CMake configuration time.

> From your reply, it seems that you propose to do Kconfig generation at build time, not CMake configuration time, since you mention this idf::sdkconfig build target. I think this approach will mean [...]

It sounds as though you are using Kconfig for two purposes.  The first for CMake Configuration - Cached parameters for various actions at configuration time, AND for configuring `#defines` for the components themselves during build operation.  If you want you can have a pre-configure stage in the CMake build stage which generates all of the CMake Cached variables for a given build from the Kconfig settings of all modules available, and then the use of them is then decided at build time.  This would allow you to include the defines on a per component level using `target_compile_definitions()`.  The menuconfig target could then decide which ones are to be displayed based on the recursive scan of used components/targets.

>This is exactly the difference I was trying to point out — CMakeLists.txt of components not used in the application (like esp_c) are going to be processed in the single stage build. And as mentioned above, Kconfig options of unused components will be processed and included into sdkconfig as well.

The CMakeLists.txt of all components used and not used in the application will be processed in the Cmake Configuration stage.  However the creation of the Kconfig options can be derived and defined by only the ones that are identified through recursively processing the `exe target`'s dependencies and generating the sdkconfig off of that - in the build stage as a target.  The CMake Configuration stage only identifies if there is a Kconfig file and does no processing of them.

> However, I don't think the developer of an application based on ESP-IDF is expected to know the list and the purpose of all the ESP-IDF components. Let me illustrate that with another example:

If the developer of an application knows which header to include to get the necessary API, then they know which component to include - or would be able to if it weren't obfuscated the way it currently is with the use of a `COMPONENT_NAME` - luckily it's easy to derive from the name of the directory as `idf::<dirname>`.  The developer should know where it's coming from and all underlying dependencies should already be defined by the ESP-IDF engineers maintaining the library dependencies.

> Who should be responsible for calling add_subdirectory(http_parser)? In the single stage build approach, there are the following alternatives:

As mentioned - the ESP-IDF engineer maintianing the esp-idf will always `add_subdirectory(http_parser)`.  and have the `add_target(http_parser STATIC EXCLUDE_FROM_ALL)` - so that it will only get built iff the http_parser is used by the http_client or http_server AND if http_server or http_client is then used by the application developer.

All of your other concerns:
 > 1.    User application, i.e. the top-level project CMakeLists.txt file, calls add_subdirectory for all the components which should be included into the build. I would like to avoid this, because the dependency on the http_parser component is an internal implementation detail of http_client and http_server. It is not something that the user of the higher level http_client/http_server API is supposed to know or care about.

You can add all components via add_subdirectory() and depending on the dependencies of what is built - only the ones used will be built.  Yes http_parser is an internal implementation used by those and that can be defined as `target_link_libraries(http_client PRIVATE http_parser...)` for example it.  This means the http_parser API and library won't be exposed beyond the scope of the http_client target.

>2  IDF itself calls add_subdirectory for all the components which are present in the components/ directory. This is also not ideal, since this adds a bunch of components (and their Kconfig options, as pointed out above) which aren't necessarily used by the application.

I don't know why you assume this is not ideal.  These components are available and you've identified how to build them if they are to be used by an application developer.

[... cut ...] s "include guards" to be implemented around the add_subdirectory calls.

This is incorrect CMake usage.  the component subdirectory should have a CMakeLists.txt which just includes all of the subdirectories.  the http_client/CMakeLists.txt should not add_subdirectory() - but just link to the http_parser that has been added at the level above.  Order of add_subdirectory at the CMake Configuration stage does not matter.

Again if you would like to retain my services please feel free to contact me.

igrr, 2023-01-25:
Thank you for your thoughtful comments in this issue. 

I generally agree with the points you made, as far as discussing "how a CMake based project should work". The only questions I have left are related to keeping compatibility with existing projects and component, while switching to a single stage build. They are questions in the sense that, some breaking changes will need to happen, and the question is how to choose the set of breaking changes which will cause least amount of trouble for component and project developers.

Since you have solved the problem you originally faced, I am going to close the issue. We do already have an internal issue about the single stage build, and I'll add more information there. Hopefully in the next IDF major version we will be able to support standard CMake projects better.

closed 2023-01-25 by igrr

phelter, 2023-01-25:
@igrr  - one way i've done and seen other projects deal with compatibility is:

* Additive: Support both modes of operation for a period of time, identifying no new features for the old one,  and phase out the old one on a given major release.
* Transitive: Provide scripts for migration of existing idf cmake files to newer methodology.

dobairoland, 2025-11-06:
We invite everyone interested in ESP-IDF build systems and CMake integration to check out the [Build System v2 Technical Preview announcement](https://github.com/espressif/esp-idf/issues/17833).