espressif/esp-idf #10582: Mock esp_wifi

ljden, 2023-01-20:
### Is your feature request related to a problem?

I would like to mock esp_wifi for host testing my AP manager

### Additional context.

I gave ti a bit of a go but got stuck. So far, I have:
```bash
vbox@vbox:~/esp/esp-idf/tools/mocks/esp_wifi$ cat CMakeLists.txt
# NOTE: This kind of mocking currently works on Linux targets only.
#       On Espressif chips, too many dependencies are missing at the moment.
message(STATUS "building ESP WIFI MOCKS")

idf_component_get_property(original_esp_wifi_dir esp_wifi COMPONENT_OVERRIDEN_DIR)

idf_component_mock(INCLUDE_DIRS "${original_esp_wifi_dir}/include"
                   REQUIRES esp_hw_support esp_event esp_netif
                   MOCK_HEADER_FILES ${original_esp_wifi_dir}/include/esp_wifi.h
                                     # ${original_esp_wifi_dir}/include/XXX.h
    )

vbox@vbox:~/esp/esp-idf/tools/mocks/esp_wifi$ cat mock/mock_config.yaml 
        :cmock:
          :plugins:
            - expect
            - expect_any_args
            - return_thru_ptr
            - array
            - ignore
            - ignore_arg
            - callback
```
and added to my `host_test/CMakeLists.txt`
```make
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_wifi/")
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_hw_support/")
```
But I am failing on finding `machine/endian.h`
```make
vbox@vbox:~/wifi_manager/host_test$ idf.py reconfigure build
Executing action: reconfigure
Running cmake in directory /home/vbox/wifi_manager/host_test/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=linux -DCCACHE_ENABLE=0 /home/vbox/wifi_manager/host_test"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test ver_gnu17_supported
-- Performing Test ver_gnu17_supported - Success
-- Performing Test ver_gnu++20_supported
-- Performing Test ver_gnu++20_supported - Success
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Building ESP-IDF components for target linux
Processing 3 dependencies:
[1/3] component_base (bf6205ccb2d90871d9318ef88289ed28c1b42896)
[2/3] idf (5.0.0)
[3/3] wifi_manager (0.0.1)
-- building ESP HW SUPPORT MOCKS
-- building ESP WIFI MOCKS
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- Project sdkconfig file /home/vbox/wifi_manager/host_test/sdkconfig
Loading defaults file /home/vbox/wifi_manager/host_test/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building ESP HW SUPPORT MOCKS
-- building ESP WIFI MOCKS
-- Component idf::main will be linked with -Wl,--whole-archive
-- Components: cmock component_base esp_common esp_event esp_hw_support esp_netif esp_rom esp_wifi freertos linux log main unity wifi_manager
-- Component paths: /home/vbox/esp/esp-idf/components/cmock /home/vbox/wifi_manager/host_test/managed_components/component_base /home/vbox/esp/esp-idf/components/esp_common /home/vbox/esp/esp-idf/components/esp_event /home/vbox/esp/esp-idf/tools/mocks/esp_hw_support /home/vbox/esp/esp-idf/components/esp_netif /home/vbox/esp/esp-idf/components/esp_rom /home/vbox/esp/esp-idf/tools/mocks/esp_wifi /home/vbox/esp/esp-idf/tools/mocks/freertos /home/vbox/esp/esp-idf/components/linux /home/vbox/esp/esp-idf/components/log /home/vbox/wifi_manager/host_test/main /home/vbox/esp/esp-idf/components/unity /home/vbox/wifi_manager
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vbox/wifi_manager/host_test/build
Executing action: all (aliases: build)
Running ninja in directory /home/vbox/wifi_manager/host_test/build
Executing "ninja all"...
[1/39] Try to find ruby. If this fails, you need to install rubyruby 3.0.2p107 (2021-07-07 revision 0db68f0233) [x86_64-linux-gnu]
[24/38] Try to find ruby. If this fails, you need to install rubyruby 3.0.2p107 (2021-07-07 revision 0db68f0233) [x86_64-linux-gnu]
[30/37] Try to find ruby. If this fails, you need to install rubyruby 3.0.2p107 (2021-07-07 revision 0db68f0233) [x86_64-linux-gnu]
[31/36] Building C object esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/mocks/Mockesp_wifi.c.oFAILED: esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/mocks/Mockesp_wifi.c.o 
/usr/bin/cc -DUNITY_INCLUDE_CONFIG_H -I/home/vbox/wifi_manager/host_test/build/config -I/home/vbox/esp/esp-idf/components/esp_wifi/include -I/home/vbox/wifi_manager/host_test/build/esp-idf/esp_wifi/mocks -I/home/vbox/esp/esp-idf/tools/mocks/freertos/include -I/home/vbox/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/vbox/esp/esp-idf/components/freertos/esp_additions/include -I/home/vbox/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/vbox/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos -I/home/vbox/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/linux/include -I/home/vbox/wifi_manager/host_test/build/esp-idf/freertos/mocks -I/home/vbox/esp/esp-idf/components/log/include -I/home/vbox/esp/esp-idf/components/esp_rom/include -I/home/vbox/esp/esp-idf/components/esp_rom/include/linux -I/home/vbox/esp/esp-idf/tools/mocks/soc/include -I/home/vbox/esp/esp-idf/components/esp_common/include -I/home/vbox/esp/esp-idf/components/cmock/CMock/src -I/home/vbox/esp/esp-idf/components/unity/include -I/home/vbox/esp/esp-idf/components/unity/unity/src -I/home/vbox/esp/esp-idf/components/esp_hw_support/include -I/home/vbox/wifi_manager/host_test/build/esp-idf/esp_hw_support/mocks -I/home/vbox/esp/esp-idf/components/esp_event/include -I/home/vbox/esp/esp-idf/components/linux/include -I/home/vbox/esp/esp-idf/components/esp_netif/include -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/home/vbox/wifi_manager/host_test=. -fmacro-prefix-map=/home/vbox/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-165-ge5675848e3\" -DconfigTICK_RATE_HZ=1000 -DESP_PLATFORM -MD -MT esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/mocks/Mockesp_wifi.c.o -MF esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/mocks/Mockesp_wifi.c.o.d -o esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/mocks/Mockesp_wifi.c.o -c /home/vbox/wifi_manager/host_test/build/esp-idf/esp_wifi/mocks/Mockesp_wifi.c
In file included from /home/vbox/esp/esp-idf/components/esp_netif/include/esp_netif.h:12,
                 from /home/vbox/esp/esp-idf/components/esp_wifi/include/esp_wifi_default.h:10,
                 from /home/vbox/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:58,
                 from /home/vbox/wifi_manager/host_test/build/esp-idf/esp_wifi/mocks/Mockesp_wifi.h:6,
                 from /home/vbox/wifi_manager/host_test/build/esp-idf/esp_wifi/mocks/Mockesp_wifi.c:6:
/home/vbox/esp/esp-idf/components/esp_netif/include/esp_netif_ip_addr.h:11:10: fatal error: machine/endian.h: No such file or directory
   11 | #include <machine/endian.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
Not sure if I'm on the right track and how to fix the current error. I suspect I need to do something similar to the mqtt mock but I don't fully understand the host test infrastructure

ljden, 2023-01-24:
A compiling mock - otherwise untested. Note - need to set `-Warray-parameter=0` to suppress issue (presumably) from CMock - see https://github.com/ThrowTheSwitch/CMock/issues/422
```bash
$ cat host_test/CMakelists.txt
cmake_minimum_required(VERSION 3.16)
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/freertos/")
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_wifi/")
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_hw_support/")

idf_build_set_property(COMPILE_DEFINITIONS "-DCONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10" APPEND)
idf_build_set_property(COMPILE_DEFINITIONS "-DCONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32" APPEND)
idf_build_set_property(COMPILE_DEFINITIONS "-DCONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1" APPEND)

project(wifi_ap_host_test)

idf_build_set_property(COMPILE_OPTIONS "-Warray-parameter=0" APPEND)
```
and the `esp-idf` diff
```diff
diff --git a/tools/mocks/esp_netif/CMakeLists.txt b/tools/mocks/esp_netif/CMakeLists.txt
new file mode 100644
index 0000000000..c86e61136e
--- /dev/null
+++ b/tools/mocks/esp_netif/CMakeLists.txt
@@ -0,0 +1,10 @@
+# NOTE: This kind of mocking currently works on Linux targets only.
+#       On Espressif chips, too many dependencies are missing at the moment.
+message(STATUS "building ESP NETIF MOCKS")
+
+idf_component_get_property(original_esp_netif_dir esp_netif COMPONENT_OVERRIDEN_DIR)
+
+idf_component_mock(INCLUDE_DIRS "${original_esp_netif_dir}/include"
+                   REQUIRES esp_event esp_hw_support
+                   MOCK_HEADER_FILES ${original_esp_netif_dir}/include/esp_netif.h
+    )
diff --git a/tools/mocks/esp_netif/mock/mock_config.yaml b/tools/mocks/esp_netif/mock/mock_config.yaml
new file mode 100644
index 0000000000..596255b0ce
--- /dev/null
+++ b/tools/mocks/esp_netif/mock/mock_config.yaml
@@ -0,0 +1,9 @@
+        :cmock:
+          :plugins:
+            - expect
+            - expect_any_args
+            - return_thru_ptr
+            - array
+            - ignore
+            - ignore_arg
+            - callback
diff --git a/tools/mocks/esp_wifi/CMakeLists.txt b/tools/mocks/esp_wifi/CMakeLists.txt
new file mode 100644
index 0000000000..250664d3a2
--- /dev/null
+++ b/tools/mocks/esp_wifi/CMakeLists.txt
@@ -0,0 +1,10 @@
+# NOTE: This kind of mocking currently works on Linux targets only.
+#       On Espressif chips, too many dependencies are missing at the moment.
+message(STATUS "building ESP WIFI MOCKS")
+
+idf_component_get_property(original_esp_wifi_dir esp_wifi COMPONENT_OVERRIDEN_DIR)
+
+idf_component_mock(INCLUDE_DIRS "${original_esp_wifi_dir}/include" "./include"
+                   REQUIRES esp_event esp_hw_support esp_netif
+                   MOCK_HEADER_FILES ${original_esp_wifi_dir}/include/esp_wifi.h
+    )
diff --git a/tools/mocks/esp_wifi/include/machine/endian.h b/tools/mocks/esp_wifi/include/machine/endian.h
new file mode 100644
index 0000000000..228316d612
--- /dev/null
+++ b/tools/mocks/esp_wifi/include/machine/endian.h
@@ -0,0 +1,2 @@
+#pragma once
+#include_next <endian.h>
diff --git a/tools/mocks/esp_wifi/mock/mock_config.yaml b/tools/mocks/esp_wifi/mock/mock_config.yaml
new file mode 100644
index 0000000000..596255b0ce
--- /dev/null
+++ b/tools/mocks/esp_wifi/mock/mock_config.yaml
@@ -0,0 +1,9 @@
+        :cmock:
+          :plugins:
+            - expect
+            - expect_any_args
+            - return_thru_ptr
+            - array
+            - ignore
+            - ignore_arg
+            - callback

0xjakob, 2023-02-27:
@ljden WiFi Mock is on our list, however, I can't give you an exact timeline. We'll keep you updated.

0xjakob, 2023-07-12:
@ljden We have a solution for mocking WiFi which will be similar to what you already tried. I think we should have this on master soon.

0xjakob, 2023-08-17:
@ljden We have merged a mock implementation internally. It will soon be available on master.

SoucheSouche, 2023-10-16:
Closing the issue since the mock was merged internally. It will be available with the next GH sync.

closed 2023-10-16 by SoucheSouche