espressif/esp-idf #10585: When Building idf_as_lib or `Using ESP-IDF in Custom CMake Projects` no way of adding EXTRA_COMPONENTS.

phelter, 2023-01-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

How do you specify the `EXTRA_COMPONENT_DIRS` when building with `idf_as_lib`?
* There is no documentation on how to add EXTRA_COMPONENT_DIRS when building with `idf_as_lib`.

I've tried multiple ways by adding the components.

I am using FetchContent to grab:
```cmake
FetchContent_Declare( esp_aws_iot
    GIT_REPOSITORY  https://github.com/Retlek-Systems-Inc/esp-aws-iot.git
    GIT_TAG         bugfix/cpp-usage-and-other-fixes #bbaf03d7d1fbf8a3f91dc18489d7bd27d5b9e9df # off of master after update feature/202210.01-LTS_update
)
...
FetchContent_GetProperties(esp_aws_iot)
if(NOT esp_aws_iot_POPULATED)
    FetchContent_Populate(esp_aws_iot)
endif()
```

And want to include this in a build via: [Using ESP-IDF in Custom CMake Projects](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects)

Note - the ESP-IDF is the `Custom CMake` and not the other way around.  The ESP-IDF is not standard CMake and does not play well with standard/native CMake at all.

I tried the following:

```cmake
    file(GLOB _component_dirs ${esp_aws_iot_SOURCE_DIR}/libraries/*)
    foreach(_component_dir ${_component_dirs})
        if(IS_DIRECTORY ${_component_dir} AND EXISTS ${_component_dir}/CMakeLists.txt)
            get_filename_component(_component ${_component_dir} NAME)
            idf_build_component(${_component_dir})
            add_subdirectory(${_component_dir} ${esp_aws_iot_BINARY_DIR}/libraries/${_component}})
        endif()
    endforeach()
```
I've tried with and without both `idf_build_component` and `add_subdirectory`

And continually get:
```txt
CMake Error at /opt/esp/idf/tools/cmake/component.cmake:425 (message):
  Called idf_component_register from a non-component directory.
Call Stack (most recent call first):
  build/_deps/esp_aws_iot-src/libraries/Device-Defender-for-AWS-IoT-embedded-sdk/CMakeLists.txt:17 (idf_component_register)
```

### What is the actual behavior?

I get errors that the component_register cannot register because it is a non-component directory.
```txt
CMake Error at /opt/esp/idf/tools/cmake/component.cmake:425 (message):
  Called idf_component_register from a non-component directory.
Call Stack (most recent call first):
  build/_deps/esp_aws_iot-src/libraries/Device-Defender-for-AWS-IoT-embedded-sdk/CMakeLists.txt:17 (idf_component_register)
```

### Steps to reproduce.

1. Step
2. Step
3. Step
...

igrr, 2023-01-21:
I think adding a component to the build should be possible via idf_build_component() before the call to idf_build_process(). That said, if you are planning to use FetchContent, you'd have to tell it to skip the configuration and build steps — as otherwise, as you have pointed out, the configuration step of a subproject fails.

I'll try to put together an example illustrating how to set this up.

igrr, 2023-01-21:
@phelter Please find attach the example of adding an extra component via `idf_build_process()`. This is a patch which can be applied to idf_as_lib example. I made it against the `master` branch, but hopefully it explains the idea. Let me know if you run into any problem replicating this with v4.4.3.

[0001-examples-demonstrate-adding-extra-component-to-idf_a.patch.txt](https://github.com/espressif/esp-idf/files/10472157/0001-examples-demonstrate-adding-extra-component-to-idf_a.patch.txt)

igrr, 2023-01-25:
Closing as the solution has been provided, please feel free to reopen if the issue is still relevant.

closed 2023-02-15 by Alvin1Zhang