espressif/esp-idf #10587: Build Fails when using `CONFIG_ESP_NETIF_LOOPBACK`

AEFeinstein, 2023-01-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expect to be able to build the [hello_world](https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/) example after using `idf.py menuconfig` to disable networking components.

### What is the actual behavior?

The project fails to build.

### Steps to reproduce.

1. Clone the IDF
2. Navigate to the [hello_world](https://github.com/espressif/esp-idf/blob/master/examples/get-started/hello_world/) example
3. Use `idf.py menuconfig` to disable networking components. My full `defconfig` and `sdkconfig` files are posted below
4. Build the project with `idf.py build`
`defconfig`:
```
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_BEFORE_NORESET=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_SPI_MASTER_IN_IRAM=y
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
CONFIG_GPIO_CTRL_FUNC_IN_IRAM=y
CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM=y
# CONFIG_ESP_TLS_USE_DS_PERIPHERAL is not set
# CONFIG_ETH_USE_SPI_ETHERNET is not set
# CONFIG_ESP_HTTP_CLIENT_ENABLE_HTTPS is not set
CONFIG_ESP_NETIF_LOOPBACK=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
# CONFIG_ESP32_WIFI_ENABLE_WPA3_OWE_STA is not set
# CONFIG_ESP_WIFI_SOFTAP_SUPPORT is not set
# CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES is not set
# CONFIG_LWIP_TIMERS_ONDEMAND is not set
# CONFIG_LWIP_SO_REUSE is not set
# CONFIG_LWIP_IP4_FRAG is not set
# CONFIG_LWIP_ESP_GRATUITOUS_ARP is not set
# CONFIG_LWIP_DHCP_DOES_ARP_CHECK is not set
# CONFIG_LWIP_DHCP_DISABLE_VENDOR_CLASS_ID is not set
CONFIG_LWIP_DHCP_OPTIONS_LEN=68
# CONFIG_LWIP_DHCPS is not set
# CONFIG_LWIP_IPV6 is not set
# CONFIG_LWIP_NETIF_LOOPBACK is not set
# CONFIG_LWIP_CHECKSUM_CHECK_ICMP is not set
# CONFIG_LWIP_ICMP is not set
[... 1640 lines cut ...]
~/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.h:50:28: error: unknown type name 'netif_related_data_t'
   50 | void esp_netif_destroy_ppp(netif_related_data_t *netif_related);
      |                            ^~~~~~~~~~~~~~~~~~~~
~/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.h:60:30: error: unknown type name 'netif_related_data_t'
   60 | esp_err_t esp_netif_stop_ppp(netif_related_data_t *netif_related);
      |                              ^~~~~~~~~~~~~~~~~~~~
~/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.h:68:38: error: unknown type name 'netif_related_data_t'
   68 | void esp_netif_ppp_set_default_netif(netif_related_data_t *netif_related);
      |                                      ^~~~~~~~~~~~~~~~~~~~
```
```

### More Information.

The solution is to wrap `/components/esp_netif/lwip/esp_netif_lwip_ppp.h` in 

```
#ifdef CONFIG_ESP_NETIF_TCPIP_LWIP
...
#endif
```

This is necessary because `esp_netif_lwip_internal.h` is already wrapped in `CONFIG_ESP_NETIF_TCPIP_LWIP` which is causing `netif_related_data_t` to be undefined during build. With this change to the IDF, the project compiles and runs.

Separately, if there is a better way to disable networking components and save RAM & ROM for projects that do not require networking, I would love to know how.

david-cermak, 2023-02-03:
Hi @AEFeinstein 

I'm sorry, this config is currently broken on `v5.0`. It's been fixed on master/`v5.1-dev` in 38dec0be489f2caffdc4683c28504af285ad9321 
That commit also adds support for non-lwip configuration, you can check [this test](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/protocols/netif_components) which validates the build without lwIP.

> Separately, if there is a better way to disable networking components and save RAM & ROM for projects that do not require networking, I would love to know how.

You can try to reduce the number of components to explicitly include only those that your project needs:
```cmake
  set(COMPONENTS main ... )   # add only components that you use
```
(similar to our [host-build](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html#requirements) config)

But that would IMO reduce just the build time, since if your project doesn't call those API of certain component it should be removed and won't consume RAM & ROM of your project binary.
So the short answer is "no need to disable the component's build, it's just enough not to use their API"

I'd recommend reading this: [Minimizing Binary Size](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#minimizing-binary-size)

and using `idf.py/size` related tooling:
* `idf.py size-components`
* `idf_size.py --files build/project.map`
* `idf_size.py --archives build/project.map`

AEFeinstein, 2023-02-04:
Thanks for fixing this in `v5.1-dev` and thanks for the tip about explicitly including components!

I had read the [Minimizing Binary Size](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#minimizing-binary-size) page before. Maybe my own memory is fuzzy, but I remember IDF 4.4 not removing as much stuff as IDF 5.0. It was also an oddball project that used ESP-NOW, but not WiFi. I'll figure it out 😃

AEFeinstein, 2023-02-18:
It's not in `v5.0.1`, but you said `v5.1`, so I'll wait for that.

david-cermak, 2023-06-09:
Backported as a3ca732e5d78de9cf1bdfd1d174a3540e6437845