espressif/esp-idf #10589: No way of adding a KConfig at a project level when using `idf_as_lib`

phelter, 2023-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I would like a means of adding KConfig file and have it as part of the build scripts when configuring the project and using idf_as_lib.
I would like a means of adding EXTRA_COMPONENTS and have it as part of the build scripts when configuring the project and using idf_as_lib method of compile.

When using the existing flow [Using ESP-IDF in Custom CMake Projects](https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32/api-guides/build-system.html?highlight=project_include%20cmake#using-esp-idf-in-custom-cmake-projects), there is no means of:
* Adding a project based `KConfig` file 
* Having `EXTRA_COMPONENTS` of third-party libraries that decided to use the `non conformant ESP-IDF Custom CMake flow`.

I would like the ability to do something like:

```cmake

set(PROJ_KCONFIGS KConfig source/any_dir/KConfig ...)
set(EXTRA_COMPONENTS ${esp_aws_iot_sdk_SOURCE_DIR}/libraries )
include($ENV{IDF_PATH}/tools/cmake/idf.cmake)
...
add_subdirectory(source)

idf_build_process("${IDF_TARGET}" ...)

# -----------------------------------------------------------------------
# Has to be at top level as per: https://github.com/espressif/esp-idf/issues/7482
add_executable(app.elf 
        source/apps/this_app/src/app_main.c)

target_link_libraries(app.elf
        PRIVATE
            <local targets>
            idf::log
            idf::esp_system
            idf::newlib
    )
# Let the build system know what the project executable is to attach more targets, dependencies, etc.
idf_build_executable(lomi_main.elf)
....

```

### What is the actual behavior?

EXTRA_COMPONENTS are only supported in project.cmake - and not in the idf.cmake flow.
KConfig are only defined if the main is added as part of the `non-conformant ESP-IDF Custom CMake flow`.

### Steps to reproduce.

1. Using the flow for `idf_as_lib`
2. before doing the changes above in `What is the expected behavior?` perform a FetchContent of esp_aws_iot_sdk
```cmake
FetchContent_Declare( esp_aws_iot
    GIT_REPOSITORY  https://github.com/Retlek-Systems-Inc/esp-aws-iot.git
    GIT_TAG         bugfix/cpp-usage-and-other-fixes #bbaf03d7d1fbf8a3f91dc18489d7bd27d5b9e9df # off of master after update feature/202210.01-LTS_update
)
FetchContent_GetProperties(esp_aws_iot)
if(NOT esp_aws_iot_POPULATED)
    FetchContent_Populate(esp_aws_iot)
endif()
```
3. Add the Expected behavior to the idf_as_lib flow.
5. Compile
...

igrr, 2023-01-21:
@phelter I don't think that the IDF build system has a concept of projet level Kconfig files — they are always part of one of the components. This is the same in "IDF project" and "IDF as library" build flows. The simple solution to this is to create a component with nothing but a CMakeLists.txt and a Kconfig file, and add that component to the build.

Let me know if this isn't clear, i can try to put together an example.

igrr, 2023-01-21:
Regarding adding extra components, I'll answer in #10585  which looks like the same issue.

phelter, 2023-01-21:
Thanks igrr.

Unfortunately there is no way that I have found to add additional idf_build_components() using the idf_as_lib flow.

I tried adding them before and after the `include($ESP(IDF_PATH)/tools/cmake/idf.cmake)` and get various errors on both locations.

I have decided to group and duplicate all of the native cmake code into the non-conforming ESP-IDF CMake and switch between the two based on a CMAKE option.  This at least allows me to include extra components with the `EXTRA_COMPONENTS_DIR` option when including `project.cmake`.  An ugly hack of skipping directories and alternating between parent and child directories using one or the other method has allowed me to mix and match the two flows in a very unorthodox manner.

Please see the following comment: https://github.com/espressif/esp-idf/issues/9929#issuecomment-1398803031

This is the style that I typically use for CMake files and has proven quite stable and useful.  I have added functions on top of cmake to create Mock targets and Test targets but all using the standard cmake properties that are provided instead of creating a secondary database of similar functions like the ESP-IDF non-compliant CMake does.

igrr, 2023-01-21:
Okay, thanks for trying that. I'll try to prepare an example for you next week.

(I've seen your comment in the other issue, and I appreciate you giving the suggestion. There are a few reasons why that approach is not used in IDF, i will write a more detailed explanation later. Your comments do make it clear that compatibility with standard CMake projects is something some of our users do care about, so we'll add weight to various build system refactoring tasks in the future!)

igrr, 2023-01-21:
> Unfortunately there is no way that I have found to add additional idf_build_components() using the idf_as_lib flow.

I have now added an example of doing this in the issue you have opened about adding extra components: https://github.com/espressif/esp-idf/issues/10585#issuecomment-1399242345

Regarding the topic of this issue, here is an example of adding Kconfig options to the idf_as_lib example. This is a patch file you can apply to the idf_as_lib example. Same is in the other issue, I made the patch relative to the `master` branch, but it should be very similar for v4.4.3. Let me know if you run into any problem replicating this in v4.4.3.

[0001-examples-demonstrate-adding-project-level-configurat.patch.txt](https://github.com/espressif/esp-idf/files/10472171/0001-examples-demonstrate-adding-project-level-configurat.patch.txt)

igrr, 2023-01-25:
@phelter I'll close this one and #10585 since the solution has been provided, and you have solved your problem in a different way in the meantime.

closed 2023-01-25 by igrr