espressif/esp-idf #10590: ESP32-S2 RISC-V ULP, Re-running once booted, needs extra call

cnlohr, 2023-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If I am running the ESP32-S2, and program in a ULP binary, and run it, all works well.  I can also perform a `ulp_riscv_halt()` and it works well.  Then if I load a new binary onto the ULP, and call `ulp_riscv_run()`  it does not run.  I have to call the following command to cause the ULP to start again:

```
CLEAR_PERI_REG_MASK( RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE );
```

Should this be included in `ulp_riscv.c`? or am I misunderstanding how this works.

cnlohr, 2023-01-21:
It looks like the following line is called from within the program.  I am not sure why it is not working for me.

```c
CLEAR_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE | RTC_CNTL_COCPU_SHUT_RESET_EN);
```

from

```c
ulp_riscv_rescue_from_monitor();
```

cnlohr, 2023-01-21:
It seems something when wrong in initialization.  Once I resolved that, it worked.  That seems really risky though.  Is there a reason not to clear from the application processor as well?

sudeep-mohanty, 2023-01-25:
Hi @cnlohr. 
`RTC_CNTL_COCPU_DONE` is cleared before the ULP RISC-V core can run. This is handled automatically via the `ulp_riscv_rescue_from_monitor()` call made in [ULP startup code](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/ulp_core/start.S#L24) when the ULP core resets. I'm not sure about the code flow you are using which prevents this. Could you maybe share some of your application code with us to review?

cnlohr, 2023-01-27:
I do not have a situation now which will result in a failure-to-start of the ULP.  I am just concerned that in this system, because if the ULP ever fails to start the code, there is no way from the main processor to make it work again.  It just stays stuck.  Verses if it was just reset when loading the image.

For our system it is reliably starting, but I think we're going to add the extra reset on load just in case something goes wrong because previously in our development cycles it did (though it was our fault).

I don't understand the rationale to making the ULP responsible for doing something to make the ULP usable.  Though I acknowledge that we may have been the only people with this issue.

sudeep-mohanty, 2023-02-02:
Hi @cnlohr,
Apologies for replying late to your last query. It is indeed an issue if the ULP RISC-V core crashes. Currently there is no way to recover it from the main core. I have been able to recreate the problem by deliberately loading a faulty firmware on the ULP and making it crash. We do not have an API to help recover the ULP to a runnable state therefore I will add a `ulp_riscv_reset()` API as follows -

```
void ulp_riscv_reset()
{
    CLEAR_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT | RTC_CNTL_COCPU_DONE);
    CLEAR_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);
    esp_rom_delay_us(20);
    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT | RTC_CNTL_COCPU_DONE);
    SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_RESET_EN);
}
```

This intended use of this API is to cover scenarios such as the one I described here or the one you faced where in the ULP fails to start for any reason. Once the core is reset, you could re-run the core again or you could load a different firmware on the core and run it.

Do have a look and let us know if this helps. I would be updating the driver API in a follow up change to this.

cnlohr, 2023-02-10:
I was already doing this :).

I just wanted to understand if there was any reason not to do it from the main CPU.

It sounds like it should be fine.

sudeep-mohanty, 2023-02-13:
Thanks for the update @cnlohr. I'll be closing this issue. Please feel free to reopen it or open a new one in case you do encounter the problem again. Thanks!

closed 2023-02-13 by sudeep-mohanty